JMRI/JMRI #2423: DCCpp Ops Mode Programmer Times Out

msunderwd, 2016-10-24:
The DCC++ Ops Mode Programmer times out waiting for a response that the DCC++ Base Station is never going to send.

Reported on Trainboard.com by Geoff McDermott

bobjacobsen, 2016-11-06:
From a jmriusers discussion of this:

> On Nov 3, 2016, at 3:31 PM, crusader2752939@gmail.com [jmriusers] <jmriusers@yahoogroups.com> wrote:
> 
> I'm talking about DCC++, which already has support, but I'd like to look at the configuration to verify that it was done correctly. I've tried multiple ways to contact the person I think was the author of the DCC++ setup for JMRI without success, so I thought I'd take a look at hiw work.
> 
> The issue is simple.....the program on main function doesn't seem to work correctly (it wait & times out when a CV is written), and I've checked the DCC++ code, and it doesn't look for a response.
> 
> So, the DCC++ code doesn't appear to bethe issue, which leaves the DCC++ to JMRI interface. I haven't the slightest idea where top begin.....I'm a retired computyer geek(HW & SW), and don't have much JAVA experience, but I can learn.
> 
> Please give me a place to start, and I'll see if I can do what I need to do to verify that the interface has been done correctly, at least related to programming on the main.

The ops-mode programmer implementation for that system is in:

java/src/jmri/jmrix/dccpp/DCCppOpsModeProgrammer.java

The write-CV code starts around line 53. It does set a timeout waiting for a reply from the command station.

It looks like this created by copying the XPressNet implementation, see comments in several places. That listens for a reply from the command station with code around line 108.

If the DCC++ command station doesn’t send anything back, you should modify the line-53 code to wait a bit (couple hundred msec) to make sure that the command was sent, then do the code from the line-108 routine.

msunderwd, 2016-11-06:
I have a fix for this in test. Will be issuing a PR as soon as I'm sure it
doesn't break anything else.

bobjacobsen, 2016-11-13:
Does #2495 fix this?

msunderwd, 2016-11-13:
Yes it does.

bobjacobsen, 2016-11-13:
Closed via #2495

closed 2016-11-13 by bobjacobsen