JMRI/JMRI #2427: Windows Installer: RFE

Breezlys, 2016-10-25:
From SF #598:

Is it possible to modify the Windows installer to allow the user to point the backup to a specific location?

I prefer to not keep my backups on the same physical drive as my "production" files. Allowing the location to be specified in the installer would save the time/trouble of moving the backups later.

Thanks!

devel-bobm, 2016-10-27:
I believe that it is possible to modify the configuration file for the JMRI Windows installer to allow users to specify the destination for installation-time back-up of pre-existing JMRI user data.  

 

I suspect that it is not a particularly difficult thing to implement, but it is beyond my current level of NSIS configuration knowledge.

mattharris, 2016-10-27:
I'll take a look at the options.

rhwood, 2016-10-27:
This is not trivial without essentially  running JMRI itself. The users data is not in a predefined location (there are defaults, but there is not, and has not been, any requirement that anything other than the (since 3.6 was released) a configuration file telling JMRI which configuration profile was last used by DecoderPro, PanelPro, or other launchers, and a file containing a list of known configuration profiles be in the default location.

To completely backup the users data, the catalog would need to be read, and every configuration profile read, and each configuration profile directory, each users files directory, and each roster directory copied to the backup location.

It might be better to recommend that users backup their data using an external backup tool prior to installation.

Randall Wood

Breezlys, 2016-10-27:
@bobm & @mattharris  - Thanks for looking at this.

@rhwood  Something is getting backed up now; I don't know if it includes user data or not.  But I'm not asking that *what* gets backed up be changed, I'd just like the option of specifying *where* those backup files are placed.

dheap, 2016-10-28:
My understanding (I can't check it just now due to a nearby thunderstorm) is that all that is getting backed up is the Settings Location, which is (necessarily) an immutable path based on the user's home directory (and is the known 

As @rhwood has already pointed out, this no longer necessarily includes user data. In many cases it will not, and this cannot really be determined without starting JMRI.

Simply having the option to backup is actually giving users a false sense of security. Expanding this to allow choice of location is giving further credibility to this myth.

> @Breezlys Something is getting backed up now; I don't know if it includes user data or not. But I'm not asking that what gets backed up be changed, I'd just like the option of specifying where those backup files are placed

mattharris, 2016-10-28:
Indeed, what's getting backed-up right now is the contents of the Settings Location.

Now, I would suggest that with a significant majority of MS Windows JMRI installations, this backup will include all the user data as I feel that only a small (but growing) proportion of users change the paths from the out-of-the-box defaults.

Now, it might be worth rewording the text in the dialog box to emphasise the fact that, if default storage locations have been changed, that these are not included in any backup. (OK, the _best_ thing would be to also backup user data from non-default locations, but that's a whole other kettle of fish).

As far as some of the history behind this feature goes, it was initially implemented way before we had the notion of movable user data locations and was also to deal with any idiosyncrasies between the old InstallerVISE-based installer and the NSIS installer and to ensure that there was a possibility to revert in the case of a borked installation.

Breezlys, 2017-02-23:
As one of the "significant majority of MS Windows JMRI installations", all my user data is in the default location**.  And my day job includes DR/BR planning, configuration, and execution, so I _really_ don't like having my backups reside next to my live data.

So what I've done is schedule a task to run a Robocopy batch job every night to mirror any changes of the JMRI backup folder to my backup location.  Problem solved, I'm closing this request.

**The exception is my Roster, which is on a network share that's managed/backed up separately.

closed 2017-02-23 by Breezlys

dheap, 2017-02-23:
That's one of the reasons many of us move our entire "User Files Location" rather than just "Roster" to  a shared location, such as Dropbox. Profiles are relatively easy to reconstruct, but "User Data" is not.

I have a history that includes time in a Data Centre and am a strong believer in the 3-2-1 rule.

Dropbox doesn't fully cover the "1" rule so that is where commercial solutions such as Backblaze and CrashPlan come into play. If I lost my entire property in a fire or similar event, none of my data would be lost.

"2" is getting harder these days as our data storage requirements far exceed optical or other non-magnetic media capacity.

-- 
Dave in Australia

On 23 Feb 2017, at 1:17 pm, Breezlys <notifications@github.com> wrote:

> As one of the "significant majority of MS Windows JMRI installations", all my user data is in the default location**. And my day job includes DR/BR planning, configuration, and execution, so Ireally [...]