JMRI/JMRI #2430: Incorrect handling of "CV number not supported" exception.

dheap, 2016-10-26:
The  NCE global programmer blocks writes to any CV>256 (firmware issue) and unless the decoder manufacturer provides an in-decoder workaround (ESU and ZIMO do), throws an exception error.

Neither the Single CV Programmer or the Indexed Programmer Facade are currently handling this exception error correctly (the problems can be demonstrated with the NCE Simulator).

The Single CV Programmer logs a status error but does not release the programmer.

The Indexed Programmer Facade throws an exception error in the log and hangs the programmer.

dheap, 2016-10-26:
Indexed Programmer Facade throws an exception error in the log and appears to hang the programmer. Subsequent attempts to write return a "Programmer in Use" message.

dheap, 2021-03-31:
Will check this one.

dheap, 2021-05-15:
This needs fixing, but is not easy to do, in my view.

danielb987, 2021-05-15:
@dheap 
Do you have some sort of idea on how it should be done? I don't have the solution and I'm not sure I understand the problem, but by open up a discussion about the problem may give a path towards a possible solution.

bobjacobsen, 2021-05-16:
Might be best to not throw the exception for this, and instead have the NCE code pop a message that says "Update your firmware" with a link.

dheap, 2021-05-16:
I can't check at this time but I have a feeling that the exception isn't thrown by NCE-specific code.

If someone can make an NCE Simulator profile with any current JMRI version and try writing any indexed CV from a TSU2/Econami definition and/or use Single CV Programmer to write any CV>256 we could get some logs.

My view is that we should treat the error as I think we do for any other write (or confirm after write) failure. I don't think they lock up with Programmer in Use, but I'll admit I haven't tried. That too could be checked easily with NCE Simulator and any low (1-256) CV by enabling confirm after write (a confirm will always return 123) Sorry I can't check tonight...

PR #9809 (2021-05-20): Better handling of NCE Power Pro write CV>256 blocking.
Resolves #2430 and numerous user complaints.

No longer locks up programmer.
closed 2021-05-20 by danielb987