JMRI/JMRI #2436: LocoNet throttle always "steals" the slot

devel-bobm, 2016-10-29:
When a JMRI-based throttle (or JMRI throttle controlled by WiThrottle or Web Server) acquires a locomotive or consist, the LocoNet code always ignores the "slot status" and always performs the <OPC_MOVE_SLOT> "Null Move" regardless of the slot status.  This means JMRI will always perform the equivalent of a "slot steal" operation whenever the slot was "in-use" by another throttle.  (When a LocoNet slot is "stolen", it is really "shared", as any other throttle using the slot does not relinquish its contol over the slot.)

@bobjacobsen and other knowledgable contributors - Is this behavior intentional?  I think that this implementation would be an easy way to avoid the whole question of how to get the associated throttle "user interface" to ask the user if he/she wants to "steal" (i.e. share) control of the slot, and then get the user reply back into the middle of the LocoNet locomotive acquisition process.

Resolution of issue #2435 may require changes to the behavior described above.

devel-bobm, 2017-08-09:
This behavior has made it possible for WiThrottle Server to grab a slot without any WiThrottle Server protocol interaction with the throttle in the case where another throttle has the slot.  So far as I can tell, the WiThrottle Server protocol did not have any sort of mechanism to handle the Digitrax "steal" mechanism.

Curiously, the introduction of the Digitrax LNWI device may have induced a change to the WiThrottle Server protocol.  A "Knowledge Base" article by Digitrax (KB # 1055, at http://www.digitrax.com/tsd/KB1055/faq-what-lnwi/ ) describing the LNWI device, includes the following quote:

> Digitrax worked with Brett Hoffman (the protocol developer) and Steve Todd another APP developer from mid-June 2017 to ensure the existing Apple and Android apps would work prior to LnWi release.  During this software collaboration a number of useful things were added to the; LnWi, Hoffman protocol and the APPS, such as automatic LocoNet Server setups and mapping the existing LocoNet throttle stealing protection back into the protocol. 

This implies that there may be new features in the WiThrottle Server protocol to allow interactive handling of the LocoNet "steal" use-case.  If so, the LocoNet throttle should be changed to take advantage of those features.  This likely will require changes also to the JMRI "Throttle", so that it too can handle the LocoNet "steal" case.

devel-bobm, 2017-08-09:
@BHoffman351, do you have any input on this?

BHoffman351, 2017-08-09:
> On Aug 9, 2017, at 4:08 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> This behavior has made it possible for WiThrottle Server to grab a slot without any WiThrottle Server protocol interaction with the throttle in the case where another throttle has the slot. So far as I can tell, the WiThrottle Server protocol did not have any sort of mechanism to handle the Digitrax "steal" mechanism.
> 
That is correct, there was no behavior in the WiThrottle protocol to handle a steal.  But if I am not mistaken, there was no JMRI behavior to handle a steal.  JMRI just does a steal type acquisition.
> This implies that there may be new features in the WiThrottle Server protocol to allow interactive handling of the LocoNet "steal" use-case. If so, the LocoNet throttle should be changed to take [...]
Yes, the WiThrottle and Engine Driver apps both have changes to handle a steal.  If steal is implemented in JMRI, we can handle making the changes to the withrottle package to integrate it.

Basically if the address is listed as in-use, it should not automatically be acquired by JMRI, a question is presented to the user whether to steal.  If yes, it steals.  If no, the request dies silently.

Regards,
Brett

pabender, 2017-08-09:
Sent from my iPhone

> On Aug 9, 2017, at 5:44 PM, BHoffman351 <notifications@github.com> wrote:
> Yes, the WiThrottle and Engine Driver apps both have changes to handle a steal. If steal is implemented in JMRI, we can handle making the changes to the withrottle package to integrate it.
> 
> Basically if the address is listed as in-use, it should not automatically be acquired by JMRI, a question is presented to the user whether to steal. If yes, it steals. If no, the request dies silently.

The current way the throttle request works is:
1) Throttle UI requests throttle from manager
2) the throttle manager calls the throttle listener's
      A) notifyThrottleFound method if the address is found
      B) notifyFailedThrottleRequest if the request fails.

I see two ways to proceed:
1) add a new method to the listener ( notifyStealAddress perhaps ) that could trigger the steal request
2) specifically test the reason code passed in the notifyFailedThrottleRequest for a trigger value that could trigger the steal request.

In either case, a new method would need to be added to cause the throttle manager to continue with the request.  If the steal is confirmed, then notifyThrottleFound needs to be called.  If the steal is denied, the notifyFailledThrottleRequest needs to be called.

Throttle managers for systems that don't support stealing ( which is most of them ) will never send the steal trigger and will have an empty method for the continue request ( or one that always respond as though the steal occurred ).

This isn't quite a silent failure, because the error message displayed with the failure request will be displayed, but it provides the least disruption to all the other systems.

Paul

BHoffman351, 2017-08-09:
> On Aug 9, 2017, at 7:20 PM, Paul Bender <notifications@github.com> wrote:
> 
> This isn't quite a silent failure, because the error message displayed with the failure request will be displayed, but it provides the least disruption to all the other systems.

Sorry, by "dies silently” I meant that no further action is taken at the WiThrottle end if the user chooses not to steal.  I was not trying to suggest that for the JMRI end.

Regards,
Brett

devel-bobm, 2017-08-10:
I have investigated this previously and can work on the necessary changes to loconet.LocoNetThrottle and loconet.SlotManager.  But we need a mechanism for the end-to-end solution, which includes at least the AbstractThrottle and "WiThrottle Server" portions of the chain.

pabender, 2017-08-10:
> On Aug 9, 2017, at 8:28 PM, Bob Milhaupt <notifications@github.com> wrote:
> I have investigated this previously and can work on the necessary changes to loconet.LocoNetThrottle and loconet.SlotManager. But we need a mechanism for the end-to-end solution, which includes at least the AbstractThrottle and "WiThrottle Server" portions of the chain.
> 
Give me a couple of hours and I will have a PR together that handles the required changes to the throttle listeners and the throttle UI, along with the default implementation of the steal reply method in the abstract throttle.

Paul

BHoffman351, 2017-08-10:
> On Aug 9, 2017, at 8:38 PM, Paul Bender <notifications@github.com> wrote:
> 
> Give me a couple of hours and I will have a PR together that handles the required changes to the throttle listeners and the throttle UI, along with the default implementation of the steal reply method in the abstract throttle.

I’ll finish the wi part after that, probably tomorrow.  Even without the wi change, it should still work normally.

The WiT/ED do not send a “cancel” if they do not want the steal action, they only send a steal confirmation.  (That’s what I meant about silent)

Regards,
Brett

devel-bobm, 2017-08-10:
I do not expect to have time to work on this until Monday.

The "silent" response case for "do not steal" (and "user didn't reply") implies that the steal response mechanism will need to implement some sort of a timer to give a "release" on LocoNet instead of a "steal".  Does the WiT/ED have any sort of a timeout on the "steal" message?  The LocoNet implementation ought to implement a timer of a longer duration thhe WiT/ED throttle.

pabender, 2017-08-10:
Sent from my iPad

> On Aug 9, 2017, at 9:14 PM, BHoffman351 <notifications@github.com> wrote:
> > On Aug 9, 2017, at 8:38 PM, Paul Bender <notifications@github.com> wrote:
> > 
> > Give me a couple of hours and I will have a PR together that handles the required changes to the throttle listeners and the throttle UI, along with the default implementation of the steal reply method in the abstract throttle.
> 
> I’ll finish the wi part after that, probably tomorrow. Even without the wi change, it should still work normally.

Ok, PR #3904 has the basic throttle manager code in place.  Standard tests are running on that now.

> 
> The WiT/ED do not send a “cancel” if they do not want the steal action, they only send a steal confirmation. (That’s what I meant about silent)

Ok.  We have to clean up some state in the throttle managers if the request is canceled. In particular, there is a list of throttle listeners for active requests that can impact future requests for the given address. ( internally, JMRI can have many things watching a throttle, hence the list of listeners for an address request ).

Paul

devel-bobm, 2017-08-10:
@pabender, be aware that newer LocoNet throttles have a feature called "StealZap".  This means that a throttle can be forcibly-removed from control of a slot.  The command station can issue the "StealZap" message "at any time" in a given throttle's operation.  But, so far as I can tell, the command station will only issue "StealZap" in the case where throttle B is trying to acquire a loco which is already acquired by throttle A, and the throttle B "steals" the slot.  In this case, the command station issues the "StealZap" message to throttle A.

In this scenario, we need to be able to tell a JMRI throttle that it no longer has control of the locomotive.  And we probably need to inform the listeners of that throttle that the throttle has lost control of the loco.

Of course, this begs the question: in this case, should any of the "throttle listeners" be "transitioned" from "listening to throttle A" to "listening to throttle B".  That seems awkward.  But a "silent death" of a throttle, and thus a listener's connection to a throttle is awkward, too.  Thoughts?

devel-bobm, 2017-08-10:
Including @mstevetodd .

devel-bobm, 2017-08-10:
A clarification on "StealZap".

A "steal" operation has two affected throttles - the one which "steals", and the one which was "stolen from".  StealZap affects the one which was "stolen from", and ensures that this throttle can no longer affect the loco.

pabender, 2017-08-10:
Sent from my iPad

> On Aug 10, 2017, at 6:16 AM, Bob Milhaupt <notifications@github.com> wrote:
> In this scenario, we need to be able to tell a JMRI throttle that it no longer has control of the locomotive. And we probably need to inform the listeners of that throttle that the throttle has lost control of the loco.

We already do this type of thing in the XpressNet throttle.  XpressNet has had a notification that another device has taken over for as long as any of us have been working with the protocol.  This doesn't cause the throttle to give up control, but it causes notifies the throttle that it needs to check with the command station about status before proceeding.  

In the XpressNet case, we start playing follow the leader when we get this message.  
That lets scripts, and other property change listeners for the throttle object, react to changes in the throttle even if JMRI is not in control.  

If JMRI needs to start controlling the locomotive again then we just take over again, which causes a notification to the other throttle.

All of this happens internally to the XpressNet throttle.

Because we still can monitor the slot, even the if we don't have control, this can, and probably should, be replicated in the Loconet throttle in the case of being forcibly removed from control of the slot.
> Of course, this begs the question: in this case, should any of the "throttle listeners" be "transitioned" from "listening to throttle A" to "listening to throttle B". That seems awkward. But a "silent death" of a throttle, and thus a listener's connection to a throttle is awkward, too. Thoughts?
> 
See above, but remember that throttleListener objects are no longer listening for changes once an address is found.  PropertyChangeListeners attached to a throttle are listing to a throttle attached to a specific address.  They expect to see changes for that address, but they usually don't care if they are in control.

Paul

pabender, 2017-08-10:
#3904 has been merged.

devel-bobm, 2017-08-11:
LocoNet will not be able to act in a way that is 100% analogous to the behavior of JMRI w.r.t. XpressNet throttles.  Some throttles have only "simplex" (unidirectional) communication with LocoNet when unplugged from LocoNet, so there is no guarantee that a LocoNet device is able to inform another device that it no longer has control.

> See above, but remember that throttleListener objects are no longer listening for changes once an address is found. PropertyChangeListeners attached to a throttle are listing to a throttle attached [...]

There's a nuance I was not aware of!

So, from a LocoNet perspective, `PropertyChangeListener`s perhaps could be attached to the associated LocoNet slot (which contains a mobile decoder address) instead of being attached to the JMRI "Throttle" object.  That way, the listener always knows what is going on with the slot, regardless of "who" has control of the slot.

But perhaps there are uses of `PropertyChangeListener`s which need to know that the JMRI "Throttle" object has been disposed of.  If the "PropertyChangeListener`s were attached to the slot, then the listener would not see a "dispose" activity.  That will require some thought and investigation...

pabender, 2017-08-11:
On 08/10/2017 10:03 PM, Bob Milhaupt wrote:
> LocoNet will not be able to act in a way that is 100% analogous to the
> behavior of JMRI w.r.t. XpressNet throttles. Some throttles have only
> "simplex" (unidirectional) communication with LocoNet [...]

In that case, the forced removal is moot.  As long as there are any 
throttles that can't be notified of the change, then you can't trust 
that operation.

Is there any information in the slot that tells us we are actually in 
control of it?  I just know it says it is "in use" or "free" or (a 
couple of consist related statuses...).

As far as mimicking what the XpressNet throttle does, JMRI currently 
does that for LocoNet throttles.  When another device is controlling a 
loconet throttle, the JMRI throttle will play follow the leader just 
from the traffic.

The only question becomes what do you do when someone using a JMRI 
throttle that previously had control but lost control.  My suggestion 
would be that if you see the forced removal message, the JMRI throttle 
sets a flag so that it remembers it is not in control.  If there comes a 
time when the throttle needs to be in control, then it can either steal 
the slot (automatically or with a question ) or just start sending 
commands (the shared slot ).

There are plenty of things that JMRI does with throttles (The Virtual 
Sound Decoder, for example) where it never actually has to have control 
of the throttle, but JMRI needs to listen for property changes from the 
throttle to function as designed.

> See above, but remember that throttleListener objects are no longer
>     listening for changes once an address is found.
>     PropertyChangeListeners attached to a throttle are listing to a [...]

Just remember that the LocoNet throttle IS a SlotListener, and you will 
break generic tools (and scripts) that are acting as 
propertyChangeListener's on a throttle object if you force them to be 
listeners on the slot.

The expectation is that we can have a throttle listener that listens for 
changes (on systems that report changes to us or that we can poll for 
status ) without JMRI ever actually controlling the throttle.

In other words, don't do anything with the listener structure here. 
You're going to break features of JMRI if you try to change it.

> But perhaps there are uses of |PropertyChangeListener|s which need to
> know that the JMRI "Throttle" object has been disposed of. If the
> "PropertyChangeListener`s were attached to the slot, then the listener
> would not see a "dispose" activity. That will require some thought and
> investigation.

If the throttle is disposed (in the java sense ) then the listeners will 
just stop receiving updates (more correctly, the the throttle object 
maintains a list of property change listeners, and it is supposed to 
clear it's list of listeners when it is disposed ).

There is no reason to dispose of the JMRI throttle object if it is no 
longer in control of the slot.  It will just keep reporting changes to 
the listeners as appropriate.

Paul

devel-bobm, 2017-08-11:
> Is  there any information in the slot that tells us we are actually in control of it?  I just know it says it is "in use" or "free" or (a couple of consist related statuses...).

Yes and no.  

With old Digitrax command stations and with new Digitrax command stations when expanded slots are disabled, it is impossible to tell who has control.  it is only possible to know if the DCC track signal gets information from the slot.  There may be some hints in the "ID" bytes about the identify of the last user, but they cannot be used for a positive identification - those bytes are not cleared when a LocoNet throttle stops using a slot, and many Digitrax throttle models do not have unique ID values assigned anyways.  

With a new Digitrax command station and with expanded slots enabled, Yes, sort of.  When used with a throttle that is not using expanded slots, or when the expanded slot feature is turned off, the same limitations as with the old Digitrax command station (above) apply.  When a new Digitrax command station has expanded slots enabled, and a throttle uses expanded slot messaging to access slots, then there is a "positive" identification of the assigned throttle.  Any device which attempts to control the slot sends its identity info as part of the expanded slot messaging.  If the command station sees that a slot control message has the wrong identity information, the LocoNet message will get a LONG_ACK "error" response, and the control message is ignored.

> As far as mimicking what the XpressNet throttle does, JMRI currently does that for LocoNet throttles.  When another device is controlling a loconet throttle, the JMRI throttle will play follow the leader just from the traffic.

This is a "classic" "slot shadowing" operation on LocoNet - two throttles "have control" of the same slot, and the most-recent command sent by either throttle "wins".  But it doesn't work with expanded slots.

> The only question becomes what do you do when someone using a JMRI throttle that previously had control but lost control.  My suggestion would be that if you see the forced removal message, the JMRI [...]

There are a couple of obvious challenges with this.  

- Only the newer command stations, with expanded slots enabled, can tell an expanded-slot throttle that it has lost control of a slot  Older command stations, and/or throttles controlling using traditional slot mesages, cannot provide such messaging.
- The Digitrax command station can move slots around.  So a JMRI throttle which was controlling loco address 715 (in slot 35), and which was forced to lose control by another throttle acquiring the loco, cannot simply remember "address 715" and "slot 35", because it may not be in slot 35 anymore!
- If the JMRI LocoNet SlotManager or LocoNet ThrottleManager must "steal-back" the throttle when JMRI or a script decides it needs to control the slot, then there is the possiblity of slot "thrashing", especially if more than one JMRI computer is tied to LocoNet, and each JMRI computer is paying attention to a given slot.

> There are plenty of things that JMRI does with throttles (The Virtual Sound Decoder, for example) where it never actually has to have control of the throttle, but JMRI needs to listen for property changes from the throttle to function as designed.

This is not going to work with expanded slots!  The act of "acquiring" a loco address on LocoNet really is "I want to control" the loco.  Something like VSD is using LocoNet messaging to "acquire" a slot, even though it only wants to "watch" the slot.  With expanded slots, as soon as the "acquire" operation is completed, the other throttle gets a message from the command station saying that it no longer has control.

Things like VSD which perform a "monitor-only" function need some sort of new interface to request that "monitor-only" function.  Other command stations may choose to implement the "monitor-only" function as a locomotive acquire, but it won't work for LocoNet and its expanded slots, per above.  For LocoNet, the "slot manager" would use that new "monitor-only" request to attach the requestor's `PropertyChangeListener` to the appropriate LocoNet slot.  And the Slot manager would be responsible for keeping up with slot moves, and move the requestor's `PropertyChangeListener` as the command station moves the controlled loco address from one slot to another.  

> The expectation is that we can have a throttle listener that listens for changes (on systems that report changes to us or that we can poll for status ) without JMRI ever actually controlling the throttle.

Per above, this is not going to work with LocoNet expanded slots if JMRI uses a JMRI throttle to "acquire" a loco in order to get this "monitor-only" function.

> In other words, don't do anything with the listener structure here.  You're going to break features of JMRI if you try to change it.

Per above, the problem is that JMRI's previous assumptions are breaking with respect to new Digitrax features.  We need to find a way that JMRI functionality can co-exist with new Digitrax features.

> There is no reason to dispose of the JMRI throttle object if it is no longer in control of the slot.  It will just keep reporting changes to the listeners as appropriate.

This is not true in some cases.  Since with newer Digitrax command stations a JMRI throttle acquisition can force another LocoNet throttle to lose control of a loco, we've lost system-level functionality.  Since a JMRI throttle is being used in some cases solely to monitor an actual throttle's behavior, and since that JMRI throttle can take away that actual throttle's ability to control the loco, we need some new mechanism for those cases.  That would restore some of that lost functionality.

The cases where JMRI wants to provide some level of control of a slot while another LocoNet agent _also_ has control of the slot is problematic in the expanded-slot environment, and needs further study.

devel-bobm, 2017-08-16:
Sorry I haven't gotten to this - life got in the way.  Perhaps I will be able to work on this Thursday or Friday this week.

devel-bobm, 2017-08-21:
Current analysis reveals a challenge.  

LocoNet process for acquiring an engine places the "steal" request before the actual acquisition of the "slot" containing the engine.  `LnThrottleManager` handles this for LocoNet, before creating the `LocoNetThrottle` object.  But creation of a `LocoNetThrottle` object actually marks the slot as "in-use" (i.e. "acquiring" the slot).  The "steal" request method which @pabender has added is an abstract method in `AbstractThrottle`.  So a LocoNet concrete implementation of this "places the cart before the horse", so to speak.

From my LocoNet-centric perspective, it would have been better to implement the "steal" request as part of the `AbstractThrottleManager` instead of the `AbstractThrottle`.

With the existing implementation, I think that it will be necessary to move at least some of the "LocoNet message checking" from `LnThrottleManager` to `LocoNetThrottle`.  I will investigate more as time permits.

pabender, 2017-08-21:
@devel-bobm,

Sent from my iPad

> On Aug 21, 2017, at 12:21 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> Current analysis reveals a challenge.
> 
> LocoNet process for acquiring an engine places the "steal" request before the actual acquisition of the "slot" containing the engine. LnThrottleManager handles this for LocoNet, before creating the LocoNetThrottle object. But creation of a LocoNetThrottle object actually marks the slot as "in-use" (i.e. "acquiring" the slot). The "steal" request method which @pabender has added is an abstract method in AbstractThrottle. So a LocoNet concrete implementation of this "places the cart before the horse", so to speak.
> 
> From my LocoNet-centric perspective, it would have been better to implement the "steal" request as part of the AbstractThrottleManager instead of the AbstractThrottle.
> 
> With the existing implementation, I think that it will be necessary to move at least some of the "LocoNet message checking" from LnThrottleManager to LocoNetThrottle. I will investigate more as time permits.
> 

You are reading that wrong.

The throttle listener interface ( that is the interface for objects requesting the throttles, which includes the throttle front ends ) implement a callback function that allows the throttle managers to notify them that a steal is required.

If the throttle listener wishes to complete the steal, it calls one of the stealThrottleRequest methods specified in the throttle manager interface.  The abstractThrottleManager provides default stealThrottleRequest methods, that do nothing.

The throttle interface hasn't been changed, so the system specific throttles don't know anything about the stealing behavior.

I'll see if I can diagram the sequence tomorrow ( ok, it probably won't be tomorrow, first day of class for me. )

Paul

devel-bobm, 2017-08-21:
@pabender, I've got a better understanding now - the GUI's `ThrottleListener` "pre-exists" when the connection-specific throttle request is made.  But I don't see how the connection-specific ThrottleManager has access to the listener from its `requestThrottleSetup` method.  Here's my reasoning.
 
The `ThrottleListener` is a callback method of the "GUI Throttle".  This pre-existing listener is passed to the `AbstractThrottleManager` as an argument to the `requestThrottle` method.
 
It appears that within the `requestThrottle` method in `AbstractThrottleManager`, the reference to the `ThrottleListener` object (from the GUI Throttle) is not passed to the connection-specific `requestThrottleSetup` method.  And `AbstractThrottleManager` keeps the `ThrottleListener` reference in a "local" ArrayList, with no visibility outside of that method.
 
So I don't see how the connection-specific ThrottleManager can have access to the GUI's `ThrottleListener`.
 
What am I missing here?

pabender, 2017-08-21:
@devel-bobm 

See line 212 of AbstractThrotlteManager:

https://github.com/JMRI/JMRI/blob/08478871aefec271db1e70b2f35a59004aaa8e9c/java/src/jmri/jmrix/AbstractThrottleManager.java#L212

That puts the throttle listener into a Hashmap where an arrayList of throttle listeners is associated with each requested throttle.  That is how the manager keeps track of the active throttle listeners.

Paul

devel-bobm, 2017-08-21:
@pabender, I'm still missing somthing, because line 122 of `AbstractThrottleManager` declares `throttleListeners` as "private", and I don't see any "setters" and "getters" declared.

pabender, 2017-08-21:
@devel-bobm 

Ok, the missing piece is how do the system specific throttle managers notify throttles that something happened.  That happens in the failedThrottleRequest and notifyThrottleKnown methods of the AbstractThrottleManager. 

So there is a missing method in the abstract throttle manager to notify all the active listeners that a steal is pending.  

Note that neither of the methods above are actually a part of the throttleManager interface, so while they are listed as public they aren't things that are expected to be called by non-descendants (My guess is they are public for testing purposes, but I would have to go back and look at the history of the file to see if that is right.)

Paul

devel-bobm, 2017-08-21:
@pabender,
In this context, we only need to notify a "steal" to the throttle which is requesting the throttle, not "all active listeners".  We don't want to tell all (GUI)throttles that someone is trying to steal their loco, but we do want to notify the throttle trying to acquire a loco that a "steal" operation would be needed, and give him/her the ability to choose.
 
Does "all active listeners" in this context only include those (GUI) throttles trying to acquire a loco, or all (GUI) throttles associated with the address?

pabender, 2017-08-21:
Sent from my iPhone

> On Aug 21, 2017, at 4:55 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> @pabender,
> In this context, we only need to notify a "steal" to the throttle which is requesting the throttle, not "all active listeners". We don't want to tell all (GUI)throttles that someone is trying to steal their loco, but we do want to notify the throttle trying to acquire a loco that a "steal" operation would be needed, and give him/her the ability to choose.
> 
> Does "all active listeners" in this context only include those (GUI) throttles trying to acquire a loco, or all (GUI) throttles associated with the address?

All active throttle listeners should be the throttles that have an outstanding request for a throttle associated with the specified address.

Remember that we only create one throttle object.  If  the throttle already exists we return that immediately.  If the throttle has already been requested, we are added to the list waiting for that throttle.

Paul

devel-bobm, 2017-08-22:
I have been able to get basic end-to-end functionality of the LocoNet "steal" process if I hack-in the following changes in `AbstractThrottleManager`:

- make the declaration of field `throttleListeners` "protected" instead of "private"
- make the declaration of class `WaitingThrottle` "protected instead of "private"
- make the declaration of method `getListener()` "public" within class `WaitingThrottle`

I don't believe that this is a good approach, though.

pabender, 2017-08-22:
Sent from my iPad

> On Aug 22, 2017, at 7:42 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> I have been able to get basic end-to-end functionality of the LocoNet "steal" process if I hack-in the following changes in AbstractThrottleManager:
> 
> make the declaration of field throttleListeners "protected" instead of "private"
> make the declaration of class WaitingThrottle "protected instead of "private"
> make the declaration of method getListener() "public" within class WaitingThrottle
> I don't believe that this is a good approach, though.
> 

I agree, it certainly doesn't follow the existing approach.

Make a new method (notifyStealRequired) that mimics how the failedThrottleRequest and notifyThrottleKnown in  AbstractThrottleManager notify the throttles of the failure to acquire and successful acquisition ( respectively ).  The new method will notify the throttle listeners of the steal request, and then the listeners can answer as they need to.

That isolates access to the listeners inside the AbstractThrottleManager class, and you can revert the changes you have made to the data structure.

Paul

devel-bobm, 2017-08-22:
I have made modifications to AbstractThrottleManager per @pabender's recommendations and backed out my hacks.

I have not yet submitted a pull-request to the JMRI repository.  Adding LocoNet "steal" functionalty to JMRI has repercussions that cast a wide net, and I haven't had a chance to think them thru.

The branch [https://github.com/devel-bobm/JMRI/tree/dbm-throttleAcquireRework](url) in my personal repository now contains this updated LocoNet and AbstractThrottleManager functionality to support the LocoNet throttle "steal" functionality.  

It can be exercised using JMRI's Throttle tools.  With a LocoNet Throttle attached to a command station, acquire a loco.  In JMRI, open a new throttle.  Select the same loco address as on the throttle, and select "Set".  JMRI will return a modal dialog box asking "Throttle #### is in use by another device.  Steal?"  "Yes" and "No" buttons are available.  

A user reply of "Yes" causes the JMRI to complete a "steal" operation on LocoNet.  "No" causes JMRI to abort the acquisition.

If a user repliess "Yes", and the command station has "expanded slots" enabled, and the other throttle was using an expanded slot to control the address, then that other throttle will get "StealZap"ed and will no longer have control of the address.

@BHoffman351 - do you need anything more than this for your work on WiThrottleServer?

BHoffman351, 2017-08-22:
> On Aug 22, 2017, at 11:10 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> @BHoffman351 <https://github.com/bhoffman351> - do you need anything more than this for your work on WiThrottleServer?

That should be fine, I found other issues that I’m trying to adjust when I began looking into the code.

Regards,
Brett

pabender, 2017-08-23:
@devel-bobm 

There are only two comments I have.  First:

https://github.com/devel-bobm/JMRI/blob/7e54eeae450b98b99a8d9be5cd41aa4f66850633/java/src/jmri/jmrix/AbstractThrottleManager.java#L519

Manager is isn't spelled correctly in the line above

Second:

https://github.com/devel-bobm/JMRI/blob/7e54eeae450b98b99a8d9be5cd41aa4f66850633/java/src/jmri/jmrix/AbstractThrottleManager.java#L525

I don't think this rises to the level of a warn.  If there aren't any throttle listeners, we can assume the throttle request was canceled, hence the steal request should never go forward.

Paul

devel-bobm, 2017-08-23:
@pabender,
 
I've committed changes to my repository based on your comments.  Deleted the line from the first comment - the method no longer has that parameter.  Downgraded to 'log.debug' on the second comment.

pabender, 2017-08-23:
Ok.  Let's get a PR together with the changes so we can get further comment on them.

Paul

devel-bobm, 2017-09-06:
Issue is solved via PR #3979.  Closing this issue.

closed 2017-09-06 by devel-bobm