JMRI/JMRI #2440: CI Tests sometimes timeout

rhwood, 2016-10-30:
Occasionally CI Tests, on both Travis CI and AppVeyor, timeout--this is most likely a sign of a JMRI test that has hung (most likely by presenting a modal dialog that isn't automatically dismissed). (Travis CI kills tests without output after 30 minutes; while AppVeyor does not. Both Travis CI and AppVeyor kill any tests that run longer than 60 minutes regardless of output.)

Since the way we build our tests using ant, any test output is swallowed by ant until ant completes (this allows ant to inspect the output for reporting purposes and so some tests run late can ensure that no error messages were logged).

I propose that we convert our tests the CI servers to use maven to resolve this, since maven is already installed on all CI servers (we use it to bootstrap the correct version of ant already). An alternate suggestion that we use run-test.sh in a script was given, but that does not work on AppVeyor, and tests do not give a consistent output (also, unlike both maven and ant, no overall report is generated).

pabender, 2016-10-30:
On Oct 30, 2016, at 9:15 AM, Randall Wood <notifications@github.com> wrote:
> I propose that we convert our tests the CI servers to use maven to resolve this, since maven is already installed on all CI servers (we use it to bootstrap the correct version of ant already). An [...]
Personally, I'm not really apply with our current Maven setup, particular the requirement that many ( but not all ) of the libraries we use are required to be in the repository twice.  

That has proven to be unreliable and prone to error ( libraries not downloading correctly on the CI engines, people not knowing we need to update libraries in two places, etc ).

Paul

rhwood, 2016-10-31:
> That has proven to be unreliable and prone to error ( libraries not downloading correctly on the CI engines, people not knowing we need to update libraries in two places, etc ).

Concerning the libraries that fail to download -- the only libraries that are failing to download are ones that we include in our own repo. Libraries that are actually published to a public repository by their own developers are not having this issue.

rhwood, 2016-11-08:
#2452 fixes all maven downloads to ensure they are cached on the CI servers (there are triggers that can invalidate the cache, but those are occasional). The only libraries that are duplicated between ant and maven in our lib directory are those that are not published to a maven distribution point (preferably maven central) by their upstream providers or a third party.

rhwood, 2016-11-24:
The timeouts have been reduced enough that I think this can be closed.

closed 2016-11-27 by rhwood