JMRI/JMRI #2449: NXFrameTest may be incorrect

rhwood, 2016-11-01:
At about lines 107-115, jmri.jmrit.logix.NXFrameTest creates a thread but does not wait for it before asserting action that I presume depends on that thread completing:
```
        jmri.util.ThreadingUtil.runOnLayout( ()->{
            try {
                sensor0.setState(Sensor.ACTIVE);
            } catch (jmri.JmriException e) { Assert.fail("Unexpected Exception: "+e); }
        });
        jmri.util.JUnitUtil.releaseThread(this);

        Assert.assertEquals("Halted/Resume message", warrant.getRunningMessage(), 
                Bundle.getMessage("Halted", block.getDisplayName(), "0"));
```
This is the likely cause of a number of false failures on our automated testing infrastructure. Should the runOnLayout call be wrapped in a JUnitUtil.waitFor block?

Also the runtimes() method (lines 181-198) does not appear to behave as documented, defeating the test at line 122. The method and its docs are:
```
    /**
     * works through a list of sensors, activating one, then the next, then 
     * inactivating the first and continuing. Leaves last ACTIVE.
     * @param sensor - active start sensor
     * @return - active end sensor
     * @throws Exception
     */
    private Sensor runtimes(Sensor sensor, String[] sensors) throws Exception {
        flushAWT();
        _sensorMgr.getSensor(sensors[0]).setState(Sensor.ACTIVE);
        for (int i=1; i<sensors.length; i++) {
            flushAWT();
            _sensorMgr.getSensor(sensors[i]).setState(Sensor.ACTIVE);
            flushAWT();           
            _sensorMgr.getSensor(sensors[i-i]).setState(Sensor.INACTIVE);
        }
        return sensor;
    }
```

pabender, 2016-11-01:
Sent from my iPad

On Nov 1, 2016, at 7:04 AM, Randall Wood <notifications@github.com> wrote:
> At about lines 107-115, jmri.jmrit.logix.NXFrameTest creates a thread but does not wait for it before asserting action that I presume depends on that thread completing: [...]

Doesn't that code only perform the action "sensor0.setState(Sensor.ACTIVE);"

I don't see the need for a separate thread to set the sensor0 state to active, even if that 
action happens more than once.
> jmri.util.JUnitUtil.releaseThread(this);
>         Assert.assertEquals("Halted/Resume message", warrant.getRunningMessage(),
>                 Bundle.getMessage("Halted", block.getDisplayName(), "0")); [...]
I agree, it seems like we should be waiting for an event notification.
> Also the runtimes() method (lines 181-198) does not appear to behave as documented, defeating the test at line 122. The method and its docs are:
>
>     /**
>      * works through a list of sensors, [...]
This does at least work through a list of sensors, but it ignores the value of sensor that was 
passed to it  and then returns that sensor.

In general, this whole package has a lot of issues related to testing, including a few methods 
that are virtually untestable.

Paul

bobjacobsen, 2016-11-01:
> Should the runOnLayout call be wrapped in a JUnitUtil.waitFor block?

The `runOnLayout` call makes sure that `setSensor(..)` has completed before returning.  But I expect (don't really understand the details of this code) that setting the Sensor causes lots of follow on actions, perhaps on multiple threads.  A `waitFor` on the Sensor state won't necessarily capture that.

What might help is changing this line:
```
        Assert.assertEquals("Halted/Resume message", warrant.getRunningMessage(), 
                Bundle.getMessage("Halted", block.getDisplayName(), "0"));
```
into a `waitFor(..)` so that the test will give those other threads whatever time they need.

petecressman, 2016-11-02:
The  runtimes() method was broken some time ago.  I will fix this after the doublePropertyChangeEvent PR is merged.  Alltest, Travis, and Appveyer always pass in my environment.  What kind of faults do you see?

rhwood, 2016-11-03:
The failure is intermittent; https://travis-ci.org/JMRI/JMRI/jobs/172545644 is a sample. The failure doesn't show often since most of us rerun commits to ensure errors are not duplicated.

petecressman, 2016-11-06:
The PR #2457 adds FlushAWT calls after sensor state changes in this test.  The test simulates running a warrant and there is a lot of status info that is sent back and forth between the warrant, its engineer and any monitoring windows.  On my machine FlushAWT adds about 150ms delay which, for me, is enough so that this test always passes.  However, if there are any more intermittent failures, please contact me and i will fashion a more dependable wait.

closed 2017-02-09 by rhwood