JMRI/JMRI #2470: Double Slip support

bobjacobsen, 2016-11-06:
(from a jmri-developers post by Gerald Wolfson <wolfsonga@gmail.com>)

Here is an interesting set of problems I have come across and could use some help solving:

1.       Need a double slip in JMRI with two turnout controls.
2.       XtrkcadReader needs code for generating the correct sequence for the implemented double slip in JMRI.

Here is a description of what this is all about:
Using Xtrkcad, I have done a large layout (whole basement) for a friend of mine and we have just added a Walthers/Shinohara Double Slip. As we have progressed, I have been running the latest .xtc file through XtrkcadReader  to produce a JMRI Panel file that we can use to control the turnouts on the actual layout.

My friend is running this file on a Raspberry Pi3 with JMRI 4.4 and I was bringing the same file up on my display under Windows 10 with JMRI 4.5.3. One other piece of info, when I upgraded from JMRI 4.4 to 4.5.3, the panels would not load on my computer  until I adjusted the .xml file by adding “.layoutEditor” in the proper place in the load string in the .xml file. Then each file would load and show correctly and everything matched on both computers.

Now when the double slip was added, the image shows the double slip on his computer after translation but does not have any control of a turnout and when I look at the .xml file, the double  slip has been converted to a level crossing instead of being a turnout. (I suspect this is happening in XtrkcadReeader as there are no double slips in JMRI.)

In addition, when I try to display the .xml file with the double slip under JMRI 4.5.3, the area where the double slip should be is empty and all of the end points of the surrounding tracks have lines going from the empty ends to the upper left corner of the display. I wonder if the library that was moved with the “.layoutEditor” added to the path is not the only library needed or something got changed in that library and level crossings are not being handled correctly in the moved version since it draws correctly under JMRI 4.4.

Let me know if there is something I missed or how these two problems can be  fixed.

geowar1, 2017-01-21:
Can you create a minimal Xtrkcad layout that reproduces this issue (with single & double slips connected to other (non-slip) tracks); use XtrkcadReader to produce a JMRI Panel file and then post that here? If so I can take a look.

geowar1, 2017-04-27:
IIRC:This should have been implemented in 4.7.2 (.3?). Can you regress?

geowar1, 2017-05-08:
This can be assigned to me and closed… Should all have been fixed via <https://github.com/JMRI/JMRI/pull/3449>

closed 2017-05-08 by bobjacobsen