JMRI/JMRI #2471: Add Check for Code Version to Decoder Definition XML

bobjacobsen, 2016-11-06:
People still want to update their decoder definition files.

To help reduce the complexity of that, we can add a check to the "remake decoder index" process that will warn the user if an XML file (including a fragment file) is tagged as needing more recent code.

- [ ] Add a `requires` element to the `decoder` (for top level), `variables`, `enumVal` and `pane` elements with major, minor, test sub-elements for semantic versioning (c.f. `jmri.Version`) to the schema
- [ ] Update `jmri.jmrit.decoderdefn.DecoderIndexFile` to first check by looping over all contained `requires` elements and checking their contents against `jmri.Version`.  For now, it's OK to directly pop a dialog warning about the mismatch between the requested and present versions, but it would be helpful to put that in a method that can be modified by subclassing later.
- [ ] Add some CI tests of this.

rhwood, 2016-11-07:
For the second bullet should the mismatch test be "requested version is newer than present version" or should it be "requested version is not equal to present version"?

bobjacobsen, 2016-11-07:
Since this is about "decoder definition relies on code feature added in N.N.N", I think the mismatch condition would be that `jmri.Version.compareCanonicalVersions(valueFromXmlFile) < 0`. 

But perhaps I've misunderstood the question.

Some really new features would be caught by the schema check:  If it's really new content that requires new schema contents (like this), it won't even get to this point.  But (1) a schema mismatch message isn't very helpful and (2) many of recent changes have been code that does interpretation within existing schema.

rhwood, 2016-11-07:
> Since this is about "decoder definition relies on code feature added in N.N.N", I think the mismatch condition would be that jmri.Version.compareCanonicalVersions(valueFromXmlFile) < 0.

> But perhaps I've misunderstood the question.

That's what I was asking.

Do we also want to support the ability to specify a "this decoder only works with JMRI between version x and y" as well as "this decoder only works with JMRI newer than x"? (perhaps as "requires=x.y.z-a.b.c" as well as "requires=x.y.z")

(Also should requires be an attribute or an element)?

closed 2021-04-14 by stale[bot]