JMRI/JMRI #2481: DecoderPro Does not load

pchristilaw, 2016-11-09:
Hello,

JMRI 4.2.1 or 4.4 will not load/start. Java 8 - 111, Windows 7Pro 64
I see the very 1st splash then nothing.

JMRI was installed on this machine for a year or more.  About 3 days ago it had a driver issue and stopped.  Uninstalled JMRI, reboot.  Reinstalled Java. Reboot.  Installed JMRI 4.4.  Does not load.
Uninstalled 4.4, reboot,  and installed 4.2.1.  Does not load.

Any pointers would be appreciated.

Cheers Paul

pchristilaw, 2016-11-09:
To add.

when I run decoderpro you can see the LaunchJMRI.exe start in the task manager processes list then stop and unload.

pchristilaw, 2016-11-09:
Test run
CommandLine: LaunchJMRI  /debug /noisy apps.InstallTest.InstallTest         
AppName: InstallTest
Class: apps.InstallTest.InstallTest
Parameters: 
Noisy: 1
Force32bit: 0
Profile: 
Testing for x64...
Result: 1
Setting x64 registry view...
Setting x86 registry view...
JavaPath: C:\Program Files (x86)\Java\jre1.8.0_111\bin\java.exe
Delete file: C:\Users\OHML\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Remove folder: C:\Users\OHML\AppData\Local\Temp\LaunchJMRI\
Create folder: C:\Users\OHML\AppData\Local\Temp\LaunchJMRI
CopyFiles: C:\Program Files (x86)\Java\jre1.8.0_111\bin\java.exe C:\Users\OHML\AppData\Local\Temp\LaunchJMRI\InstallTest.exe
Result: 0
JExePath: C:\Users\OHML\AppData\Local\Temp\LaunchJMRI\InstallTest.exe
PhysicalMemory: 8092m
Check if 4046m is between 192m and 768m
Greater than HIMAXMEM: 768m
MinMemory: 96m
MaxMemory: 768m
Checking user.home...
Reading Desktop Shell Folder registry key...
...read: C:\Users\OHML\Desktop
Checking if equal to %USERPROFILE%...
Comparing: C:\Users\OHML
to: C:\Users\OHML
user.home OK
Options:  -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx768m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\OHML\JMRI\systemfiles" -Djmri.log.path="C:\Users\OHML\JMRI\systemfiles\\"
ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.0.6.jar;lib\jackson-core-2.0.6.jar;lib\jackson-databind-2.0.6.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.2.7.v20150116.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.2.7.v20150116.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.2.7.v20150116.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.2.7.v20150116.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.2.7.v20150116.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.2.7.v20150116.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath.jar;lib\websocket-api-9.2.7.v20150116.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.2.7.v20150116.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.2.7.v20150116.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.2.7.v20150116.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.2.7.v20150116.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbee-api-0.9.jar;lib\xercesImpl.jar
MaxLen: 8192
ExeString: "C:\Users\OHML\AppData\Local\Temp\LaunchJMRI\InstallTest.exe"  -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx768m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\OHML\JMRI\systemfiles" -Djmri.log.path="C:\Users\OHML\JMRI\systemfiles\\" -Djava.class.path=".;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.0.6.jar;lib\jackson-core-2.0.6.jar;lib\jackson-databind-2.0.6.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.2.7.v20150116.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.2.7.v20150116.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.2.7.v20150116.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.2.7.v20150116.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.2.7.v20150116.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.2.7.v20150116.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath.jar;lib\websocket-api-9.2.7.v20150116.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.2.7.v20150116.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.2.7.v20150116.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.2.7.v20150116.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.2.7.v20150116.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbee-api-0.9.jar;lib\xercesImpl.jar" apps.InstallTest.InstallTest 
[... cut ...] v20160517.jar;lib\websocket-common-9.2.7.v20150116.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.2.7.v20150116.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.2.7.v20150116.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbee-api-0.9.jar;lib\xercesImpl.jar" apps.InstallTest.InstallTest 
Return code from process: -1
To copy this text to the clipboard, right click then choose
  'Copy Details To Clipboard'
Completed

rhwood, 2016-11-15:
Can you provide a copy of the file "session.log" (I think it's in the folder C:\Users\OHML\JMRI\systemfiles) right after running PanelPro or DecoderPro?

pchristilaw, 2016-11-15:
Hi Randall

Here you go.

Started JMRI, saw the splash screen disappear, copied log file.

Paul S. Christilaw
1014 Trailview Dr.
Oakville, Ontario, CANADA
L6M 3K9
905-469-0797

rhwood, 2016-11-15:
@pchristilaw Can you try attaching the log file again (at the GitHub website for this issue) or cut and paste it into the reply? I apologize for not specifying how to attach the file earlier.

pchristilaw, 2016-11-15:
No Problem

Here you are:

2016-11-15 07:38:08,403 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-11-15 07:38:08,508 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\OHML\JMRI\log\messages.log [main]
2016-11-15 07:38:08,509 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\OHML\JMRI\log\session.log [main]
2016-11-15 07:38:08,515 apps.AppsBase                         INFO  - DecoderPro version 4.2.1-R6b484b4 starts under Java 1.8.0_111 at Tue Nov 15 07:38:08 EST 2016 [main]
2016-11-15 07:38:08,825 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f2f2172 [main]
2016-11-15 07:38:08,990 jmrix.ConnectionConfigManager         INFO  - No connections configured. [main]
2016-11-15 07:38:09,177 node.NodeIdentity                     INFO  - Using jmri-E46F1390C57F-3f2f2172 as the JMRI Node identity [main]
2016-11-15 07:38:09,658 managers.DefaultShutDownManager       INFO  - Normal termination complete [Thread-3]

Paul S. Christilaw
1014 Trailview Dr.
Oakville, Ontario, CANADA
L6M 3K9
905-469-0797

mattharris, 2016-11-15:
@pchristilaw Could you also provide a copy of the file `ProfileConfig.xml` from the directory `C:\Users\OHML\JMRI\My_JMRI_Railroad`.

Thanks.

rhwood, 2016-11-15:
There's something wrong with the configuration profile currently in use (its triggering an immediate shutdown). I'm going to suggest that we move the existing configuration profile to force a new one to be generated.

Please move the folder C:\Users\OHML\JMRI\My_JMRI_Railroad to your desktop and rename it to something like BAD_JMRI_Railroad (We may want to look at it later).

When you start JMRI, it should attempt to create a new profile automatically.

pchristilaw, 2016-11-15:
Here are the contents of the file

<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/panelfile-2-9-6.xsl" type="text/xsl"?>
<layout-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/layout-2-9-6.xsd">
  <jmriversion>
    <major>3</major>
    <minor>10</minor>
    <test>1</test>
    <modifier>.1</modifier>
  </jmriversion>
  <gui LAFclass="Metal" class="apps.configurexml.GuiLafConfigPaneXml" LocaleLanguage="en" LocaleCountry="CA" LocaleVariant="" nonStandardMouseEvent="no" />
  <connection userName="LocoNet" systemPrefix="L" manufacturer="Digitrax" disabled="no" port="COM1" speed="fixed at 16600 baud" class="jmri.jmrix.loconet.ms100.configurexml.ConnectionConfigXml">
    <options>
      <option>
        <name>CommandStation</name>
        <value>DCS100 (Chief)</value>
      </option>
      <option>
        <name>TurnoutHandle</name>
        <value>Normal</value>
      </option>
    </options>
  </connection>
  <fileLocations class="apps.configurexml.FileLocationPaneXml">
    <fileLocation defaultScriptLocation="program:jython/" />
    <fileLocation defaultUserLocation="F:/Plant Automation Files/JMRI Roster Files/" />
  </fileLocations>
  <console class="apps.configurexml.SystemConsoleConfigPanelXml" scheme="0" fontfamily="Monospaced" fontsize="12" fontstyle="0" wrapstyle="2" />
  <programmer verifyBeforeWrite="no" showEmptyPanes="no" class="jmri.jmrit.symbolicprog.configurexml.ProgrammerConfigPaneXml" />
  <roster class="jmri.jmrit.roster.configurexml.RosterConfigPaneXml" ownerDefault="OHML" />
  <managerdefaults class="jmri.managers.configurexml.ManagerDefaultSelectorXml">
    <setting>
      <key>jmri.CommandStation</key>
      <value>LocoNet</value>
    </setting>
    <setting>
      <key>jmri.ThrottleManager</key>
      <value>LocoNet</value>
    </setting>
    <setting>
      <key>jmri.ProgrammerManager</key>
      <value>LocoNet</value>
    </setting>
    <setting>
      <key>jmri.PowerManager</key>
      <value>LocoNet</value>
    </setting>
  </managerdefaults>
  <perform name="jmri.jmrit.withrottle.WiThrottleCreationAction" type="Action" class="apps.configurexml.PerformActionModelXml" />
  
</layout-config>

Paul S. Christilaw
1014 Trailview Dr.
Oakville, Ontario, CANADA
L6M 3K9
905-469-0797

pchristilaw, 2016-11-15:
Hi

A process has the DIR locked.
Is it a JMRI process, a Java process, or something else?

Paul S. Christilaw
1014 Trailview Dr.
Oakville, Ontario, CANADA
L6M 3K9
905-469-0797

rhwood, 2016-11-15:
Most likely it's a left over JMRI process that is locking that directory. It might be easiest to restart the computer to clear that lock and move the directory before starting JMRI.

Are you using an MS100 to connect to your command station?

pchristilaw, 2016-11-15:
Hi

I am no at the machine now so will do this tomorrow.

Yes – using MS100 to connect.

Paul S. Christilaw
1014 Trailview Dr.
Oakville, Ontario, CANADA
L6M 3K9
905-469-0797

rhwood, 2016-11-15:
The [MS100](http://jmri.org/help/en/html/hardware/loconet/MS100.shtml) is not supported on Windows 7 with JMRI. You will want to replace that before attempting to get JMRI working with your command station.

I highly recommend the [LocoBuffer-USB](http://jmri.org/help/en/html/hardware/loconet/LocoBufferUSB.shtml) from [RR-Cirkits](http://www.rr-cirkits.com/) as a replacement for the MS100.

pchristilaw, 2016-11-15:
Thanks.

Will do this tomorrow and let you know.

Cheers.

Paul S. Christilaw
1014 Trailview Dr.
Oakville, Ontario, CANADA
L6M 3K9
905-469-0797

rhwood, 2016-11-24:
@pchristilaw Have you gotten JMRI working with your railroad?

pchristilaw, 2016-11-24:
Hi Randall

All seems good at the moment.
Thanks for the follow up.

Cheers

Paul S. Christilaw
1014 Trailview Dr.
Oakville, Ontario, CANADA
L6M 3K9
905-469-0797

closed 2016-11-24 by rhwood