JMRI/JMRI #2489: Creating TabbedPreferences for a test

pabender, 2016-11-10:
I'm working on constructor tests for the swing package in the ecos subsystem.

As part of construction of the objects, every one of the tests eventually constructs a jmri.jmrix.ecos.EcosPreferences object, which in turn registers a preferences panel 
with an apps.gui3.TabbedPreferences object that is retrieved from the instance manager.

The problem with this is that during the testing there isn't a registered TabbedPreferences object, so we get back a null pointer at this point during the object construction.

So there really are two questions:
1) Should we actually be registering preferences in this way from a jmri.jmrix object? (that seems like a layering violation to me)
2) Do we have any existing utility method to construct a TabbedPreferences object for the tests? (I couldn't find one).

Paul

rhwood, 2016-11-10:
The EcosPreferences class is IIRC unique in how it registers the ECOS PreferencesPanel, and has done that since sometime in 2010 (the code has been updated since then, but the same model (EcoPreferences registers a Panel in the Preferences window on creation) was introduced then. I'm not aware of anything else that ever registered itself this way.

We probably should not be registering preferences this way, but I'm not sure how else to register connection-specific PreferencesPanels any other way either.

pabender, 2016-11-11:
Thanks @rhwood 

I think I have an alternative solution to loading this that should work:
1) Instead of registering the ecos PreferencesPane object with the TabbedPreferences directly, store it in the Instance Manager.

2) Modify the TabbedPreferences so that in addition to asking the ServiceLoader to provide a list of classes to add to the tabbed preferences, we also ask the instance manager for a list of TabbedPreferences.

I've already coded that, and as long as the list from the InstanceManager is checked first, then everything looks like it does with the current code in JMRI/JMRI master.

I'm going to go ahead and push up a PR with this code change in it.

pabender, 2016-11-11:
The change noted above was implemented in #2493

closed 2016-11-11 by pabender