JMRI/JMRI #2506: Cause CI Tests to fail if Test Coverage drops

pabender, 2016-11-14:
Hello All,

We hit a bit of a milestone with our testing today.  Jacoco on Jenkins is now reporting 50% class coverage.  When we started running the Jacoco builds back in early summer, we were getting somewhere around 25-30% coverage (I can't go back in the history that far) so this is a huge improvement.

(Build 391 actually shows we've covered 3408 classes and have missed 3439, so this is a rounded 50%, and we need to have another 16 classes covered to hit the actual mark).

Moving forward, I want to make sure we don't slip on our coverage numbers.  The Jacoco ant plugin actually supports failing a build if the coverage numbers don't hit a certain threshold, but we haven't been using that feature.  I'm in the process of modifying the build scripts to turn that feature on in at least one of the CI tests.  This will let us know before merging code if we slip.

and just FYI, here's what the output looks like (for a check of the coverage ratio):
alltest-coveragecheck:
[jacoco:report] Loading execution data file /home/Paul/JMRI/git/JMRI/jacoco.exec
[jacoco:report] Writing bundle 'AntTestReporting' with 6847 classes
[jacoco:report] Rule violated for bundle AntTestReporting: classes covered ratio is 0.49, but expected minimum is 0.50

Paul

rhwood, 2016-11-14:
I get the feeling that unless we:
- are really reviewing the tests to make sure they are real,
- ensure tests work in AllTest as well as individually,
- and require that a PR cannot be merged unless it passes
that this is only going to lead to people ignoring the CI services entirely before merging commits (do note that we've had people with commit privileges state that our "code wins" rule means that ignoring the PR process and CI services is entirely legitimate).

pabender, 2016-11-14:
@rhwood,

The biggest thing we need to make sure is that people understand that tests are important, and not adding/changing tests when you add/change code isn't right, so your 3rd point is valid.  I think most of us are to the point where we don't merge PRs until the tests pass.

On your first two points, I personally believe that even a test without any assertions provides some value if it exercises a part of the code (At least with JUnit4 tests, it tells us the part of the code under test doesn't throw any exceptions).

Running tests individually is important, because it shows there isn't a dependency between test classes, but even dependent tests can provide value.  (Again, the exercise some part of the code...)

The coverage reports tell us not only what classes we hit, but also other information about the lines we hit in those classes.  Just like findbugs and other tools we use, this gives us a picture of where we need to improve things.

The eventual goal is to set the coverage rules so we can tell we have good coverage of the program in those tests.  Our current test coverage is still pretty poor (We hit 50% of classes, but only 35% of methods... and an even small percentage of lines).

Paul

KenC57, 2016-11-14:
Paul,

 

One thing that will help with this goal of improving the test coverage is
educating some of us on how and what should be written for the tests. Your
work seems to be doing one important part of providing the scaffolding so us
who don't have time to do a lot of programming can follow along with simple
stuff.

 

So how up to date is:

http://jmri.org/help/en/html/doc/Technical/JUnit.shtml

with the work you have been doing?? I suspect it has grown along the way and
a quick look at it I wonder if the organization of it is all valid or if it
might need some reshuffling to improve the reading.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-11-14:
I really do value the CI tests and associated infrastructure.  A lot of work has gone into making the code more robust. Multiple people have worked hard on this. Having more robust, maintainable code is important in the long run.

At the same time, the rate at which new people contribute has gone way down. Adding new people to the developer team is also important in the long run, probably more important.

I think it's reasonable to hold seasoned JMRI developers to a high standard on tests and other aspects of code quality.  

I'm not at all sure it's wise to do that for everybody. In particular, I think it's counterproductive to _require_ somebody new who contributes a small fix or addition (new code) to provide tests for it. Better to get the fix or addition, make that experience rewarding for the new contributor, and work from there. 

So having CI count the coverage of classes (generally, first contributions aren't entire classes, but it has happened) seems OK to me.  Even raising a fail when it drops below a threshold is fine. I don't think that'll happen very often.  But I'd rather not get a lot tighter than that.

pabender, 2016-11-14:
On Mon, Nov 14, 2016 at 9:17 AM, Ken Cameron <notifications@github.com>
wrote:

> One thing that will help with this goal of improving the test coverage is
> educating some of us on how and what should be written for the tests. Your
> work seems to be doing one important part of [...]

There are several philosophies to testing, but I've come around to some
form of "test driven development" is beneficial.  The experience of
removing the ActiveFlag classes is only the most recent example of that

I'm not yet in the camp that prescribes the development cycle:
1) Write a Test describing desired result (or behavior, or use case)
2) See where the test fails
3) Write code to make the pass the test
4) re-run the test to make sure there is no failure
5) refactor
6) repeat.

That said, all of the Agile software development methods stress testing as
an important piece of the puzzle.

So how up to date is:
>
> http://jmri.org/help/en/html/doc/Technical/JUnit.shtml
>
> with the work you have been doing?? I suspect it has grown along the way
> and
> a quick look at it I wonder if the organization of it is all valid or if it
> might need some reshuffling to improve the reading.
>

It is fairly up to date.  There are a few sections on adding new class and
package tests that haven't ever been fleshed out.  I'll work on that over
the next couple of days.

Paul

pabender, 2016-11-14:
On Mon, Nov 14, 2016 at 9:33 AM, Bob Jacobsen <notifications@github.com>
wrote:

> I'm not at all sure it's wise to do that for everybody. In particular, I
> think it's counterproductive to *require* somebody new who contributes a
> small fix or addition (new code) to provide tests [...]

What I'm actually looking at is requiring (once I get through all of the
classes that are currently at 0% class coverage) that each package has at
least one test and that the overall testing doesn't fall below the current
threshold.

That would mean that new developers only need to worry about tests if they
are adding a package.

The plan will be to increase the thresholds as time progresses, so we can
make sure we don't slide backwards significantly.  In the new ant targets,
I set thresholds of 50% coverage for classes coverage and 28% for line
coverage (which are the current values reported by the Jacoco plugin on
Jenkins).  We can move those up when we hit the next 10% mark for each, and
look at adding the other statistics later.  Eventually I would like to get
us to thresholds of 90 or 95%, but I expect that to take a while.

I can't put the rule in about the number of tests in a package yet because
that will fail right now.

Paul

pabender, 2016-11-15:
PR #2523 gets the necessary infrastructure in place to handle the tests on travis (only on the HEADLESS=false builds)

Paul

pabender, 2016-11-15:
closed via 2523

closed 2016-11-15 by pabender