JMRI/JMRI #2508: Create test release 4.5.7

bobjacobsen, 2016-11-14:
For discussion of timing and what to include in 4.5.7.

This is (intended to be) the last "from HEAD of master" test release in the 4.5/4.6 cycle this Fall.

After this, there will be a short-cycle of "targeted changes only" test releases as needed.

mattharris, 2016-11-28:
The build process for this release is to be started later on today (around 18:00 UTC).

As a reminder, this is to be the last "from HEAD of master" test release this cycle - further changes will be part of the 4.7.x/4.8 release cycle.

rhwood, 2016-11-28:
Does this mean that neither #1382 and #1472, both IIRC originally requested by Windows users will not be addressed for Windows users in 4.6?

mattharris, 2016-11-28:
@rhwood I'll take a look at those items beforehand.

mattharris, 2016-11-28:
#2610 covers these updates.

Build process start for this release is delayed - might now not be until tomorrow.

mattharris, 2016-11-29:
The release-4.5.7 branch has been created. 

From now on, please document your changes in the [jmri4.7.1.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.7.1.shtml) release note file.

Maintainers, please set the 4.7.1 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.7/)

mattharris, 2016-11-29:
First JMRI 4.5.7 files are available in the usual way at:

http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.7

Feedback appreciated. I would like to release this later today or tomorrow morning.

mattharris, 2016-11-29:
A new build has been launched to (hopefully) fix the launching issues seen with certain Linux and Windows configurations.

Build 2 or later on the Jenkins job will contain the fix.

mattharris, 2016-11-30:
Closing as now released

closed 2016-11-30 by mattharris