JMRI/JMRI #2510: Cannot Add Loco Marker in Layout Editor

spassmo, 2016-11-14:
While viewing a Layout Editor panel and selecting "Add Loco" from "Marker" menu, a label no longer appears on the screen.   If I look in the "Window" menu I do see an "Enter Loco Marker" selection which when chosen shows a the top bar of a window.  If I maximize the window I do see a field to enter a Loco ID but there is no "OK" type button to save/close the window.

The "Add Loco from roster" and "Remove labels" tools in the "Marker" menu do appear to work as expected.

I first noticed this problem in 4.5.4 and it is happening in 4.5.5 and 4.5.6 as well.

I get the following in the JMRI System Console when selecting the "Add Loco" tool

`2016-10-29 00:18:54,300 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.util.MissingResourceException: Resource not found
        at jmri.Bundle.retry(Bundle.java:191)
        at jmri.Bundle.handleGetMessage(Bundle.java:150)
        at jmri.jmrit.Bundle.retry(Bundle.java:106)
        at jmri.Bundle.handleGetMessage(Bundle.java:150)
        at jmri.jmrit.display.Bundle.retry(Bundle.java:94)
        at jmri.Bundle.handleGetMessage(Bundle.java:150)
        at jmri.Bundle.handleGetMessage(Bundle.java:130)
        at jmri.jmrit.display.Bundle.getMessage(Bundle.java:41)
        at jmri.jmrit.display.Editor.locoMarkerFromInput(Editor.java:1509)
        at jmri.jmrit.display.layoutEditor.LayoutEditor.access$57(LayoutEditor.java:1)
        at jmri.jmrit.display.layoutEditor.LayoutEditor$57.actionPerformed(LayoutEditor.java:1756)
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
        at javax.swing.AbstractButton.doClick(Unknown Source)
        at javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
        at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
        at java.awt.Component.processMouseEvent(Unknown Source)
        at javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.awt.Component.processEvent(Unknown Source)
        at java.awt.Container.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)`

bobjacobsen, 2016-11-14:
```
jmri.jmrit.display.Editor.locoMarkerFromInput(Editor.java:1509) at 
```

The "OK" key is missing from `java/src//jmri/jmrit/display/DisplayBundle.properties` (it's present in `java/src//jmri/jmrit/display/DisplayBundle_cs.properties` but none of the other translations) Testing a fix, will create PR shortly.

bobjacobsen, 2016-11-14:
Problem seems to have arrived in [Clean up LE and CPE Bundles](https://github.com/JMRI/JMRI/commit/6b0ece94781541c41c55c8716b1050053f2d64ef) on August 12th.  `jmrit.Editor` is one of those Really Big Classes that doesn't have a lot of test coverage, so CI didn't detect this.

bobjacobsen, 2016-11-14:
@spassmo: Thanks for the trace, that helped a lot.  There's a (likely) fix in #2510, which should be in the development builds at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/ later today.  Will send a note when that's done.

closed 2016-11-14 by bobjacobsen

bobjacobsen, 2016-11-14:
Fix is now in dev builds, will be in 4.5.7

spassmo, 2016-11-15:
@bobjacobsen Thank you,  It's working well now.

PR #2515 (2016-11-14): Fix layout editor properties file (Issue 2510)
Adds the "OK" string to the base properties file, removing it from the _cs file, to fix #2510.