JMRI/JMRI #2511: Invalid UTF-8 not caught

rhwood, 2016-11-14:
The test ```apps.ValidateConfigFilesTest``` fails when run individually (as ```./runtest.csh apps.ValidateConfigFilesTest``` or ```ant test-single -Dtest.includes=apps.ValidateConfigFilesTest```) but not when run as part within ```ant alltest``` or ```ant headlesstest```.

This is since #2484 was merged, which introduces invalid, non-UTF-8 characters into some config files, but which isn't being caught by any CI servers (that is a separate issue).

I can't figure out why this test is failing to catch invalid files when run in AllTest or HeadlessTest.

mattharris, 2016-11-14:
Oh poop. Will have to fix those invalid characters.

mattharris, 2016-11-14:
OK, I know (or, at least suspect) the cause of the UTF-8 error - it's likely the same as that with the `decoderIndex.xml` that I was battling with in #2484. It looks like the `fixcrlf` ant task defaults to other than UTF-8 encoding when processing and, until now, we've gotten away with it.

mattharris, 2016-11-14:
#2513 contains a fix for the broken config file as well as an update to the ant build config file to force UTF-8 on the `fixlineends` task.

rhwood, 2016-11-14:
Even on OS X (where UTF-8 is the OS-wide default, the fixcrlf task passed these files without modification; attempts to the automated tools available to me to fix this led to truncated files (that is I think the task did not force a file encoding change)).

I'm still more concerned about why this wasn't caught in the CI or Jenkins tests than that it happened.

@mattharris Thanks for the quick fix.

mattharris, 2016-11-14:
@rhwood Indeed, I'm more concerned that it was somehow OK in the tests.

Anyhow, what seems to have happened is that, during the merge, the two versions of the file were encoded differently and the 'wrong' encoding was taken over. The commit cdc9e5228a8186bc1de38334bf41a6fef7e751d0 as part of #2484 that contained the Czech translation switched the file encoding from, for want of a better term, ASCII, to UTF-8.
The commit 3e4f16511976b571736a203770bcde59e1b572ab from #1804 that contained Catalan translations kept the file encoding as-is (i.e. not UTF-8).
As the cdc9e5228a8186bc1de38334bf41a6fef7e751d0 commit was actually dated _earlier_ than the 3e4f16511976b571736a203770bcde59e1b572ab commit (despite being in a _later_ PR), the encoding from PR #1804 seems to have won in the merge.

rhwood, 2018-03-07:
Likely fixed.

closed 2018-03-07 by rhwood