JMRI/JMRI #2524: serialdriver.SerialDriverAdapter Error in System console

spassmo, 2016-11-15:
I'm seeing the following errors in the System Console when a C/MRI node is configured with a serial connection,  Errors do not appear when a Simulated C/MRI connection is configured.

2016-11-14 23:44:15,369 serialdriver.SerialDriverAdapter      ERROR - getInputStream called before load(), stream not available [main]
2016-11-14 23:44:15,369 serialdriver.SerialDriverAdapter      ERROR - getOutputStream called before load(), stream not available [main]
2016-11-14 23:44:15,369 jmrix.AbstractMRTrafficController     ERROR - Failed to start up communications. Error was java.lang.NullPointerException [main]

bobjacobsen, 2016-11-15:
This is usually a configuration error and/or problem with the port hardware or driver.  Do you have another connection _besides_ C/MRI configured?  E.g. does your preferences show two tabs under "Connections"?

rhwood, 2016-12-17:
Closing since OP has not responded.

closed 2016-12-17 by rhwood

spassmo, 2017-01-29:
Sorry I missed the question on this one.   No there is only one connection tab.
#2953