JMRI/JMRI #2529: Unexpected failure running test jmri.jmrit.PackageTest

rhwood, 2016-11-15:
I found this failure does not occur when AllTest or HeadlessTest are run, but when run individually, or within the jmri.jmrit.PackageTest, I get the following errors:
```
Calculus:RunTests rhwood$ ant realclean tests ; ./runtest.csh -Djava.awt.headless=true jmri.jmrit.display.configurexml.LoadAndStoreTest
.log4j:WARN No appenders could be found for logger (jmri.configurexml.LoadAndStoreTest).
log4j:WARN Please initialize the log4j system properly.
log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
.E.E..
Time: 0.961
There were 2 errors:
1) Test load&store&compare matches: java/test/jmri/jmrit/display/configurexml/load/OneOfEach.3.3.3.xml(jmri.configurexml.LoadAndStoreTestBase$CheckOneFilePasses)org.junit.ComparisonFailure: expected:<    [  <operation name="Raw" class="jmri.configurexml.turnoutoperations.RawTurnoutOperationXml" interval="300" maxtries="1" /]>> but was:<    [</operations]>>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at jmri.configurexml.LoadAndStoreTestBase.checkFile(LoadAndStoreTestBase.java:99)
	at jmri.configurexml.LoadAndStoreTestBase.loadLoadStoreFileCheck(LoadAndStoreTestBase.java:159)
	at jmri.configurexml.LoadAndStoreTestBase$CheckOneFilePasses.runTest(LoadAndStoreTestBase.java:175)
	at jmri.jmrit.display.configurexml.LoadAndStoreTest.main(LoadAndStoreTest.java:32)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)
2) Test load&store&compare matches: java/test/jmri/jmrit/display/configurexml/load/OneOfEach.xml(jmri.configurexml.LoadAndStoreTestBase$CheckOneFilePasses)org.junit.ComparisonFailure: expected:<    [  <operation name="Raw" class="jmri.configurexml.turnoutoperations.RawTurnoutOperationXml" interval="300" maxtries="1" /]>> but was:<    [</operations]>>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at jmri.configurexml.LoadAndStoreTestBase.checkFile(LoadAndStoreTestBase.java:99)
	at jmri.configurexml.LoadAndStoreTestBase.loadLoadStoreFileCheck(LoadAndStoreTestBase.java:159)
	at jmri.configurexml.LoadAndStoreTestBase$CheckOneFilePasses.runTest(LoadAndStoreTestBase.java:175)
	at jmri.jmrit.display.configurexml.LoadAndStoreTest.main(LoadAndStoreTest.java:32)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)

FAILURES!!!
Tests run: 5,  Failures: 0,  Errors: 2

Calculus:RunTests rhwood$
```

bobjacobsen, 2016-11-15:
Looks like a missing manager when run stand-alone, will check.

bobjacobsen, 2017-01-21:
Both sure where this was fixed, but it's not failing now in the current HEAD of master.

closed 2017-01-21 by bobjacobsen

rhwood, 2017-01-21:
This probably got fixed in #2769