JMRI/JMRI #2534: Edge Connection "Edit Link" not working

tynewydd, 2016-11-17:
I have several panels with Edge connections that have never been connected and when I hit "Edit Link" on them - absolutely nothing happens instead of an expected pulldown of available panels and other connections to attach to.

I tried the simplest use case of two new panels with with just one edge connection each and got the same result. 

I took a peek at the XML and all looked clean. 

Level 4.4 - Mac OSX El Capitan

tynewydd, 2016-11-17:
And actually the other edit tasks "Set Signals ..." and "Set Signal Masks..." on edge connectors also have zero result.  I assume this means that a properties page/pane is not being properly displayed.

mattharris, 2016-11-17:
@tynewydd Could you check if there are any messages output to the system console? (Help > System Console...)

If so, paste a copy of them back here (use `Copy to clipboard` button in console window)

tynewydd, 2016-11-18:
Well that reveals the error message "Could not set link until we have a connecting track with a block assigned".   I suppose that it was necessary to be able to pass trains over the link?  Maybe the documentation should state that (and/or the user should see the error and not just the log)?

Anyway, I added blocks to both sides of the link - and now I do see the pull-downs - great - but I also see lots of sets of "Did not decode a ts.getType2() of 1 (A)" messages however.

pnbgit, 2017-10-30:
This has definitely been a gotcha for a while - which is fine once the user knows about it :-)   A good fix would be for one of George's grayed out menu entries on the right click popup menu - such as "Cannot set link until blocks assigned".

closed 2021-04-14 by stale[bot]