JMRI/JMRI #2535: Trailing Blanks in Turnout Names not suppressed

tynewydd, 2016-11-17:
If you define a turnout in the table with a trailing blank, you can't use that name to link to it in the layout editor. Removing the blank in the table does the trick - but it would be better if the turnout editor suppressed trailing blanks on data entry/update.

bobjacobsen, 2016-12-19:
I haven't been able to recreate this.  Names with trailing blanks, entered consistently, seem to work throughout.  Can you give a bit more specifics of how to recreate the problem?

(I'm not sure it's a good idea to use names with trailing blanks, but either they should be prevented they should work)

geowar1, 2017-01-21:
Just spent a whole lot of time converting all the LayoutEditor(/Turnout/Slip, etc.) to use editable JComboBox's instead of JTextFields to enter Blocks, Turnouts, Signals, Sensors, etc. and all the JTextField code used .getText().trim() (as did the JComboBox code I wrote to replace it). I can check all the (Block, Turnout, Sensor, etc.) table code and add the .trim() call in the appropriate places.

geowar1, 2017-01-21:
Looking at /java/src/jmri/jmrit/beantable/BeanTableDataModel.java:654 (in the renameBean routine):

        String value = _newName.getText().trim();
So it appears that the "Rename" function in all the tables already include the .trim().

Looking at /java/src/jmri/jmrit/beantable/TurnoutTableAction.java:1359:
                String user = userName.getText();
There's no .trim(); should I add it?

geowar1, 2017-02-17:
https://github.com/JMRI/JMRI/pull/3084

closed 2021-04-14 by stale[bot]