JMRI/JMRI #2536: Spurious test output?

bobjacobsen, 2016-11-17:
When I run `ant ci-test` on OS X it has started to produce some extra output:

```
   [junit] ------------- Standard Output ---------------
    [junit] 
    [junit] Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Block Table"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Block Table"" has been opened in 2 milliseconds
    [junit]     jmri.jmrit.beantable.AbstractTableAction$1[frame0,26,24,1875x486,layout=java.awt.BorderLayout,title=Block Table,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,1875x464,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] 
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Signal Head Table"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Signal Head Table"" has been opened in 0 milliseconds
    [junit]     jmri.jmrit.beantable.AbstractTableAction$1[frame1,26,24,880x486,layout=java.awt.BorderLayout,title=Signal Head Table,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,880x464,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Memory Table"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Memory Table"" has been opened in 2 milliseconds
    [junit]     jmri.jmrit.beantable.AbstractTableAction$1[frame2,26,24,847x486,layout=java.awt.BorderLayout,title=Memory Table,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,847x464,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Script Output"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Script Output"" has been opened in 1 milliseconds
    [junit]     jmri.util.JmriJFrame[frame3,26,24,424x269,invalid,layout=java.awt.BorderLayout,title=Script Output,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,424x247,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Display Log Categories"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Display Log Categories"" has been opened in 1 milliseconds
    [junit]     jmri.util.JmriJFrame[frame4,26,23,578x1417,layout=java.awt.BorderLayout,title=Display Log Categories,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,578x1395,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Sensor Group"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Sensor Group"" has been opened in 1 milliseconds
    [junit]     jmri.jmrit.sensorgroup.SensorGroupFrame[frame5,26,24,554x428,invalid,layout=java.awt.BorderLayout,title=Define Sensor Group,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,554x406,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Define Follower"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Define Follower"" has been opened in 0 milliseconds
    [junit]     jmri.jmrit.ussctc.FollowerFrame[frame6,26,24,396x169,layout=java.awt.BorderLayout,title=Define Follower,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,396x147,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] Trace:
    [junit] Start to wait frame "Frame with title "Define OS Indicator"" opened
    [junit] Trace:
    [junit] Frame "Frame with title "Define OS Indicator"" has been opened in 1 milliseconds
    [junit]     jmri.jmrit.ussctc.OsIndicatorFrame[frame7,26,24,396x146,layout=java.awt.BorderLayout,title=Define OS Indicator,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,396x124,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
    [junit] ------------- ---------------- ---------------
```

Some quick grepping doesn't find any related strings in the source code, so I'm not sure where this is coming from.  Any suggestions on the cause and how to suppress it?

rhwood, 2016-11-18:
It's from Jemmy, a GUI test framework that works with JUnit 4 tests.

rhwood, 2016-11-18:
Hit send too soon. This output may help us understand tests that fail if the theory that some tests are off screen on the CI tests, so I don't think it should be suppressed.

bobjacobsen, 2017-01-31:
Can we now remove this now?  It's a large volume, and makes it harder to get straight to the true errors causing failure.

commit 39324f3f (2017-02-01): Silence most outputs from Jemmy. (#2980)

This silences most outputs from the Jemmy GUI testing library while
allowing individual tests to override that behavior. To ensure that
Jemmy remains silences, any call to Log4JFixture.setUp also silences
Jemmy.

Closes #2536
closed 2017-02-01 by rhwood