JMRI/JMRI #2549: Request An Enhancement - Panel Pro - Turnout Icons With Colour Option When Operated

peterriley2, 2016-11-20:
I would like to request an addition of turnout colours in panel pro. The turnout icon displays the closed and open status. I would like the colour option when operated. Im using Ithrottle and turnouts are displayed using green and red for closed and open. I would like to be able to display this on panel pro.

bobjacobsen, 2016-11-20:
You can make up your own graphics files and have the panel icons set to show those.  That way, you can make it look any way you want.

If they're the right size, you can use these colored squares that are distributed with JMRI:  http://jmri.org/resources/icons/

For instructions on how to change what a turnout icon shows, please see http://jmri.org/help/en/package/jmri/jmrit/display/IconAdder.shtml

peterriley2, 2016-12-03:
Hi Bob

thank you for the information, sorry about the delay in replying
I am currently using layout editor and not panel editor. Would you suggest I change over as I'm not currently using block detection etc. I did start with panel editor, but transfered over as I struggled to increase the icon track length using the icons available. Peter

bobjacobsen, 2016-12-04:
> I am currently using layout editor and not panel editor. Would you suggest I change over as I'm not currently using block detection etc. I did start with panel editor, but transfered over as I struggled to increase the icon track length using the icons available. Peter

Layout Editor is really aimed at editing the _layout_ information: How things are connected, where blocks and signals are, etc.  

Panel Editor (and its sibling, Control Panel Editor) are about setting up something that looks exactly the way you want.

On the other hand, Panel Editor deals with individual objects like turnouts and sensors, and doesn’t know much about the overall layout.  And layout editor has very limited ability to change how it looks.

I expect those basic characters will continue, which means it’s unlikely that somebody will add custom graphic capability to Layout Editor.  But we can leave this issue open just in case.

As a workaround, if you have your layout panel the way you like it, take a screen shot of it.  Then use that image as the background for a Panel Editor panel, which you can customize exactly the way you want.

peterriley2, 2016-12-04:
Bob

I now understand a lot more about the different options of panel pro. I have just started to build a selection of panels to operate the layout in panel editor. I have managed to change icons along with producing my own versions in colour Many thanks for the replying. I'm amazed by what all the JMRI team have produced

Peter

closed 2016-12-04 by bobjacobsen