JMRI/JMRI #2565: New B&O-Semaphore signal system

SuzieTall, 2016-11-22:
[B&O-Semaphore test.zip](https://github.com/JMRI/JMRI/files/607323/B.O-Semaphore.test.zip)
[B&O-Semaphore.zip](https://github.com/JMRI/JMRI/files/607322/B.O-Semaphore.zip)

New signal system for inclusion in to JMRI and a test panel with test strategy. Will work in conjunction with B&O-2009 system.

Suzie x

KenC57, 2016-11-26:
Missed seeing this zip. Will deal with it after the dust settles on the other two.

SuzieTall, 2016-11-26:
Hi Ken

Not sure what Bob normally does with the testing bit but B&O-Semaphore.zip is the important bit.

The test zipfile just contains a demonstration panel and a set of manual test cases.

Suzie x

SuzieTall, 2016-11-27:
Sorry, there is an error in the path for the graphics files. Corrections in #2594

https://github.com/JMRI/JMRI/issues/2594

mattharris, 2016-11-28:
@SuzieTall can this one now be closed as, from what I can see, it's all either merged or been updated.

Thanks.

SuzieTall, 2016-11-28:
It can be closed thankyou. Test zipfile is just there in case ongoing testing is implemented but it is not important.

Suzie x

closed 2016-11-29 by KenC57