JMRI/JMRI #2566: Table window size not saved/restored

KenC57, 2016-11-23:
If you open the tables using a button instead of from the menu, it opens full size. But if you open from the menu, it will save and restore using the last size when closed. I've not tested to see if this also happens if the table is opened as a startup action. But I suspect something about either the name or something not being saved right for what the restore is looking for.
-ken c

KenC57, 2016-11-23:
Further testing shows same behavior for 'startup action'.

I now think the issue is that the menu is saving separate window values based on which table. But the other two methods both don't have a 'which table' when it is being selected. So the real question may be if there shouldn't just be one entry for the tables, since they always open as a table within the whole set of tables and not an individual table anyway. Doesn't depend on which 'table' is being opened.
-ken c

geowar1, 2017-01-21:
I added a Preferences -> Start Up -> "Add button to main window" to "Open Turnout Table"… When I click the button the turnout table opens as a separate (individual) window (table).

[image]

[image]

rhwood, 2017-02-12:
This might be fixed by #3052

closed 2017-05-07 by bobjacobsen