JMRI/JMRI #2577: RPS Algorithm Test Failures

rhwood, 2016-11-24:
The following tests throw errors:
```
Calculus:JMRI rhwood$ ./runtest.csh jmri.jmrix.rps.Ash1_0AlgorithmTest
.ERROR - Mismatch: 4 readings, 5 receivers [main] jmri.jmrix.rps.Ash1_0Algorithm.?()
.ERROR - Mismatch: 3 readings, 4 receivers [main] jmri.jmrix.rps.Ash1_0Algorithm.?()
E.ERROR - Mismatch: 3 readings, 4 receivers [main] jmri.jmrix.rps.Ash1_0Algorithm.?()
.ERROR - Mismatch: 3 readings, 4 receivers [main] jmri.jmrix.rps.Ash1_0Algorithm.?()
E
Time: 2.814
There were 2 errors:
1) testCalc4_not1(jmri.jmrix.rps.Ash1_0AlgorithmTest)java.lang.AssertionError: z close expected:<true> but was:<false>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:118)
	at jmri.jmrix.rps.Ash1_0AlgorithmTest.testCalc4_not1(Ash1_0AlgorithmTest.java:60)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.jmrix.rps.Ash1_0AlgorithmTest.main(Ash1_0AlgorithmTest.java:109)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)
2) testCalc4_not3(jmri.jmrix.rps.Ash1_0AlgorithmTest)java.lang.AssertionError: x close expected:<true> but was:<false>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:118)
	at jmri.jmrix.rps.Ash1_0AlgorithmTest.testCalc4_not3(Ash1_0AlgorithmTest.java:96)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.jmrix.rps.Ash1_0AlgorithmTest.main(Ash1_0AlgorithmTest.java:109)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)

FAILURES!!!
Tests run: 4,  Failures: 0,  Errors: 2

Calculus:JMRI rhwood$ ./runtest.csh jmri.jmrix.rps.Ash1_1AlgorithmTest
.ERROR - Mismatch: 4 readings, 5 receivers [main] jmri.jmrix.rps.Ash1_1Algorithm.?()
.ERROR - Mismatch: 3 readings, 4 receivers [main] jmri.jmrix.rps.Ash1_1Algorithm.?()
.ERROR - Mismatch: 3 readings, 4 receivers [main] jmri.jmrix.rps.Ash1_1Algorithm.?()
.ERROR - Mismatch: 3 readings, 4 receivers [main] jmri.jmrix.rps.Ash1_1Algorithm.?()

Time: 6.058

OK (4 tests)

Calculus:JMRI rhwood$ ./runtest.csh jmri.jmrix.rps.Analytic_AAlgorithmTest
....Test 4 x not close -99999.0 3.0
ERROR - Can't solve, iteration limit exceeded.  it = 6 [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
ERROR - failed to converge [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
E.Test 5 x not close -99999.0 3.0
ERROR - Can't solve, iteration limit exceeded.  it = 6 [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
ERROR - failed to converge [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
E.Test 6 x not close -99999.0 3.0
ERROR - Can't solve, iteration limit exceeded.  it = 6 [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
ERROR - failed to converge [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
E.Test 7 x not close -99999.0 3.0
ERROR - Can't solve, iteration limit exceeded.  it = 6 [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
ERROR - failed to converge [main] jmri.jmrix.rps.Analytic_AAlgorithm.?()
E
Time: 2.4
There were 4 errors:
1) testCalc4(jmri.jmrix.rps.Analytic_AAlgorithmTest)java.lang.AssertionError: Test 4 x close expected:<true> but was:<false>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:118)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc(AbstractAlgorithmTest.java:176)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc4(AbstractAlgorithmTest.java:90)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.jmrix.rps.Analytic_AAlgorithmTest.main(Analytic_AAlgorithmTest.java:23)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)
2) testCalc5(jmri.jmrix.rps.Analytic_AAlgorithmTest)java.lang.AssertionError: Test 5 x close expected:<true> but was:<false>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:118)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc(AbstractAlgorithmTest.java:176)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc5(AbstractAlgorithmTest.java:104)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.jmrix.rps.Analytic_AAlgorithmTest.main(Analytic_AAlgorithmTest.java:23)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)
3) testCalc6(jmri.jmrix.rps.Analytic_AAlgorithmTest)java.lang.AssertionError: Test 6 x close expected:<true> but was:<false>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:118)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc(AbstractAlgorithmTest.java:176)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc6(AbstractAlgorithmTest.java:118)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.jmrix.rps.Analytic_AAlgorithmTest.main(Analytic_AAlgorithmTest.java:23)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)
4) testCalc7(jmri.jmrix.rps.Analytic_AAlgorithmTest)java.lang.AssertionError: Test 7 x close expected:<true> but was:<false>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:118)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc(AbstractAlgorithmTest.java:176)
	at jmri.jmrix.rps.AbstractAlgorithmTest.testCalc7(AbstractAlgorithmTest.java:132)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.jmrix.rps.Analytic_AAlgorithmTest.main(Analytic_AAlgorithmTest.java:23)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:21)

FAILURES!!!
Tests run: 7,  Failures: 0,  Errors: 4
```

These tests were commented out of any suites in 2008 (739482aa42c28ec65997a703cd2931a41c3b850f), but I do not know if they were bad then, or removed for some other reason. Should these tests be removed (they aren't in use), and if not, how should they be fixed (I do not understand the code enough to know if the test is failing correctly)?  

(I added logging setup and teardown configurations to these tests in #2576 to get the Mismatch error statements instead of improper logging configuration warnings.)

pabender, 2016-11-24:
On Nov 24, 2016, at 7:01 AM, Randall Wood <notifications@github.com> wrote:

> These tests were commented out of any suites in 2008 (739482a), but I do not know if they were bad then, or removed for some other reason. Should these tests be removed (they aren't in use), and if not, how should they be fixed (I do not understand the code enough to know if the test is failing correctly)?
> 

Bob would be the one to know if they ever worked.

I would:
1) convert the. Test class to JUnit4 ( if they aren't already )
2) mark these tests with an ignore annotation that explains how they fail.

Paul

bobjacobsen, 2016-11-24:
There are only a few RPS installations, and the manufacturer sold the line to another company. What happens next isn't at all clear: The new owner periodically says he'll be bringing out a better version, but it's been quite a while.  I don't know whether there will ever be motivation to update this support.

The computational part (*Algorithm* classes) was developed by a 3rd party in FORTRAN, and then transliterated to Java.  That makes it pretty opaque, hence hard to really test. 

I don't think it's worth figuring out how to update the code and/or test right now.

I think leaving the code, but flagging them to be ignored as known failures is the right approach.

rhwood, 2016-11-24:
Thanks for the direction. Will do so this weekend.

PR #2582 (2016-11-25): Fix commented out test classes
Fix test classes commented out because they failed by ignoring them (this required porting the tests to JUnit 4 first). Fixes #2577.
closed 2016-11-25 by rhwood