JMRI/JMRI #2583: Conditional redefines NamedBean constants inconsistently

rhwood, 2016-11-25:
Most NamedBean subclasses use the constant UNKNOWN with a value of 1, and, if appropriate, use the values 2 and 4 in a state pair (ON/OFF, CLOSED/THROWN, ACTIVE/INACTIVE, etc).

```jmri.Conditional```, however, defines its state pair (TRUE/FALSE) as 1 and 2 and UNKNOWN as 4.

Do we have any sense of how much damage would be caused by making the Conditional consistent with the other NamedBeans?

Within what we ship, there would be no damage, even in the Jython scripts, but the question is how likely is it that people have embedded the values of those constants in scripts instead of references to the constants themselves?

bobjacobsen, 2016-11-25:
> Within what we ship, there would be no damage, even in the Jython scripts, but the question is how likely is it that people have embedded the values of those constants in scripts instead of references to the constants themselves?

Are the numerical values stored in panel files?

rhwood, 2016-11-26:
It looks like the actual states are *not* stored in panel files, so no, the numerical values of the UNKNOWN, TRUE, and FALSE states is not stored in panel files.

bobjacobsen, 2017-02-12:
@rhwood: I spent a bit of time on this (as part of working on #3047).  As far as I can tell, changing this won't break anything in CATS, nor in anybody's stored JMRI files.  So I think we should go ahead and change it.

closed 2017-07-03 by bobjacobsen