JMRI/JMRI #2599: Block table doesn't save all changes: sensor, length

KenC57, 2016-11-27:
Possibly other fields, but sensor was the main one. I was updating a layout and filling in the sensors that match the blocks after the wiring was done (and the sensors became known). It only saved if you did edit on a single block. So if those fields don't update the block, they shouldn't be writable IMHO.

Cliff-Anderson, 2019-02-25:
Not certain that this is related to Ken's issue, but there is a potential overlap.  

Do not change the userName of a Block once it has been saved!  

The immediate results seem OK, but if the panel is saved and reopened, a new Block with the previous userName is automatically created.  The original systemName Block has the new userName, but any internal cross-connections are broken.  

A simple Panel has been created with an intentional and very obvious typo.  A copy of that Panel that includes an attempt to correct that typo has also been created.  Both of those simple Panel files are included in the attached zip file.  

Incidentally, PanelPro version 4.15.3+R7949343 causes an error when any Table window is opened when the connection is "None" but version 4.14 does not.  Both of these panels use "None" but that is not my primary concern.  

Changing a Block userName or a Sensor userName is something that I often need to do when developing a new Panel.  Unfortunately, many layers of changes had been made in a complicated Panel prior to discovering that I had made this mistake.  

```
2019-02-24 21:21:20,101 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
2019-02-24 21:21:20,161 beantable.ListedTableFrame            ERROR - An error occurred while trying to create the table for ID Tags [Listed Table Generation]
java.lang.NullPointerException
	at jmri.jmrit.beantable.AbstractTableTabAction.addToBottomBox(AbstractTableTabAction.java:121)
	at jmri.jmrit.beantable.IdTagTableAction.addToPanel(IdTagTableAction.java:330)
	at jmri.jmrit.beantable.IdTagTableTabAction.createModel(IdTagTableTabAction.java:73)
	at jmri.jmrit.beantable.AbstractTableTabAction.getPanel(AbstractTableTabAction.java:81)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.addPanelModel(ListedTableFrame.java:416)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.<init>(ListedTableFrame.java:362)
	at jmri.jmrit.beantable.ListedTableFrame.initComponents(ListedTableFrame.java:113)
	at jmri.jmrit.beantable.ListedTableAction$1.run(ListedTableAction.java:68)
	at java.lang.Thread.run(Unknown Source)
2019-02-24 21:21:20,165 beantable.AbstractTableTabAction      ERROR - java.lang.ArrayIndexOutOfBoundsException: -1 in add to Frame -1 null [Listed Table Generation]
```

[Panels.zip](https://github.com/JMRI/JMRI/files/2902505/Panels.zip)

dsand47, 2019-02-26:
The block userName problem is caused by the layoutBlock object.  The LE track component block assignment actually refers to the layoutBlock.  The userName for the layoutBlock matches the userName of the related Block object.

When a block userName is changed, the layoutBlock now points to nothing.  When the panel xml file is saved and loaded after start up, a new Block is created during panel loading to provide a related Block for the layoutBlock.  track block --> layoutBlock --> Block.

If a block is renamed and the track components are checked using the Edit menu, they no longer have a block assignment.  A new one has to be selected.  If all relevant track components are checked and fixed, the orphan layoutBlock goes away.

Dealing with this has been on my mental TODO list.  I will move it to the top.

Dave Sand