JMRI/JMRI #2601: Duplicate decoder definitions

rhwood, 2016-11-27:
Some duplicate decoder definitions have crept into our list of decoders. This is verifiable by running the jmri.jmrit.decoderdefn.DuplicateTest as an individual test (this test is not run on the CI servers).

```
$ ant test-single -Dtest.includes=jmri.jmrit.decoderdefn.DuplicateTest
Buildfile: /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/build.xml
[...output truncated...]
test-single:
    [junit] Running jmri.jmrit.decoderdefn.DuplicateTest
    [junit] Testsuite: jmri.jmrit.decoderdefn.DuplicateTest
    [junit] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 28.742 sec
    [junit] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 28.742 sec
    [junit] 
    [junit] ------------- Standard Output ---------------
    [junit] checked total of 2151
    [junit] ------------- ---------------- ---------------
    [junit] ------------- Standard Error -----------------
    [junit] log4j:WARN No appenders could be found for logger (jmri.jmrit.XmlFile).
    [junit] log4j:WARN Please initialize the log4j system properly.
    [junit] log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
    [junit] Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
    [junit] Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
    [junit] found duplicate for QSI Articulated Steam Ver. 7][QSI Magnum Articulated Steam
    [junit] found duplicate for QSI Diesel Ver. 6][BLI E3/E6 w/Mars
    [junit] found duplicate for QSI Diesel Ver. 6][BLI E3/E6 wo/Mars
    [junit] found duplicate for QSI Diesel Ver. 7][BLI E3/E6 w/Mars
    [junit] found duplicate for QSI Diesel Ver. 7][BLI E3/E6 wo/Mars
    [junit] found duplicate for QSI Diesel Ver. 7][BLI E7 w/Mars
    [junit] found duplicate for QSI Diesel Ver. 7][P2k F7B
    [junit] found duplicate for TCS DP2X decoder][DP2X
    [junit] found duplicate for TCS DP2X decoder][DP2X
    [junit] found duplicate for TCS DP2X decoder][DP2X-UK
    [junit] found duplicate for TCS DP2X decoder][DP2X-UK
    [junit] found duplicate for TCS Mx series decoders][M1
    [junit] ------------- ---------------- ---------------
    [junit] Testcase: testForDuplicateModels(jmri.jmrit.decoderdefn.DuplicateTest):	Caused an ERROR
    [junit] test failed, see System.err
    [junit] java.lang.AssertionError: test failed, see System.err
    [junit] 	at jmri.jmrit.decoderdefn.DuplicateTest.testForDuplicateModels(DuplicateTest.java:37)
    [junit] 
    [junit] 
    [junit] Test jmri.jmrit.decoderdefn.DuplicateTest FAILED

BUILD FAILED
/Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/build.xml:623: condition satisfied

Total time: 36 seconds
```

dheap, 2016-11-27:
@rhwood I'm looking into these. There should be a PR soon.

rhwood, 2016-11-27:
@dheap Thank you.

dheap, 2016-11-28:
@rhwood #2607 resolves QSI duplicates.

TCS duplicates will be the subject of a new issue.

AlanUS, 2016-11-28:
@dheap 
I did some clean up on the non-sound TCS decoders more than one year ago. There was an attempt by the previous authors to cover several firmware releases for each model, where applicable. There might be some overlap remaining, but generally speaking there are slight differences between each FW release. All this is obviously undocumented in official TCS literature, so difficult to check unless contacting TCS.
Note that during that "spring cleaning" I tried to build a clear chain between successive models using "show" and "replacementModel" attributes, leaving shown only for the very latest version.

dheap, 2016-11-28:
@AlanUS Unfortunately the technique used for the TCS replacements does not (and cannot be made to) work. The current situation has unfortunate side effects.

I need to take a break for possibly a day or so. Please be patient and (as alread promised) I will raise an issue explaining why the current technique is not workable and we can then look for a way forward.

dheap, 2017-01-22:
Sorry, had to take quite a long break due to personal reasons. Was just in the process of getting back to this today. As @rhwood has said #2915 is a duplicate of this.

rhwood, 2017-04-04:
Would the fix for the TCS models be to allow the ```replacementFamily``` and ```replacementModel``` attributes of the decoder also be attributes of the ```versionCV``` element (and to handle that change correctly within the JMRI code)?

dheap, 2017-04-04:
@rhwood I'm in the process of replying to your recent direct email to me. I'll send it very soon. Could we please take this offline to a direct email  consultation between the two of us for a bit? That would be most helpful and productive at this stage.

dheap, 2017-11-05:
#4372 resolves the `DuplicateTest` issue and also re-enables the `DuplicateTest` in `pom.xml`.

Can this be closed now?

bobjacobsen, 2017-11-05:
> Can this be closed now?

@dheap Thank you for all the work on this.  Yes, it can be closed.

closed 2017-11-05 by bobjacobsen