JMRI/JMRI #2617: Java Exception when modifying Signal Mast Logic

dsand47, 2016-11-29:
When I add or remove an optional sensor from an existing signal mast logic definition a Java exception occurs.  This interrupts signal logic until JMRI is restarted.

To test this, I added a sensor, tx/Active, to the SML for VMast2.  After saving the changes and restarting JMRI, I tested the SML and it worked as expected.  Subsequently removing the sensor results in the Java exception.

To recreate the problem:
1. Edit the panel
2. Right click **VMast2** and select Signal Mast Logic: VMast2
3. Select **Edit** 
4. Click on **Sensors**
5. Uncheck **tx**
6. Click on **Update Signal Logic**
7. Java Exception occurs

I suspect that this problem applies to other manual SML modifications.

This problem was discovered while helping with the **Signal Mast Logic an Art not a Science?** thread on the JMRI Yahoo group.

Test panel:
[sig test.xml.txt](https://github.com/JMRI/JMRI/files/619678/sig.test.xml.txt)

Console log:
[java exception.txt](https://github.com/JMRI/JMRI/files/619683/java.exception.txt)

KenC57, 2016-11-29:
I've also had this issue. And is is also indirectly hitting MSL, like updating a Block for the sensor assignment will trigger these sorts of NPE's.

dsand47, 2017-01-15:
I have narrowed the scope of this problem.

New steps to re-create the problem using the same xml file:

1. Edit the panel.
2. Right click **VMast1** and select **Signal Mast Logic: VMast1**.  The **Signaling Mast Pairs** window opens.
3. Close the **Signaling Mast Pairs** window.
4. Click the lower button icon to change the aspects.  The Java exception occurs.

- An aspect change for other masts does not create a Java exception.
- Changing the SML for a mast pair using the edit function from the Signal Mast Logic table does **NOT** create a subsequent Java exception.
- Stopping and re-starting JMRI eliminates the Java exception.

The root problem appears to be memory corruption induced by opening the Signaling Mast Pairs window which only applies to the selected signal mast.

[console-4.6.txt](https://github.com/JMRI/JMRI/files/706722/console-4.6.txt)

silverailscolo, 2017-02-04:
The NPE does no longer occur when I repeat your steps after PR #2998. Could you confirm?

dsand47, 2017-02-04:
I tested using build #1218 and the error no longer occurs.  Thank you.

Dave Sand

bobjacobsen, 2017-02-05:
@dsand47 is it OK to close this?  Or are there additional issues involved?  Thanks

dsand47, 2017-02-05:
Sorry, I forgot to do the close option.  Still learning.

Dave Sand

closed 2017-02-05 by dsand47

bobjacobsen, 2017-02-05:
> Sorry, I forgot to do the close option. Still learning.

Not a problem!  Thanks.