JMRI/JMRI #2619: Exception during "Add Marker" on a panel.

tgretton, 2016-11-30:
Pressing "Add Marker" on a panel window generates the following exception:

2016-11-29 20:25:22,848 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.util.MissingResourceException: Resource not found
	at jmri.Bundle.retry(Bundle.java:191)
	at jmri.Bundle.handleGetMessage(Bundle.java:150)
	at jmri.jmrit.Bundle.retry(Bundle.java:106)
	at jmri.Bundle.handleGetMessage(Bundle.java:150)
	at jmri.jmrit.display.Bundle.retry(Bundle.java:94)
	at jmri.Bundle.handleGetMessage(Bundle.java:150)
	at jmri.Bundle.handleGetMessage(Bundle.java:130)
	at jmri.jmrit.display.Bundle.getMessage(Bundle.java:41)
	at jmri.jmrit.display.Editor.locoMarkerFromInput(Editor.java:1509)
	at jmri.jmrit.display.controlPanelEditor.ControlPanelEditor.access$7(ControlPanelEditor.java:1)
	at jmri.jmrit.display.controlPanelEditor.ControlPanelEditor$6.actionPerformed(ControlPanelEditor.java:330)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

JMRI appears to become non-responsive.
 
Issue exists in both 4.4 and 4.5.4

Tom

tgretton, 2016-11-30:
Issue also exists in 4.5.6

bobjacobsen, 2016-11-30:
Is this the same as #2510 and #2515?  If so, it should be fixed in 4.5.7

tgretton, 2016-11-30:
Bob

It appears that it is.

Sorry for the duplication... I got 'add marker' in my head when I scanned the issues and missed 2510.

Thank you

Tom

closed 2016-12-01 by bobjacobsen