JMRI/JMRI #2624: Create test release 4.5.8

mattharris, 2016-11-30:
This is the next test release, building on 4.5.7 (not from HEAD of master). It will form the basis of the Production Release towards the end of the month.

Only minor changes and bug fixes to be considered.

mattharris, 2016-12-02:
For any proposed patches, please start from branch `pre-production-proposed-patches`.

It should be possible to then create 2 pull requests - one to `JMRI:pre-production-proposed-patches` and one to `JMRI:master`

bobjacobsen, 2016-12-12:
Currently contains #2634 and #2681 in addition to 4.5.7 contents.  First files building at usual place: http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.8/

bobjacobsen, 2016-12-14:
Second files built and available in the usual place.

Note: if 4.5.8 succeeds, we'll skip 4.5.9.  Only new, immediate fixes will be considered for 4.5.9.

bobjacobsen, 2016-12-15:
Released.

closed 2016-12-15 by bobjacobsen