JMRI/JMRI #2629: Added aspect numbers to NS-2008 signal system

SuzieTall, 2016-12-01:
[NS-2008-v1.21.zip](https://github.com/JMRI/JMRI/files/624931/NS-2008-v1.21.zip)

Added default aspect numbers, permissive specific appearance and references to indicate that this is the former Conrail signal system (rules N281-N292) rather than the native NS signal system (rules 301-310) in the referenced rulebook.

Aspect numbers match similar aspects in the UP-2008 and BNSF-1996 systems where possible to allow simple transfer between signal systems on common hardware.

Suzie x

KenC57, 2016-12-01:
I'll work this over. The spacing and linefeed cleanup needed.

 

-ken c

bobjacobsen, 2017-01-21:
@KenC57 Has this been merged, or can we go ahead with it?

SuzieTall, 2017-01-21:
@bobjacobsen it is showing up on the web already:-

http://jmri.org/xml/signals/NS-2008/aspects.xml

KenC57, 2017-01-21:
It says it was merged 3 Dec. I see the aspect values in the file
(dccAspect). So I think it should be closed now.

 

-ken c

bobjacobsen, 2017-01-22:
@SuzieTall Thanks!  Closing the item.

closed 2017-01-22 by bobjacobsen