JMRI/JMRI #2638: DecoderPro 4.4 hangs on OSX 10.11.16 after splash screen

Skengrobot, 2016-12-03:
My Java version is 1.8.0_111-b14.
DecoderPro version 4.4.
OSX 10.11.16.

I run the starter script and the splash screen shows up but nothing else happens. The debug output form the started script is:
```
PWD: '/Applications/JMRI'
CLASSPATH: '.:classes:java/classes:lib/RXTXcomm.jar:lib/Serialio.jar:lib/WinRegistry-4.4.jar:lib/beansbinding-1.2.1.jar:lib/bluecove-2.1.1-SNAPSHOT.jar:lib/bluecove-bluez-2.1.1-SNAPSHOT.jar:lib/bluecove-gpl-2.1.1-SNAPSHOT.jar:lib/ch.ntb.usb.jar:lib/commons-io-1.3.1.jar:lib/commons-lang3-3.2.1.jar:lib/gluegen-rt.jar:lib/jackson-annotations-2.0.6.jar:lib/jackson-core-2.0.6.jar:lib/jackson-databind-2.0.6.jar:lib/jakarta-regexp-1.5.jar:lib/javacsv.jar:lib/javax.servlet-api-3.1.0.jar:lib/jdom-2.0.5.jar:lib/jdom.jar:lib/jetty-http-9.3.9.v20160517.jar:lib/jetty-io-9.3.9.v20160517.jar:lib/jetty-security-9.3.9.v20160517.jar:lib/jetty-server-9.3.9.v20160517.jar:lib/jetty-servlet-9.3.9.v20160517.jar:lib/jetty-util-9.3.9.v20160517.jar:lib/jhall.jar:lib/jhidrawplugin.jar:lib/jinput.jar:lib/jlfgr-1_0.jar:lib/jmdns.jar:lib/joal.jar:lib/jul-to-slf4j-1.7.13.jar:lib/jython.jar:lib/log4j-1.2.17.jar:lib/mailapi.jar:lib/openlcb.jar:lib/pi4j-core.jar:lib/pi4j-device.jar:lib/pi4j-gpio-extension.jar:lib/pi4j-service.jar:lib/slf4j-api-1.7.13.jar:lib/slf4j-log4j12-1.7.13.jar:lib/smtp.jar:lib/vecmath.jar:lib/websocket-api-9.3.9.v20160517.jar:lib/websocket-client-9.3.9.v20160517.jar:lib/websocket-common-9.3.9.v20160517.jar:lib/websocket-server-9.3.9.v20160517.jar:lib/websocket-servlet-9.3.9.v20160517.jar:lib/xAPlib.jar:lib/xbee-api-0.9.jar:lib/xercesImpl.jar::jmri.jar:lib/RXTXcomm.jar:lib/Serialio.jar:lib/WinRegistry-4.4.jar:lib/beansbinding-1.2.1.jar:lib/bluecove-2.1.1-SNAPSHOT.jar:lib/bluecove-bluez-2.1.1-SNAPSHOT.jar:lib/bluecove-gpl-2.1.1-SNAPSHOT.jar:lib/ch.ntb.usb.jar:lib/commons-io-1.3.1.jar:lib/commons-lang3-3.2.1.jar:lib/gluegen-rt.jar:lib/jackson-annotations-2.0.6.jar:lib/jackson-core-2.0.6.jar:lib/jackson-databind-2.0.6.jar:lib/jakarta-regexp-1.5.jar:lib/javacsv.jar:lib/javax.servlet-api-3.1.0.jar:lib/jdom-2.0.5.jar:lib/jdom.jar:lib/jetty-http-9.3.9.v20160517.jar:lib/jetty-io-9.3.9.v20160517.jar:lib/jetty-security-9.3.9.v20160517.jar:lib/jetty-server-9.3.9.v20160517.jar:lib/jetty-servlet-9.3.9.v20160517.jar:lib/jetty-util-9.3.9.v20160517.jar:lib/jhall.jar:lib/jhidrawplugin.jar:lib/jinput.jar:lib/jlfgr-1_0.jar:lib/jmdns.jar:lib/joal.jar:lib/jul-to-slf4j-1.7.13.jar:lib/jython.jar:lib/log4j-1.2.17.jar:lib/mailapi.jar:lib/openlcb.jar:lib/pi4j-core.jar:lib/pi4j-device.jar:lib/pi4j-gpio-extension.jar:lib/pi4j-service.jar:lib/slf4j-api-1.7.13.jar:lib/slf4j-log4j12-1.7.13.jar:lib/smtp.jar:lib/vecmath.jar:lib/websocket-api-9.3.9.v20160517.jar:lib/websocket-client-9.3.9.v20160517.jar:lib/websocket-common-9.3.9.v20160517.jar:lib/websocket-server-9.3.9.v20160517.jar:lib/websocket-servlet-9.3.9.v20160517.jar:lib/xAPlib.jar:lib/xbee-api-0.9.jar:lib/xercesImpl.jar:/System/Library/Java'
Java CMD: '/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java'
APPNAME: 'DecoderPro'
OPTIONS: ' -noverify -Djava.security.policy=lib/security.policy -Djava.rmi.server.codebase=file:java/classes/ -Djava.library.path=.:lib/macosx:lib -Dapple.laf.useScreenMenuBar=true -Dcom.apple.macos.useScreenMenuBar=true -Dfile.encoding=UTF-8 -Xms96m -Xmx768m -Dgnu.io.rxtx.NoVersionOutput=true'
DOCK_OPTIONS: '-Xdock:name=DecoderPro -Xdock:icon=/Applications/JMRI/DecoderPro.app/Contents/Resources/DecoderPro.icns'
2016-12-03 14:01:56,289 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-12-03 14:01:57,124 util.Log4JUtil                        INFO  - This log is appended to file: /Users/gabitibi/Library/Preferences/JMRI/log/messages.log [main]
2016-12-03 14:01:57,125 util.Log4JUtil                        INFO  - This log is stored in file: /Users/gabitibi/Library/Preferences/JMRI/log/session.log [main]
2016-12-03 14:01:57,139 apps.AppsBase                         INFO  - DecoderPro version 4.4-R1ccf76b starts under Java 1.8.0_111 on Mac OS X x86_64 v10.11.6 at Sat Dec 03 14:01:57 PST 2016 [main]
2016-12-03 14:01:57,408 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f270841 [main]
2016-12-03 14:01:57,823 node.NodeIdentity                     INFO  - Using jmri-600308F1DBDC-3f270841 as the JMRI Node identity [main]
2016-12-03 14:01:57,962 jmrix.ConnectionConfigManager         INFO  - No connections configured. [main]
2016-12-03 14:02:00,519 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 07 00 40 00 21 21 00 consecutive timeouts = 0 [Transmit]
2016-12-03 14:02:03,513 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 21 21 00 consecutive timeouts = 0 [Transmit]
```

rhwood, 2016-12-05:
Please try using JMRI 4.5.7; it's a release candidate for 4.6 due later this month.

rhwood, 2016-12-10:
@Skengrobot Did 4.5.7 work correctly for you?

rhwood, 2016-12-17:
Closing since OP has not responded.

closed 2016-12-17 by rhwood

Skengrobot, 2016-12-28:
Hi, sorry for the silence. It was an issue was a stale save file in `~/Libraries`. Deleting that and reinstalling solved the problem.