JMRI/JMRI #2641: DCC++ turnouts don't invert

bobjacobsen, 2016-12-04:
On Dec 4, 2016, at 2:02 AM, Joff Hamilton joffham@btconnect.com [jmriusers] <jmriusers@yahoogroups.com> wrote:

>I have manged to get my sensors working through an Arduino using DCC++. Now
the only problem is that checking the "Inverted" box in the Sensor table
makes no difference. So, at the moment I have reversed the Track colour,
but just wanted to know if anyone else has this problem?

A quick look at jmrix.dccpp.DCCppTurnout shows an odd structure:  Calling setInvert changes the value of the _mThrown and _mClosed members, which are supposed to somehow do the inversion, but it's not clear how.  There's no local implementation of newCommandedState(..)  So the whole thing is a bit of a mystery.  

There aren't any test cases for jmrix.dccpp.DCCppTurnout

msunderwd, 2017-02-21:
There are several problems with the Sensors & Turnouts table... I believe the lack of functioning inversion is one of them.  I'm working it.

geowar1, 2017-02-21:
There's definitely something wack with those tables… the logical (to me) procedure should be to setup the sensor and set its invert checkbox so the turnout (in the turnout table) closed/thrown state  matches the physical reality on the layout; then if the turnout in the layout view doesn't match the sensor & reality then click the turnout's "invert" checkbox… except that then sometimes  "Direct Turnout Control" doesn't work. When it doesn't the only workaround I've found is to turn off both the sensor & turnout inverts… the sensor state will be wrong in the table but the physical & virtual layouts will match and "Direct Turnout Control" works.

bobjacobsen, 2017-02-21:
I think the "invert" function never was completed in the DCC++ implementation. What you may be seeing there is vestiges of whatever system was copied to set it up.  `jmri.jmrix.nce.NceTurnout`has a nice clean approach if you're looking for a model.

KenC57, 2017-02-21:
Keep in mind each implementation has the choice of supporting or not things
like inversions. So those are edits in the DCC++ sensor or turnout classes.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

geowar1, 2017-02-21:
On Feb 21, 2017, at 1:28 PM, Ken Cameron <notifications@github.com> wrote:
> Keep in mind each implementation has the choice of supporting or not things
> like inversions. So those are edits in the DCC++ sensor or turnout classes.

Seems like the option to “invert” a sensor should be in JMRI… not in an individual sensor's (driver?) implementation.

-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

pabender, 2017-02-22:
On 02/21/2017 05:47 PM, George Warner wrote:
> On Feb 21, 2017, at 1:28 PM, Ken Cameron <notifications@github.com> wrote:
> things
> classes.
>
> Seems like the option to “invert” a sensor should be in JMRI… not in
> an individual sensor's (driver?) [...]

Sensors are objects.

Objects make the decision about what they support and what they don't.

The basic premise is that it may not make sense for a particular 
implementation to support inversion, so you leave it to up to the 
implementation to make that decision.  You don't force behavior on an 
object from above.

For digital inputs, yes, we could usually force the behavior on a sensor 
implementation, but not all sensors are digital.

You could, for example, implement an analog sensor by setting a 
threshold representing an active signal.  Just inverting the decision 
usually isn't a good idea because there typically is more grey area to 
the decision than the threshold represents.  To deal with that grey 
area, you probably want set a different threshold when you invert the 
output.

Paul

bobjacobsen, 2017-02-22:
> Seems like the option to “invert” a sensor should be in JMRI… not in an individual sensor's (driver?) implementation.

Not sure what you're saying here. 

The decision whether to invert a specific individual sensor is (mostly) "in JMRI", in the sense that the code that sets the specific object to invert or not is in JMRI outside the object.

By a specific sensor _type_ may or may not be able to do that. That's up to the implementer of the class. They probably should, if they have time, no reason not to, etc. but nothing compels them to.

geowar1, 2017-02-22:
That's my point; why have the sensor (for lack of a better term) driver handle the inversion? Why can't JMRI just take what ever state is reported to it by the sensor software and invert it (if asked to by the user)? That would remove the necessity of the developer of the sensor software to implement the inversion code and it would make the inversion consistent between the different implementations for different sensors.

geowar1, 2017-02-22:
On Feb 21, 2017, at 6:28 PM, Paul Bender <notifications@github.com> wrote:
> On 02/21/2017 05:47 PM, George Warner wrote:
> > On Feb 21, 2017, at 1:28 PM, Ken Cameron <notifications@github.com <mailto:notifications@github.com>> wrote:
> > > Keep in mind each implementation has the choice of supporting or not 
> > things
> > > like inversions. So those are edits in the DCC++ sensor or turnout 
> > classes.
> >
> > Seems like the option to “invert” a sensor should be in JMRI… not in 
> > an individual sensor's (driver?) implementation.
> 
> Sensors are objects.
> 
> Objects make the decision about what they support and what they don't.
> 
> The basic premise is that it may not make sense for a particular 
> implementation to support inversion, so you leave it to up to the 
> implementation to make that decision. You don't force behavior on an 
> object from above.

Ok, so I have a sensor that says that a turnout is closed… but I look at my layout and the turnout is diverted. Ok, so I want to invert the sensor… BUZZ, sorry but the sensor object doesn’t support inversion.

Huh? So what do I do? Rewire my layout because the sensor software developer didn’t feel like implementing that feature? That isn’t right. If the sensor is reporting the wrong state then the end user should be able to invert it regardless of what “features” a sensor developer felt like implementing. 
-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

KenC57, 2017-02-22:
Some systems the concept of invert is not simple. Due to things like
feedback to the physical system care needs to be taking that it doesn't
cause a loop in the logic between JMRI and the system. Therefore it is
written in each system interface in the way that works for that system.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

msunderwd, 2017-02-22:
> George Warner, (408)249-7533 <(408)%20249-7533>

I think I found the problem.

There's a convenience method DCCppReply.getTOIsThrown() that looks into the
reply message from the base station and returns true==thrown /
false==closed.

Unfortunately, this method was looking at the turnout address (always
nonzero) instead of the turnout state (0 or 1), so it always returned
THROWN.

Once I fixed that, both DIRECT and MONITORING mode started working
"normally" -- including inversion using the checkbox in the JMRI turnout
table.

However, EXACT mode is still not working, but it looks like probably a
similar issue with either the DCCppMessage or DCCppReply not handling the
message cleanly.  I'll look into that tomorrow.

I also figured out that the text in the drop-down box in the turnout
configurator can be manipulated... so I'm doing that.  Could use some
feedback on the naming, though.  Note this only changes the text in the
dropdown, not the internal mnemonic value.

MONITORING mode sends a "T" turnout command to use the Base Station's
internal Turnout lookup table to send a command to an accessory decoder.
I'm renaming that to "BSTURNOUT"

EXACT mode sends a "Z" command to use the Base Station's Output lookup
table to manipulate a Base Station IO Pin.  I'm renaming that to "BSOUTPUT".

I think those names "will do" but there might be a better ... maybe
TURNOUTTABLE / OUTPUTTABLE, or something...

Thoughts?

- Mark

KenC57, 2017-02-22:
George,

 

The issue is that not all systems act the same way. And remember that a
sensor is just one type of object. A turnout for example can be inverted,
sometimes. In some systems this can create some side-effects, so those
systems don't support invert. But as long as it might not make sense for one
connection, it has to be optional for all so it can be enabled where it make
sense.

 

I will agree that sensors have the least amount of issues with being
inverted. And the code change is only a line or three most of the time.

 

Also in your example the real case is the wiring was backwards. Being able
to invert the sensor is just a lot easier than crawling under the layout.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

msunderwd, 2017-02-22:
Fixed the OUTPUT (EXACT / BSOUTPUT) mode case, too.  Had a bad if/else case
in the DCC++ Output Message interpretation.

Bugs, bugs, bugs.  Does anyone have a can of RAID around?

Going to bed now.

Thanks!

- Mark

bobjacobsen, 2017-02-22:
> That's my point; why have the sensor (for lack of a better term) driver handle the inversion? Why can't JMRI just take what ever state is reported to it by the sensor software and invert it (if [...]

> Ok, so I have a sensor that says that a turnout is closed… but I look at my layout and the turnout is diverted. Ok, so I want to invert the sensor… BUZZ, sorry but the sensor object doesn’t support [...]

You're conflating a couple of things here, so I may not be understanding your points entirely, but here goes:

- In your turnout example, it's not a-priori clear whether the problem is the turnout wiring (hence inversion) or the sensor wiring (hence inversion).  This matters for some kinds of hardware (e.g. Digitrax, RR-CirKits) but not others.  

 - Some systems really don't have a concept of "inverted" for their sensors and/or turnouts.  RFID sensors, for example, simply don't handle that.  It doesn't really make sense for some of the CAN (MERG) types IIRC.  Etc.  Making "invert" available in those cases is, in the opinion of the person who implemented them, Not A Good Idea.  Somebody else might not agree, but that's not the point:  They didn't write that code.

- Some times the person implementing a system just didn't spend the time to add inversion, or there's a bug in it, or the code is otherwise incomplete: maybe they spent their time on something else.  That's how these things go.  Nobody else gets to complain, because they didn't write that code

- One can certainly structure code different ways.  For example, AbstractSensor implements the `setInvert` and `getInvert` methods, but not the actual inversion.  Subclasses have to have code to implement do the inversion.  One could certainly do it other ways that would result in less duplication and decentralization. For example, we could have central code in the AbstractSensor class (though not every Sensor has to descend from that) that uses a `canInvert` method to decide whether to permit inversion.  (Sensor probably should copy that from Turnout anyway, so the table can show the invert checkbox disabled)  There are always lots of better ways to code something, but people shouldn't complain, because they didn't write that code.

If you want to improve this area of the code, that would be _great_.  That would certainly be a Big Contribution.  Improving the inheritance structure so there's less duplication, figuring out a good common implementation that works with all the various options and system-specific weird bits would be great.  

Improving the existing code is good.

But please don't beat up on the approach that the person who wrote the existing code, because they did the most important thing that nobody else did:  They wrote the code you see in front of you.  That's better than having none at all.

geowar1, 2017-02-22:
On Feb 21, 2017, at 9:44 PM, Bob Jacobsen <notifications@github.com> wrote:
> You're conflating a couple of things here, so I may not be understanding your points entirely, but here goes:
>
> In your turnout example, it's not a-priori clear whether the problem is the turnout [...]
My turnout sensor is 100% independent of the turnout… I expect it to tell me the state of the turnout regardless of what the turnout (driver or motor, etc.) thinks that it’s doing… or not.
But not unlike turnout motors I can’t always put the turnout sensor on the same side of the turnout… so sometimes active(-low) doesn’t mean thrown… it means closed.
> Some systems really don't have a concept of "inverted" for their sensors and/or turnouts. RFID sensors, for example, simply don't handle that. It doesn't really make sense for some of the CAN (MERG) [...]
That makes sense… except why then have the (user) option to invert it? Is there a way to query the sensor _driver_ ((?)for lack of a better word) to determine if it supports inversion (and hide/disable the invert GUI if not)? Currently it silently fails if not implemented… and that’s not a good user experience.
> Some times the person implementing a system just didn't spend the time to add inversion, or there's a bug in it, or the code is otherwise incomplete: maybe they spent their time on something else.  That's how these things go. Nobody else gets to complain, because they didn't write that code
> 
That really not what I’m complaining about. (Pause to genuflect toward all developers who’ve volunteer their time to write code for JMRI! ;-)
> One can certainly structure code different ways. For example, AbstractSensor implements the setInvert and getInvert methods, but not the actual inversion. Subclasses have to have code to implement [...]
My military sergeant always said “Don’t complain about something if you’re not willing to fix it…”. I think that applies here. ;-)
> If you want to improve this area of the code, that would be great. That would certainly be a Big Contribution. Improving the inheritance structure so there's less duplication, figuring out a good [...]
I suspect that that would just contribute to the mess… since there isn’t any way to force everyone to use my (new & improved!) superclass.
> Improving the existing code is good.
> 
> But please don't beat up on the approach that the person who wrote the existing code, because they did the most important thing that nobody else did: They wrote the code you see in front of you. That's better than having none at all.
> 
Again, not my intent… (and again I genuflect in their general direction…);

-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

pabender, 2017-02-22:
On Wed, Feb 22, 2017 at 1:51 PM, George Warner <notifications@github.com>
wrote:

>
> On Feb 21, 2017, at 9:44 PM, Bob Jacobsen <notifications@github.com>
> wrote:
> > In your turnout example, it's not a-priori clear whether the problem is
> the turnout wiring (hence inversion) or the sensor wiring (hence
> inversion). This matters for some kinds of hardware (e.g. Digitrax,
> RR-CirKits) but not others.
> >
> My turnout sensor is 100% independent of the turnout… I expect it to tell
> me the state of the turnout regardless of what the turnout (driver or
> motor, etc.) thinks that it’s doing… or not.
> But not unlike turnout motors I can’t always put the turnout sensor on the
> same side of the turnout… so sometimes active(-low) doesn’t mean thrown… it
> means closed.

Your sensor is not 100% independent of the turnout if the feedback of the
sensor is being referenced by the turnout object.  In OneSensor feedback
mode, for example, the turnout object actually interprets the input from
the sensor to determine which position the turnout is in.  It uses that
information so that it knows what the appropriate next command is to send
to the turnout.

> and/or turnouts. RFID sensors, for example, simply don't handle that. It
> doesn't really make sense for some of the CAN (MERG) types IIRC. Etc.
> Making "invert" available in those cases is, in the [...]

Turnout objects have a canInvert() method so you can ask if they are
invertable or not.

It looks like that may be missing on the sensor interface.

> AbstractSensor implements the setInvert and getInvert methods, but not the
> actual inversion. Subclasses have to have code to implement do the
> inversion. One could certainly do it other ways that [...]

 You can't force it, but you can make it happen.  Just refactor the
existing classes to use your new class as appropriate.

Paul

KenC57, 2017-02-22:
Paul,

 

I think one key part is the look and feel of the table view shows the invert
flag for all turnouts and sensors, but doesn't check if the particular
implementation responds true for canInvert(). I can't recall which systems
but I've added inversion to at least three of them over the years and it
only came after clicking the box, but it didn't make any difference. So this
would be easy in the table when it is showing only one connection type. The
'all' tab would make it difficult but just more code since it would be a
line by line thing.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2017-02-22:
On Wed, Feb 22, 2017 at 2:37 PM, Ken Cameron <notifications@github.com>
wrote:

> I think one key part is the look and feel of the table view shows the
> invert
> flag for all turnouts and sensors, but doesn't check if the particular
> implementation responds true for canInvert(). I [...]

jmri.jmrit.beantable.TurnoutTableAction defines the table model for the
turnout table.

It sets the editable property for the invert cell for the turnout in a row
based on the return value of the turnout's canInvert method.

I think that's all we really need to do.

The sensor table can behave the same way.

Paul

KenC57, 2017-02-22:
Paul,

Does it do anything like gray the box or just make it so you can't check the
box?? I'm thinking is there any visual difference??

-ken c

bobjacobsen, 2017-02-22:
Turnouts and the Turnout table support the `canInvert()` call. The default implementation (in `AbstractTurnout`) returns `false`.  That causes the table support to disable the invert checkboxes:
[image]
Note that the MERG turnout's invert checkbox is greyed out.

Adding `canInvert()` to Sensor and the corresponding table support might be useful.

msunderwd, 2017-02-23:
Specific to the Sensor ignoring the "inverted" checkbox... my DCCppSensor class was totally ignoring the inversion flag.  I'm testing a fix for that now.

msunderwd, 2017-02-27:
Also, I added a testcase for DCCppTurnout.  It verifies that the canInvert() and setInvert() interface methods work as expected.  So both the turnouts and sensors should behave properly with 4.7.1.

bobjacobsen, 2017-02-27:
This is a long thread that raises a bunch of issues. But the original issue that opened it was "DCC++ turnouts don't invert". That seems fixed now. 

Would it make sense to close this, and have people concerned about Sensor behavior and/or other items open new Issues?

msunderwd, 2017-02-27:
We could, but the Sensor issues should also be fixed... so the only purpose
in opening a separate issue would be to close it again and have it as
documentation.

I would leave this closed for both Turnouts and Sensors.  If (and only if!)
4.7.1 shows further or continuing problems with Sensors and Turnouts, we
can open a new issue to track that.

I can insert some commentary in the 4.6 release notes that indicate the
problems are known issues that are (hopefully) fixed in 4.7.x

As best I understood the flow of the discussion, it stemmed from a bug in
the Sensor/Turnout handling but diverged (incorrectly IMHO) from "fix the
bug" to "maybe it's not a bug but an architectural issue" (which it isn't,
again, IMHO)

- Mark

KenC57, 2017-02-27:
I think any other systems found with the same problem should be a new issue.
But as this went, was it found that for systems that report canInvert() ==
false didn't enable the checkbox in the table??

-ken c

msunderwd, 2017-02-27:
No, I don't think so.

I think the whole problem was that I had not IMPLEMENTED the conversion
logic in the DCC++ sensors, and it was incorrectly implemented in the
Turnouts.

I saw no evidence at all that there was actually a problem outside of the
above.  I think the discussion just broadened itself into speculation...

- Mark

KenC57, 2017-02-27:
Mark,

 

One part as I understood was that the table let you play with the check box
for invert, but it didn't do anything. So the DCC++ specific was it didn't
support the invert. The table was in error since it led the user to think
the check box should have worked since there was no message that it wasn't
supported.

-ken c

msunderwd, 2017-02-27:
Right.  That was part of what was wrong with the DCC++ implementation (not
the GUI).  At least on the Turnouts side, I was storing the checkbox state,
which meant that I was responding to the GUI correctly...

... but I was not USING the checkbox state (correctly) when forming the
messages to the DCC++ Base Station...

... so while the checkbox in the GUI worked as expected (as far as being
able to click it), it had no *practical* effect on the system behavior.

Hence the SYMPTOM was a GUI thing... but the ROOT CAUSE was in the DCC++
interface implementation.

- Mark

bobjacobsen, 2017-03-12:
Can we close this now?

closed 2017-04-16 by bobjacobsen