JMRI/JMRI #2647: Layout editor cross-over icon does not use track size for diverging route

dsand47, 2016-12-06:
Turnouts use the connected track size to set the size of the legs.  Cross-overs only set the straight routes to the adjacent track sizes.  The diverging route is not changed when it should have the **mainline** size.  This makes it more difficult to see that the cross-over is active when the cross-over is connecting tracks that have been defined as **mainline**.
[image]
Ideally it should look the same as the 2 turnout approach.

The single and double slip icons behave differently.  If any leg is connected to track defined as **mainline**, all segments of the slip are changed.

geowar1, 2017-01-21:
Looks like this was intentional… but I don't know why… I'll fix this for now but if someone complains (for whatever reason) I may have to change it back. ;-)

geowar1, 2017-02-15:
I'm assigned but I can't close this (no "close" button)… how do it close it?

geowar1, 2017-02-15:
Unless the OP has to regress and close it?

dsand47, 2017-02-16:
The change works fine.  I like that the cross-over leg is small when not active but expands when the cross-over is thrown.

I will close this issue.

Dave Sand

closed 2017-02-16 by dsand47