JMRI/JMRI #2657: Marker trouble

bobjacobsen, 2016-12-07:
>With my updated JMRI 4.5.6, when I try to place markers on my layout panel, "Add loco from roster..." works but the text cannot be edited.  Therefore, the marker must be deleted and a new one added, which takes too much time while dispatching.  "Add loco..." will bring up a window, but the marker does not appear on the panel.  This worked well in the previous versions of JMRI.  Any suggestions?
>
>Ted Smith <tfsmith365@yahoo.com>
><tfsmith18@q.com> (Please also respond to this email address in case I do not receive your response on my Yahoo Mail. Thanks.)

silverailscolo, 2016-12-08:
No problem creating and editing such a label in 4.5.7 (right mouse click > Edit text), so updating to that version from 4.5.6 is advised:
[image]

silverailscolo, 2016-12-10:
I asked the user by mail what OS and Locale (language setting) he uses, and if this happens in Layout Editor (as the message suggests) or instead in Panel Editor or Control Panel Editor.

bobjacobsen, 2017-01-21:
Thought to be fixed, see above.

closed 2017-01-21 by bobjacobsen