JMRI/JMRI #2663: Add Menu to Console pane

silverailscolo, 2016-12-09:
Does anyone remember if there was a reason not to display a menu bar - with the JMRI and Windows menus as a bare minimum - when the Console is frontmost? On macOS I find it cumbersome to minimize/drag the Console out of the way in order to see other JMRI panes. Switching windows (panes) using the menu is a handy method everywhere else in JMRI.

mattharris, 2016-12-09:
Probably because the person who originally wrote the JMRI console support isn't a mac user and didn't notice.

I wonder who that was... :-)

silverailscolo, 2016-12-10:
Sounds logical to me. Will try to add a bar. (and thanks again for creating the Console)

geowar1, 2016-12-12:
Using command-~ (tilda) to switch windows is embedded in my (Mac) muscle memory…

silverailscolo, 2016-12-12:
Sure, but PanelPro (Java) is just one icon to Command-~
And access to our excellent JMRI Help menu comes as a bonus.

silverailscolo, 2016-12-16:
Addressed in PR #2679.

closed 2016-12-16 by silverailscolo