JMRI/JMRI #2664: Create Range of Turntouts, sensors, reporters can fail silently

devel-bobm, 2016-12-09:
When adding a range of turnouts, sensors, or reporters using the "Add..." button on the appropriate object "Table" GUI, the tool can fail silently under certain circumstances.

The tool implies that it will create a range of objects, starting with the address entered in the "Hardware Address" GUI text field, and extending to that number plus the value from the "Number to add" GUI text field minus 1.  When that range does not overlap any existing objects in that implied range of SystemNames, the tool succeeds without issue.

When that implied range overlaps one or more existing objects in that implied range, the tool will not behave as expected, and does not provide any feedback to the user, other than by not creating the expected objects.  Known failure modes include:
. If the tool encounters a group of 9 or fewer pre-existing objects while attempting to create all of the sensors in the range, the tool skips those pre-existing objects and continues creation of new objects following that group of pre-existing objects.
. If there are more than 9 consecutive pre-existing objects within the implied range of SystemNames, the tool stops after creating the last new sensor numerically before the group of 10+ pre-existing objects.  The tool does not create the remaining requested objects and the tool does not inform the user in any way.
. A combination of the above two above, where one or more occurence of the first case can happen before an occurence of the second case.

This behavior appears to be related to:
.  A loop counter in the<AbstractSensorManager.getNextValidAddress()>, <AbstractTurnoutManager.getNextValidAddress()>, and <AbstractReporterManager.getNextValidAddress()>,  methods, and within those connection-specific classes which override those methods.
. A mechanism in the GUI-related code to which notices the failure to create a new sensor but which fails to notify the user.

devel-bobm, 2016-12-09:
Would it be better to pre-check that it is possible to create all of the new objects, sequentially, with the system names implied by the "hardware address" field and "Nnumber to add" field, and fail, with user notification, before creating any objects if the entire requested range cannot be created?

KenC57, 2016-12-09:
I would go with the pretest that validates the whole range is available from
where the start is given. If no start point is given, then I'd say the first
available range that fits the length should be given. Both of these become
simple if the test getNextValidAddress was given the range in addition to
the starting point. For no given start, it should use 1, if no range given,
then a range of 1 is used.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

devel-bobm, 2016-12-09:
Current behvior, when value for "Hardware Address" field is empty or not a number, is to show a dialog box and refuse to create any sensors.  Current behavior, when "Add a sequential range" is checked and the "Number to add" field is empty or not a number, is to show a dialog box and refuse to create any sensors.

silverailscolo, 2017-01-10:
I agree there should be some feedback about what happened. Won't be hard to insert.
I noticed creation jumping over existing items but hadn't noticed it maxed at 9. Stopping after adding one more item after a bigger block is unexpected.
When we change this, may I add that I never liked the first new item lacking i.e.``:0`` in the name. Never looks clean when naming the first Sensor LS20, an the second LS20:1
In PR #2839 I changed the (renamed) "Number of items" field into a JSpinner, so it's impossible not to enter something. Spinner starts at "1", max. is 100 with a dialog asking for confirmation if set above 65.

bobjacobsen, 2022-05-09:
I tried to recreate it and got something a bit different:

1) Start a clean LocoNet Simulator version of PanelPro
2) In the Sensor table, add sensor `10` and a sequential range of 11 (i.e. LS10 through LS20
3) Now add sensor `1` and a sequential range of 11 more

You get: 

[image]

So there is an error message, but it's really unhelpful when it hits one _before_ the first existing one.

bobjacobsen, 2022-05-09:
Another oddness:
1) Start a clean LocoNet Simulator version of PanelPro
2) In the Sensor table, add sensor 10 and a sequential range of 5 (i.e. LS10 through LS14)
3) Now add sensor 1 and a sequential range of 20

The result is not the sensors LS1 through LS20 one might expect, but rather LS1 through LS24.  It's as if it's saying "Well, you wanted 20 _new_ ones, so I'm skipping over the existing ones".  That seems counter-intuitive.

icklesteve, 2022-05-09:
@bobjacobsen I keep on meaning to go back to this,

Links heavily to  #8022

The conclusion I came to, after various false starts, is that the jmrix class should not be telling the UI what the next **_available_** address is.
The jmrix should be telling the UI what the **_next_** address is, then the UI determines if that address is available.
If already in use, request the next one up to limit / jmrix decides it can't do a next.

The UI doing the available or not would provide better feedback to user on what is happening, especially when UserNames are involved in the mix.

HW Address - UserName
```
About to create Sensors LS1 - with sequential 3
LS1 -My Sensor1
LS2 - MySensor2 
LS3 - MySensor3
```
or how about if MySensor2 was already defined?
```
NOT About to create Sensors LS1 - with sequential 3
LS1 -My Sensor1
LS2 - MySensor2 ALREADY FOUND SENSOR WITH UserName MySensor2 - 
```

Imho first step is to recode the jmrix manager classes to provide the **next** address, available or not.

bobjacobsen, 2022-05-09:
I can work on that a little.

One remaining question:  What happens when a Sensor/Turnout/Reporter/Light already exists?

1) It seems to me that the request is to "make sure the next N are available", in which case I think it should silently leave those in place.  No user name changes, etc.

2) But it could also be "create these next N", in which case having one exist is an error.

Any preference?

bobjacobsen, 2022-05-10:
Thanks!

One more question on how best for it to work:  What should it do when it reaches a hardware discontinuity?

 - C/MRI nodes have only a certain number of pins.  If more Sensors/Lights/Turnouts are requested than available pins, should it stop there, or perhaps go on to the next node if it exists?
 - NCE AIUs only have 14 pins per AIU, so the same kind of question arises:  Stop at 14, regardless of number requested, or go on to the next AIU? Note that JMRI might not know that AIU exists until Sensors on it are defined.

devel-bobm, 2022-05-10:
In my _opinion_, attempting to create a range which spans a known hardware discontinuity is something that deserves positive user notification.  Whether that notification is generated when the creation process gets to the discontinuity or when the initial request is made is a different question, and I do not have a strong opinion on that.  

It would be "nice to have" a feature where the user could respond "yes, I really mean to create across the boundary", and the hardware-specific implementation would "do the right thing" (or, alternately, "the assumed, logical thing") with respect to new object addresses and the hardware boundary (or boundaries).  

That way, a user with 5 new NCE AIUs at sequential cab addresses could add all 5 at the same time using a sequence size of 70 objects (5 AIUs x 14 inputs per AIU), if he responded to the dialog with "yes, I really mean to do that".

bobjacobsen, 2022-05-11:
PR #1093 has a re-implementation of this function that should work better. 

- It doesn't attempt to properly handle hardware boundaries, i.e. skipping from one NCE AIU to the next. There's a place to add that if somebody wants to.
- It does(attempt to) create N sequential items regardless of whether they already exist or not.
- If there is a problem, it will create all the items up to the system name that has a problem
- Several error messages and warnings have been clarified.

It would be great if people would try it. It's currently flagged as for after 5.0, but we could conceivably merge it before that if people find that it's an improvement.

danielb987, 2022-05-13:
Note that it's PR #10973 and not PR  #1093 that @bobjacobsen refers to in the comment above. I have updated that comment with the correct PR.

bobjacobsen, 2022-05-13:
Closed by #10973

closed 2022-05-13 by bobjacobsen