JMRI/JMRI #2671: Frequent build failures in Travis Headless builds

rhwood, 2016-12-10:
We are persistently getting the following error messages during builds on the Travis CI Headless tests:
```
[junit] ERROR - run: Exception: java.io.IOException: Pipe broken [testInvoke] jmri.jmrix.AbstractMRTrafficController.?()
[junit] ERROR - Exit from rcv loop [testInvoke] jmri.jmrix.AbstractMRTrafficController.?()
```

The only testInvoke methods are in:
- jmri.jmrit.beantable.BlockTableActionTest
- jmri.jmrit.beantable.OBlockTableActionTest
- jmri.jmrit.beantable.RouteTableActionTest
- jmri.jmrit.beantable.SignalHeadTableActionTest
- jmri.jmrit.ussctc.OsIndicatorTest

pabender, 2016-12-12:
@rhwood,

The testInvoke is a red herring.

This happens on successful builds as well, for example:
    [junit] ERROR - run: Exception: java.io.IOException: Pipe broken [Receive] jmri.jmrix.AbstractMRTrafficController.reportReceiveLoopException()

    [junit] ERROR - Exit from rcv loop [Receive] jmri.jmrix.AbstractMRTrafficController.receiveLoop()

(copied from the successful headless build of #2677 ).

The problem is a test that uses a TrafficController either:
a) not shutting down the TrafficController properly.
or
b) not initializing the TrafficController completely.

This may or may not be an error from the point of view of the test.  In other words, if the test that causes is only a constructor test for some object, then the test can succeed even if there is a communication failure (which the error message actually indicates).

My inclination is to say something isn't completely shutdown.  The reason I say this is that the error message must be occurring at some point in the processing where the check for unexpected Error messages has already occurred.  Otherwise, we would be getting a test failure from this, and we're not.

(I bet if you check the log of the test run you pulled that from, you'll find some other error also occurred).

Paul

rhwood, 2016-12-12:
Attaching complete log containing a failure [133797028.txt](https://github.com/JMRI/JMRI/files/645624/133797028.txt)

bobjacobsen, 2016-12-19:
#2726 added more debug info for this.  Now merged.  If you see this error, please record the traceback and class name in a comment here.

bobjacobsen, 2016-12-19:
One example: Failure was in the Lenz XNetTrafficController's receive thread:
```
    [junit] ERROR - run: Exception: java.io.IOException: Pipe broken in class jmri.jmrix.lenz.XNetPacketizer [testInvoke] jmri.jmrix.AbstractMRTrafficController.?()
    [junit] java.io.IOException: Pipe broken
    [junit] 	at java.io.PipedInputStream.read(PipedInputStream.java:321)
    [junit] 	at java.io.PipedInputStream.read(PipedInputStream.java:377)
    [junit] 	at java.io.DataInputStream.read(DataInputStream.java:149)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:779)
    [junit] 	at jmri.jmrix.lenz.XNetTrafficController.loadChars(XNetTrafficController.java:273)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:860)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:714)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:649)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
    [junit] ERROR - Exit from rcv loop in class jmri.jmrix.lenz.XNetPacketizer [testInvoke] jmri.jmrix.AbstractMRTrafficController.?()
```

pabender, 2016-12-19:
Thanks @bobjacobsen

What that doesn't tell us still is what test actually started the traffic controller that is exiting due to the broken pipe.

I'm convinced the current cause is actually something happening in one of the Roco Z21 tests, but I can't figure out which file is causing it.

(and yes, the error reported from the XNetTrafficController actually can be coming from the Roco Z21s XPressNet tunnel.)

Paul

bobjacobsen, 2016-12-20:
> What that doesn't tell us still is what test actually started the traffic controller that is exiting due to the broken pipe.
>
> I'm convinced the current cause is actually something happening in one [...]

All agreed.

The only thing I can think of is to put the test name (e.g. from @Rule TestName) into the thread names in the tests.  If we’re certain that it’s the XNetTrafficController and kin, that’s a manageable amount of work. Would that help?

rhwood, 2016-12-20:
Other than two other tests that fail (and total failure on the Linux CI servers) the run tests in maven PR would should which shows logging by test.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2016-12-20:
> Other than two other tests that fail (and total failure on the Linux CI servers) the run tests in maven PR would should which shows logging by test.

Perhaps I’m missing something, but part of the problem seems to be that the threads eventually fail _outside_ the test interval.  The tests are supposed to be catching the ERROR messages, but they miss them due to the delay (or some other reason).

Maybe the reason I haven’t been able to track this down is that I’m following that wrong cause.  Have I misunderstood?

pabender, 2016-12-20:
Sent from my iPad

On Dec 20, 2016, at 10:25 AM, Bob Jacobsen <notifications@github.com> wrote:

> Perhaps I’m missing something, but part of the problem seems to be that the threads eventually fail _outside_ the test interval. The tests are supposed to be catching the ERROR messages, but they [...]

Yes, that is what I understand.

Paul

pabender, 2016-12-20:
Sent from my iPad

On Dec 20, 2016, at 10:13 AM, Bob Jacobsen <notifications@github.com> wrote:

> All agreed.
>
> The only thing I can think of is to put the test name (e.g. from @Rule TestName) into the thread names in the tests. If we’re certain that it’s the XNetTrafficController and kin, that’s [...]

That might help, but what I am not sure of is if the test failure is because of a specific test for the traffic controller or a test that needs a traffic controller.

Paul

pabender, 2016-12-22:
I think I actually understand what is going on here.

I'm debugging a problem for a Roco Z21 user, and I noticed that when I shutdown, I get this sequence (this is from a run of DecoderPro):

     [java]  61096 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
     [java]  63122 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 07 00 40 00 21 81 A0 consecutive timeouts = 0 [Transmit]

That timeout message we are also seeing in the tests.  

The problem here is that there are actually two traffic controllers involved for the single connection.  We have the Roco Z21TrafficController and an XPressNetTrafficController associated with the XPressNet tunnel.

What I believe the issue is is that the Z21TrafficController is terminating it's transmit thread before the XPressNetTrafficController executes its shutdown hook.  

The shutdown hook on the XPressNetTrafficController makes it so the last thing we do on an XPressNet system is exit service mode (which is what the 21 81 A0 does.  The 07 00 40 00 are headers for the Z21).

Paul

bobjacobsen, 2017-01-02:
This is costing people a lot of time, and it's becoming a source of real frustration. We should bypass the test if it can't be reasonably fixed.

pabender, 2017-01-02:
> On Jan 1, 2017, at 6:14 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> This is costing people a lot of time, and it's becoming a source of real frustration. We should bypass the test if it can't be reasonably fixed.
> 
This is one of the things I've been working on.

Basically, this is a thread synchronization problem at shutdown.  In the tests, the Z21 traffic controller is shutting down before the Z21XPressNet tunnel's traffic controller shuts down, and I am working out the best way to synchronize the shutdown process.

Paul

bobjacobsen, 2017-01-02:
Thank you for working on the underlying problem.

But a review of the last 48 hours of CI builds shows that headless Linux is failing more than 3/4 of the time, resulting in a lot of time wasted on reruns for just this one test.

Do you know which JUnit test or tests are triggering it?  If we can localize it, we can disable just the test(s) causing this failure.  Otherwise, I think it makes sense to disable the entire`jmrix.roco.z21` test tree.

pabender, 2017-01-02:
Sent from my iPhone

> On Jan 2, 2017, at 1:32 PM, Bob Jacobsen <notifications@github.com> wrote:
> Thank you for working on the underlying problem.
> 
> But a review of the last 48 hours of CI builds shows that headless Linux is failing more than 3/4 of the time, resulting in a lot of time wasted on reruns for just this one test.
> 
Yes, I understand that.
> Do you know which JUnit test or tests are triggering it?
> 
I thought I had it isolated, but that did not make the problem go away.

It is in one of the tests that actually causes the XPressNet tunnel to be initialized.
> If we can localize it, we can disable just the test(s) causing this failure. Otherwise, I think it makes sense to disable the entirejmrix.roco.z21 test tree.
> 
You can't disable a whole test tree any longer.  

That will cause every test run to fail due to missing test coverage.

Paul

bobjacobsen, 2017-01-03:
> You can't disable a whole test tree any longer.
>
> That will cause every test run to fail due to missing test coverage.

Sure we can.  We just back off the test coverage requirement. See PR #2812.

pabender, 2017-01-03:
On 01/02/2017 08:20 PM, Bob Jacobsen wrote:
>     You can't disable a whole test tree any longer.
>
>     That will cause every test run to fail due to missing test coverage.
>
> Sure we can. We just back off the test coverage requirement. See PR
> #2812 <https://github.com/JMRI/JMRI/pull/2812>.

 From a technical point of view, yes, we can.

 From a software engineering point of view we can't.

The fact is that by disabling the minimum test coverage requirement, we:
a) set a bad precedent
b) allow ourselves to use a hatchet to fix a problem when what we need 
is a scalpel.
c) remove a tool that has caught several errors when new tests have been 
submitted in the last month alone, one of which basically disabled all 
testing.

Frankly, I think we need to get out of the bad habit of disabling tests 
that do something we don't like. (running too long, or causing periodic 
(even sometimes frequent errors.)

By disabling the tests, we are not letting the tests do their job.  The 
job of the test is to show us there are errors.  Yes, sometimes those 
errors are hard to find, and don't occur all the time, but that's why we 
write the tests in the first place.

Now, I just got back home, and I have my notes in front of me, I can 
tell you precisely that the tests that are causing the problem are 
either the Z21XPressNetStreamPortController tests or the 
Z21XPressNetTunnel tests.

I've submitted a new pull request that comments just those two test 
files out.

Paul

rhwood, 2017-01-05:
https://travis-ci.org/JMRI/JMRI/jobs/189169329 failed about two hours ago with:
```
[junit] ERROR - run: Exception: java.io.IOException: Pipe broken in class jmri.jmrix.roco.z21.Z21XNetPacketizer [testInvoke] jmri.jmrix.AbstractMRTrafficController.?()
    [junit] java.io.IOException: Pipe broken
    [junit] 	at java.io.PipedInputStream.read(PipedInputStream.java:321)
    [junit] 	at java.io.PipedInputStream.read(PipedInputStream.java:377)
    [junit] 	at java.io.DataInputStream.read(DataInputStream.java:149)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:779)
    [junit] 	at jmri.jmrix.lenz.XNetTrafficController.loadChars(XNetTrafficController.java:271)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:860)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:714)
    [junit] 	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:649)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
    [junit] ERROR - Exit from rcv loop in class jmri.jmrix.roco.z21.Z21XNetPacketizer [testInvoke] jmri.jmrix.AbstractMRTrafficController.?()
```

rhwood, 2017-01-15:
I am able to persistently generate this issue in my maven-tests branch when running all tests (```mvn test -Djava.awt.headless=true```) except when I exclude the ```jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest``` test.

When run alone (```mvn test -Dtest=jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest```), this test does not throw any errors.

When run with the other tests in its package (```mvn test -Dtest=jmri.jmrix.roco.z21.simulator.**```), this test throws a different warning:
```
Running jmri.jmrix.roco.z21.simulator.configurexml.ConnectionConfigXmlTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.843 sec - in jmri.jmrix.roco.z21.simulator.configurexml.ConnectionConfigXmlTest
Running jmri.jmrix.roco.z21.simulator.configurexml.PackageTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 sec - in jmri.jmrix.roco.z21.simulator.configurexml.PackageTest
Running jmri.jmrix.roco.z21.simulator.ConnectionConfigTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.107 sec - in jmri.jmrix.roco.z21.simulator.ConnectionConfigTest
Running jmri.jmrix.roco.z21.simulator.PackageTest
Tests run: 17, Failures: 0, Errors: 0, Skipped: 2, Time elapsed: 2.012 sec - in jmri.jmrix.roco.z21.simulator.PackageTest
Running jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest
ERROR - Exception opening socket [main] roco.z21.simulator.Z21SimulatorAdapter.?()
java.net.BindException: Address already in use
	at java.net.PlainDatagramSocketImpl.bind0(Native Method)
	at java.net.AbstractPlainDatagramSocketImpl.bind(AbstractPlainDatagramSocketImpl.java:93)
	at java.net.DatagramSocket.bind(DatagramSocket.java:392)
	at java.net.DatagramSocket.<init>(DatagramSocket.java:242)
	at java.net.DatagramSocket.<init>(DatagramSocket.java:299)
	at java.net.DatagramSocket.<init>(DatagramSocket.java:271)
	at jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapter.run(Z21SimulatorAdapter.java:78)
	at java.lang.Thread.run(Thread.java:745)
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 2.896 sec - in jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest
Running jmri.jmrix.roco.z21.simulator.Z21SimulatorLocoDataTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 sec - in jmri.jmrix.roco.z21.simulator.Z21SimulatorLocoDataTest
Running jmri.jmrix.roco.z21.simulator.Z21SimulatorTest
Tests run: 2, Failures: 0, Errors: 0, Skipped: 2, Time elapsed: 0 sec - in jmri.jmrix.roco.z21.simulator.Z21SimulatorTest
Running jmri.jmrix.roco.z21.simulator.Z21XNetSimulatorAdapterTest
Tests run: 7, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 sec - in jmri.jmrix.roco.z21.simulator.Z21XNetSimulatorAdapterTest
```

pabender, 2017-01-16:
@rhwood,

That address in use error is because the Z21 SimulatorAdapter actually creates a UDP server socket and then connects to it ( the simulator is simulating the actual device, which we connect to using UDP)

Also, the test causing the error is happening twice in rapid succession.  Your log above shows:
Running jmri.jmrix.roco.z21.simulator.PackageTest
Tests run: 17, Failures: 0, Errors: 0, Skipped: 2, Time elapsed: 2.012 sec - in jmri.jmrix.roco.z21.simulator.PackageTest
Running jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest
ERROR - Exception opening socket [main] roco.z21.simulator.Z21SimulatorAdapter.?()

The Z21SimulatorAdapterTest is referenced inside of jmri.jmrix.roco.z21.simulator.PackageTest.  Operating systems sometimes take some time to release server ports, even after an application has completed.

You either need to eliminate the duplicated test run or provide enough time between the tests that the OS releases the port.

This is not an inherent problem with the test itself.

Paul

pabender, 2017-01-24:
#2920 may have actually hit on the real cause of this.

The Z21Adapter has a keepalive timer that periodically sends messages.

The dispose method in that class wasn't killing the timer, so the timer would fire off after the test had finished, which explains why we were seeing delayed failure when we started all this.

rhwood, 2017-02-08:
[Complete log](https://github.com/JMRI/JMRI/files/760473/145864879.txt) from [failed Travis CI build](https://travis-ci.org/JMRI/JMRI/jobs/199216750).

rhwood, 2017-02-08:
Log of a failed build is at https://travis-ci.org/rhwood/JMRI/jobs/199578258

pabender, 2017-02-25:
I haven't seen any new reports of this since #3032 was merged, is this error still happening, or can we close this issue?

rhwood, 2017-02-25:
I think this has been fixed and can be closed.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

closed 2017-02-25 by pabender