JMRI/JMRI #2699: Create test release 4.7.1

bobjacobsen, 2016-12-14:
JMRI test release 4.7.1 is made from a JMRI 4.7.7 base, with all PRs to master since then applied.

silverailscolo, 2016-12-15:
That would be 4.5.7 base?

bobjacobsen, 2016-12-15:
Sorry, that's a typo in the original comment. Fixed.

In effect, it's really just being made from "master" after we make sure that everything that was brought in as patches to 4.5.7 (e.g. for 4.5.8 and perhaps 4.5.9) are merged onto master also.

rhwood, 2016-12-17:
Can we include #936 in 4.7.1 or should that wait until 4.7.2?

silverailscolo, 2016-12-17:
Would be nice to have PR #2693 included. All checks have passed.

mattharris, 2016-12-20:
> Can we include #936 in 4.7.1 or should that wait until 4.7.2?

Part of me would like to have a release where this is the only change so, either we have 4.7.1 being 4.6 + #936 or we do 4.7.2 being 4.7.1 + #936.

This is just so that we can be sure if any issues crop up that there is a clear known state to go back to for verification.

The reason for the proposed cautious approach is due to the fact that this is quite a fundamental change to the connections infrastructure.

Thoughts?

bobjacobsen, 2016-12-20:
I agree that we should have a clear strategy for tracking down any issues that arise in the wild.  

Having 4.7.1 be just 4.6+#936 helps make the cause of any problem reports clearer.  But that would normally lead to #936 being in 4.7.2, etc.  Are we ready to do that right away?  There's a lot of change stacked up for a next 4.7.* release that we should also get out into the wild.

We could also have a special-purpose set of downloads that are 4.6+#936 and ask the community to try them.  (Not sure of the status of the rxtx-to-pjc-3 branch, this might be tedious to create but should be possible)   We don't usually get much uptake of those, but we could ask for a couple of specific cases and probably get a response.  I'd be particularly interested in getting a few examples along the spectrum of control line use, e.g. EasyDCC, C/MRI with original adapter, original LocoBuffer, etc. A month of that, and we could merge.

In any case, I think it's time to get this out where people can start to use it and give us feedback.

KenC57, 2016-12-20:
If we want that kept separate, then you'd have the 4.7.1 as all except #936,
and then follow quickly with 4.7.2. Or we'd have to make a special branch
for this, I'm not sure on what's better. But since lots of things have piled
into 4.7.1 already it is far from being able to do 4.6 + #936 as 4.7.1. At
least that's my understanding of Git, which isn't very good but I think this
one is correct.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

mattharris, 2016-12-20:
@KenC57 It's not too difficult to do 4.6 + #936 in git in another branch - you just need to choose the right spot to branch from (doesn't have to be HEAD of master).

In essence, take the 4.6 tag; create a branch from that; merge #936 into that branch.

rhwood, 2016-12-20:
#936 was last updated from JMRI/master on 12 Dec 2016. 4.5.7 was branched from master before that, and I don't know which changes between 4.5.7 and 12 Dec 2016 in master would be in the release (I think most were unit testing changes).

Randall Wood

bobjacobsen, 2016-12-20:
> #936 was last updated from JMRI/master on 12 Dec 2016. 4.5.7 was branched from master before that, and I don't know which changes between 4.5.7 and 12 Dec 2016 in master would be in the release (I think most were unit testing changes).

Thanks! I think both the following are straight-forward:

- A test release that’s v4.6 and #936
- A test release that’s head-of-master and #936, either before, after or simultaneous with a regular head-of-master test release

I’m leaning toward 4.6+#936 because it’ll be most reliable on _other_ things.  

Then

- put that out as 4.7.1, and decide in a month about whether 4.7.2 branches from that (but that delays a bunch of other change)
- put that out as a special release, e.g. 4.7.0 which we label as a special test which we’d like our friends to help us with, and put out an all-contents 4.7.1 at the same time

I’m leaning toward the second of those.  I’d be happy to build and release Wednesday/Thursday this week. I think early is good, as people might have time to test over the next 10 days.

Should we move this to JMRIdevelopers?

mattharris, 2016-12-20:
I've created a new issue to continue this discussion - #2733

rhwood, 2017-02-07:
I think all known instances of the schema causing existing, created by JMRI, panels to fail have been resolved. Can we release 4.7.1 for feedback?

silverailscolo, 2017-02-09:
By and large I would think so. Today I found one more panel that won't load. Added it as a as issue #3033 with xml attached.

bobjacobsen, 2017-02-19:
The release-4.7.1 branch has been created. All future PR merges and commits should please get the 4.7.2 milestone, not 4.7.1. Thanks.

bobjacobsen, 2017-02-27:
4.7.1 is being released from the current head of master.

closed 2017-02-27 by bobjacobsen