JMRI/JMRI #2704: Athearn OEM SoundTraxx Econami and Tsunami2 Decoder Profiles

cpalomarez, 2016-12-15:
Athearn has developed proprietary default configurations for the Econami and Tsunami2 products.
Would like to help add them into JMRI Decoder Profiles

Tsunami2
EMD - Broken into three different decoders and sorted by Prime Mover and broken into dynamic and Non-Dynamic by muting the DB sound in the mixer
           1) EMD#1 - 567 Prime Movers
                              A) Dynamic
                              B) Non-Dynamic
           2) EMD#2 - 645 Prime Movers
                              A) Dynamic
                              B) Non-Dynamic
           3) EMD#3 - 710 Prime Movers

GE - At this time a single decoder with all GE Prime Movers on it

Horns - A targeted selection of horns honed per prototype application

Econami
EMD - A single decoder with three distinct EMD Prime Movers
                              A) Dynamic
                              B) Non-Dynamic

[E9900.001B Checklist AR1.docx](https://github.com/JMRI/JMRI/files/655601/E9900.001B.Checklist.AR1.docx)
[E9900.002B Checklist AR1.docx](https://github.com/JMRI/JMRI/files/655600/E9900.002B.Checklist.AR1.docx)
[T0102.001B Checklist AR1.docx](https://github.com/JMRI/JMRI/files/655655/T0102.001B.Checklist.AR1.docx)
[T0102.002B Checklist AR1.docx](https://github.com/JMRI/JMRI/files/655654/T0102.002B.Checklist.AR1.docx)
[T0203.001B Checklist AR1.docx](https://github.com/JMRI/JMRI/files/655602/T0203.001B.Checklist.AR1.docx)
[T0203.003B Checklist AR1.docx](https://github.com/JMRI/JMRI/files/655603/T0203.003B.Checklist.AR1.docx)

KenC57, 2016-12-15:
Christopher P,

 

A quick look at two of the docs looks like this will be a breeze for one of
the decoder folk to set up. I just wish more vendors would do something like
this for us. It would improve the results from sometimes us having to guess
at some of these details. That would make life better for customer, us
supporting JMRI, and the vendor.

 

Only thing I'd have to say is too bad this wasn't 3 weeks ago. As you know
we do a twice a year 'production' release and this one is just about wrapped
up and should hit the site as 4.6 within the next few days. But the work on
these decoders will take a bit of time for somebody, so they won't  be in
that production release but in the test releases that would be starting with
4.7.1.

 

Thank you for providing such nice clear data for us to work from. Way to
often it ends up being some amount of guesswork.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

cpalomarez, 2016-12-16:
Thank you for the feedback! My desire is to continue this moving forward if this is useful to the JMRI community. We're still working on the EMD 710 Tsunami2 board and the GE Tsu2. Both of these decoders drive larger 1" speakers. This means different graphic EQ levels once we settle on a speaker.

closed 2021-04-14 by stale[bot]