JMRI/JMRI #2710: Remove support for file: and resource: portable paths

rhwood, 2016-12-17:
Support for ```resource:``` and ```file:``` portable file paths should be removed.

From https://sourceforge.net/p/jmri/mailman/message/35556824 (with some markdown formatting applied):

The ```resource:``` and ```file:``` portable file paths were deprecated in jmri.util.FileUtil sometime before 17 July 2011 (the first release after that is 2.13.1, but that is the farthest back I can search through the history for [FileUtil][1]). The JavaDocs for that class at that time indicate they were already deprecated. Sometime after that, they were encapsulated in constants, and I am getting deprecation warnings in the three methods in the one class using them.

[1]: https://github.com/JMRI/JMRI/blob/1c52b95d0ac4e442618e3b2da69c585d0bdf8c9b/java/src/jmri/util/FileUtil.java

rhwood, 2016-12-17:
There are two questions I have about doing this:
- [ ] What path should [VSDSound.java][1] use for its vid_file_base on line 35? As far as I can tell, the current value never pointed to an existing file or directory.
- [ ] [general.xsd][2] specifically allows the ```resource:``` portable path on line 247. How should that no longer be allowed? (It also allows the ```file:``` portable path, but since that also matches the file: URL scheme, I'm not concerned about that prefix.)

[1]: https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/vsdecoder/VSDSound.java#L35
[2]: https://github.com/JMRI/JMRI/blob/master/xml/schema/types/general.xsd#L247

klk32003, 2016-12-18:
> What path should VSDSound.java use for its vid_file_base on line 35? As far as I can tell, the current value never pointed to an existing file or directory.

Could that path be program:java/src/jmri/jmrit/vsdecoder/vsdfiles?

vsd_file_base on line 35 in VSDSound.java is not used. Please see [VSDPreferences.java](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/vsdecoder/VSDecoderPreferences.java) on line 77.

rhwood, 2016-12-18:
>> What path should VSDSound.java use for its vid_file_base on line 35? As far as I can tell, the current value never pointed to an existing file or directory.
>
> Could that path be program:java/src/jmri/jmrit/vsdecoder/vsdfiles?

No. Nothing under java/src is available as a file in a shipping version of JMRI; not only that the path you propose does not exist either.

> vsd_file_base on line 35 in VSDSound.java is not used. Please see VSDPreferences.java on line 77.

It is used in [SoundBite.java][1], as a quick grep reveals:
```
$ grep -rn vsd_file_base *
java/src/jmri/jmrit/vsdecoder/Diesel3Sound.java:495:                //buf.setURL(vsd_file_base + filename);
java/src/jmri/jmrit/vsdecoder/Diesel3Sound.java:519:                    //buf.setURL(vsd_file_base + filename);
java/src/jmri/jmrit/vsdecoder/SoundBite.java:103:                        sound_buf.setURL(vsd_file_base + filename);
java/src/jmri/jmrit/vsdecoder/SoundBite.java:295:        sound_buf.setURL(vsd_file_base + filename);
java/src/jmri/jmrit/vsdecoder/VSDSound.java:35:    protected String vsd_file_base = "program:resources/sounds/vsd/";
```

[1]: https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/vsdecoder/SoundBite.java#L103

rhwood, 2016-12-18:
Am using ```program:resources/vsdecoder``` for vsd_file_base in VSDSound.java in #2719.

klk32003, 2016-12-18:
That's working for me, thanks. Sorry about my non productive comment.

KenC57, 2016-12-18:
I've got a lot of panel files from many places and times. Many of these are
left over from helping a user with an issue. LOTS of the files have
resources: typically using that to find the icons. So my question on this is
how would a user migrate from one to the other?? We can do it with editing
of the panel files but users shouldn't need that. Yes it is the same path as
program: how would a normal user convert??

 

The most frequent use seems to be the <signalheadicon entries for accessing
all the indication icons. I'll retest and see what the range of ages are for
these files. And I'll test if I open it and do a save if that changes it.
Ok, many of these files are from the <2010 range but I'll need to inspect
closer to figure pattern of dates.

 

But key in my mind is how to provide a way for normal users who had
something with these paths change them to the modern names.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-12-18:
> my question on this is
> how would a user migrate from one to the other?

I think if a file with old-style references is loaded and then stored from a current JMRI release (4.5 series or 4.6), the references will update. 

Am away from main machine, can't easily check, but IIRC some of the LoadStore JUnit tests check that. But it would be good to confirm it with one of the user files you have too.

KenC57, 2016-12-18:
Trying a few old files, first one threw this:

2016-12-18 11:08:05,890 configurexml.ErrorHandler             ERROR -
ClassNotFoundException while loading from file
/C:/Users/kcameron/JMRI/CrossoverExample.xml Exception:
java.lang.ClassNotFoundException:
jmri.configurexml.AbstractSignalHeadManagerXml [AWT-EventQueue-0]

java.lang.ClassNotFoundException:
jmri.configurexml.AbstractSignalHeadManagerXml

       at java.net.URLClassLoader.findClass(Unknown Source)

       at java.lang.ClassLoader.loadClass(Unknown Source)

 

Another file threw:

2016-12-18 11:20:27,444 configurexml.ErrorHandler             ERROR -
ClassNotFoundException while loading from file
/C:/Users/kcameron/JMRI/Phil-CTCClinic1.xml Exception:
java.lang.ClassNotFoundException: jmri.configurexml.DefaultLogixManagerXml
[AWT-EventQueue-0]

java.lang.ClassNotFoundException: jmri.configurexml.DefaultLogixManagerXml

       at java.net.URLClassLoader.findClass(Unknown Source)

       at java.lang.ClassLoader.loadClass(Unknown Source)

 

Or others seemed to be one of the above cases. No panel contents seemed to
get loaded. So this may show those old files are already 'lost' due to other
issues like old class names that changed over time.

-ken c

pabender, 2016-12-18:
Sent from my iPhone

> On Dec 18, 2016, at 11:24 AM, Ken Cameron <notifications@github.com> wrote:
> Or others seemed to be one of the above cases. No panel contents seemed to
> get loaded. So this may show those old files are already 'lost' due to other
> issues like old class names that changed over time.

We should probably fix those though. There should be regression tests for things like this to make sure we can still open those older files.

Can you provide a minimal file that produces the two errors above?

Paul

rhwood, 2016-12-18:
> We should probably fix those though. There should be regression tests for things like this to make sure we can still open those older files.

We shouldn’t fix those, since we made the deliberate decision to break those in #1441:

"Deprecations around loading XML files wait until the second major release: JMRI 4.* doesn't have to load files written by JMRI 2., but does need to be able to load files written by any JMRI 3."

pabender, 2016-12-18:
Sent from my iPhone

> On Dec 18, 2016, at 12:15 PM, Randall Wood <notifications@github.com> wrote:
> 
> > > On Dec 18, 2016, at 11:24 AM, Ken Cameron <notifications@github.com> wrote:
> > > Or others seemed to be one of the above cases. No panel contents seemed to
> > > get loaded. So this may show those old files are already 'lost' due to other
> > > issues like old class names that changed over time.
> > 
> > We should probably fix those though. There should be regression tests for things like this to make sure we can still open those older files.
> 
> We shouldn’t fix those, since we made the deliberate decision to break those in #1441:
> 
> "Deprecations around loading XML files wait until the second major release: JMRI 4.* doesn't have to load files written by JMRI 2., but does need to be able to load files written by any JMRI 3."
> 

That may be, but what version caused the class not found exceptions  Ken's trace generated to occur?  

That's actually a separate issue, even though it came up here.

Paul

KenC57, 2016-12-18:
Paul,

 

I've got one around 22k, another 58k, and larger. The 22k was a test for a
crossover and signals as I recall. It works with LocoNet Simulator. Last mod
date was 2007. Do you think that would be useful??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2016-12-18:
I would expect that Ken’s issues would be duplicated in 4.5.1 if not in 4.4. The activity from multiple developers on the JMRI/remove-deprecations branch includes explicitly removing tests for 2.x compatibility since that was intentionally broken.

Randall Wood

KenC57, 2016-12-18:
On the case of depreciations where there is a XXX replaces ZZZ, I think we
should provide a migration script that gives a window, file chooser, etc.
and does the conversion. Doing one tool and adding blocks for 'what we cut
out from 2' to bring it forward would be easy enough. In fact I don't think
the tool would even have to offer 'migrate 2.x to 4.x' but just bring it
forward to 'now'. 

 

Granted each major rev we have to add to this those things we have
depreciated, but I'd see it as a cost of removing the old to go on with the
new. Not providing users a way forward seems a modern idea and that's sad.
And given the environment and history of our sorts of users, keeping
something old happens a lot. They would not do a new save unless they had a
known reason. I've been to layouts that didn't change a panel in the last 3
years.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-12-19:
> On Dec 18, 2016, at 5:53 PM, Randall Wood <notifications@github.com> wrote:
> 
> I would expect that Ken’s issues would be duplicated in 4.5.1 if not in 4.4. The activity from multiple developers on the JMRI/remove-deprecations branch includes explicitly removing tests for 2.x compatibility since that was intentionally broken.

That may be, but we can't silently fail on this, which is the situation Ken describes.  

We need to let the users know that the file format is no longer supported and what the migration path to restore the files to functional state is.  That is going to save a lot of headaches in the long run.

Paul

rhwood, 2016-12-19:
That's why I asked the original question. How do I invalidate the schema so it will fail? Should I just update the existing schema file so the files relying on long deprecated forms fail validation, or should cause JMRI to reject the old schema itself entirely, so that we then have to figure out if the file can be forced to validate against a different schema?

At this point any silent failures would be because some else decided that silent failures when failing to find an expected file are acceptable.

pabender, 2016-12-19:
Sent from my iPhone

Sent from my iPhone
> On Dec 18, 2016, at 9:35 PM, Randall Wood <notifications@github.com> wrote:
> That's why I asked the original question. How do I invalidate the schema so it will fail? Should I just update the existing schema file so the files relying on long deprecated forms fail validation, or should cause JMRI to reject the old schema itself entirely, so that we then have to figure out if the file can be forced to validate against a different schema?
> 
The problem is that these files Ken has may very well not fail validation. ( I have not checked yet )

The schema doesn't ( on purpose) specify the class names that are valid, and the failure is a class not found exception.

Paul

KenC57, 2016-12-19:
And the 'class not found' is not silent. I do get a pop up and then the
details in the console. One surprise is that nothing from the file seems to
get kept IF the error is not the one about the editors. It seems that even
when the error is from signals, the sensors and turnouts don't show in the
tables. So the other part is you only get one error at a time and not that
there are multiples. Only when editing a file did I find another same sort
of 'class not found'.

 

Both cases are testing to be simple sed script type fixes. But for all our
users, something nicer would be in order. My feeling is not something
automatic, that would burden the runtime with the growing baggage of old
stuff. But a script so it is standalone from the program. This keeps the
fixing stuff in one clean place.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2016-12-19:
The schema does specify which file prefixes are valid for JMRI portable paths, which is the concern of this PR. Please let this PR conversation stick to the problem this PR is dealing with, and not other issues.

So, should a file fail validation if the schema does not allow the portable path, and since the schema file in question has no version statement, or should the failure occur when a file not found error is displayed (assuming all paths that check for files display those errors)?

KenC57, 2016-12-19:
If the 'file not found' is not silent and the console message contains the
portable path, then us support people would know the issue when we see the
log message. Anything displayed is seldom caught and passed on. And I
suspect at the point the 'file not found' happens it is in some generic part
of the code. So the exception has to float up high enough so the portable
path is displayed as that is the clue. This I don't know if it happens or
not. But now I have at least one file that I manually fixed the classes, so
it has the depreciated portable path, in this case resource:. So it will
show quickly how it gets handled.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-12-19:
Sent from my iPhone

> On Dec 18, 2016, at 10:13 PM, Randall Wood <notifications@github.com> wrote:
> 
> The schema does specify which file prefixes are valid for JMRI portable paths, which is the concern of this PR. Please let this PR conversation stick to the problem this PR is dealing with, and not other issues.

I did mention earlier that the class not found error was not directly related to this discussion.

However, if the class not found prevents us seeing a validation failure, that makes these issues related.

Ultimately the issues are related in the sense that the cause is the removal of deprecated code.

> So, should a file fail validation if the schema does not allow the portable path, and since the schema file in question has no version statement, or should the failure occur when a file not found error is displayed (assuming all paths that check for files display those errors)?

I think it should occur as early as possible.

As this relates to the class not found exception Ken reported, it is likely that a file containing the deprecated portable path prefixes will also contain deprecated class names, so failing at file validation time will prevent at least some of those errors.

Paul

rhwood, 2016-12-19:
I am proposing to generally solve the bad prefix problems by forcing schema validation whenever a panel file is loaded (through either a startup option or the Panels->Open Panels... menu item). This will force verbose logging and a dialog with the error to open (users can currently ignore the dialog error for the session, should that be removed?).

I think having the error log point to a URL explaining the different validation errors and how to fix them (and the dialog providing a link to the built in help with the same content) might be the best way forward (later providing scripts that can fix errors, and referring to those scripts in the help content can improve upon this).

There are elements in the schemas that are still considered valid that have been deprecated since 2.8, but which will force file not found errors since they will be skipped over in the schema validation, but I think it unlikely that a schema validation and load of an older panel will pass without getting us to the point where we can guide a user through updating their panels manually.

KenC57, 2016-12-19:
This sounds like a reasonable path, only issue is more drag at startup by
testing the panel file before trying to load it. But you are hinting it will
be simpler to provide a more meaningful error message this way than trying
to figure out the offending line from what happens in the depths of loading.

 

As we spot things, and figure scripted solutions to fix them, we just keep
building updating script to handle the forward conversions.

 

Anybody have a good idea for what we should name the tool??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2016-12-19:
> This sounds like a reasonable path, only issue is more drag at startup by
testing the panel file before trying to load it. But you are hinting it will
be simpler to provide a more meaningful error message this way than trying
to figure out the offending line from what happens in the depths of loading.

I think it will be simpler; it will also prevent the offending file from being loaded until the issue is fixed (currently, some errors allow the loading to continue, and I don't know what happens if a file is loaded with errors and immediately saved (or saved with only 6 of 8 errors fixed)--most likely whatever is saved is not what the user wants.

> As we spot things, and figure scripted solutions to fix them, we just keep
building updating script to handle the forward conversions.
>
> Anybody have a good idea for what we should name the tool??

Don't assume there would be one tool. It may turn out that more than one tool is better than one "fix all the things" tool.

commit 40fd535a (2016-12-22): Merge pull request #2719 from rhwood/deprecated-portable-paths

Remove deprecated portable paths.
Fixes #2710.
closed 2016-12-22 by rhwood