JMRI/JMRI #2729: Untitled Window, since 4.5.7 at least

KenC57, 2016-12-19:
It seems to happen late in the startup sequence. Window has only the window control bar (minimize, maximize, close) but no other contents. It will close and go away just fine. But what is creating it?? It may predate 4.5.7 but that's when I first remember seeing it. It is clearly in 4.6 and 4.7,1ish.
-ken c

rhwood, 2016-12-19:
Is anything shown in the logs?

bobjacobsen, 2016-12-19:
Does it show in the Window menu?  If you do 
```
import jmri
print jmri.util.JmriJFrame.getFrameList()
```
as a Python script, does it list more than one (there should be a main window, but nothing else)

KenC57, 2016-12-20:
I've not been able to find any clues. One thing I do see is a number of
threads popping up and going out during the startup, some I can't read quick
enough. I suspect there may be a clue in there. But I've not been able to
figure how to trap what it might be.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

KenC57, 2016-12-20:
Here is what I see from that test:

>>> import jmri

... print jmri.util.JmriJFrame.getFrameList()

[jmri.util.JmriJFrame[frame0,629,389,661x302,invalid,layout=java.awt.BorderL
ayout,title=PanelPro,resizable,normal,defaultCloseOperation=DO_NOTHING_ON_CL
OSE,rootPane=javax.swing.JRootPane[,9,38,643x255,invalid,layout=javax.swing.
JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,ma
ximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true],
jmri.jmrit.logix.NXFrame[frame1,0,0,0x0,invalid,hidden,layout=java.awt.Borde
rLayout,title=,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootP
ane=javax.swing.JRootPane[,0,0,0x0,invalid,layout=javax.swing.JRootPane$Root
Layout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,min
imumSize=,preferredSize=],rootPaneCheckingEnabled=true],
jmri.util.JmriJFrame[frame2,0,0,557x290,invalid,layout=java.awt.BorderLayout
,title=Script
Entry,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax
.swing.JRootPane[,9,38,539x243,layout=javax.swing.JRootPane$RootLayout,align
mentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,pr
eferredSize=],rootPaneCheckingEnabled=true],
jmri.util.JmriJFrame[frame3,37,37,386x289,invalid,layout=java.awt.BorderLayo
ut,title=Script
Output,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=java
x.swing.JRootPane[,9,38,368x242,invalid,layout=javax.swing.JRootPane$RootLay
out,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimu
mSize=,preferredSize=],rootPaneCheckingEnabled=true]]

bobjacobsen, 2016-12-20:
Thanks!  That shows that the 2nd window created is:

jmri.jmrit.logix.NXFrame[frame1,0,0,0x0,invalid,hidden,layout=java.awt.BorderLayout,title=,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,0,0x0,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]

So this is a NXFrame with a null title, probably the thing you’re looking for.

But I don’t see anything causing that when I do a clean run.  Is it perhaps being referenced from some file you load at startup?

Bob

> Here is what I see from that test:
>
>
> ... print jmri.util.JmriJFrame.getFrameList()
>
> [jmri.util.JmriJFrame[frame0,629,389,661x302,invalid,layout=java.awt.BorderL [...]

KenC57, 2016-12-20:
I just created a brand new profile, didn't set anything. It is still there.
Now on timing, I'm getting the main screen before this shows in the window
list. But shortly after that, like move to a different menu item and back,
and there it is.

 

-ken c

KenC57, 2016-12-20:
PS: also tried the new clean profile from the 4.6 exe, same results.

 

-ken c

bobjacobsen, 2016-12-26:
I haven’t been able to recreate this, sorry.

I suggest adding something like

```
new Exception(“creating that odd frame”).printStackTrace();
```

in the `NXFrame()` ctor at line 111 in `java/src//jmri/jmrit/logix/NXFrame.java` to see who’s creating this. It’s remotely possible that it’s OK that it’s being created early, but it’s not expected to be shown until later, so you could also put a stack traceback in `init()`e.g. near line 115.

If that fails to track it down, you might need to add a local setVisible(..) methods to NXFrame, something like this:

```
	public void setVisible(boolean t) {
		new Exception(“setVisible(“+t+”)").printStackTrace();
		super.setVisible(t);
	}
```
which should make it really clear who’s showing the window.

Sorry I can’t get it to show, so I can’t test this for you.

Bob

KenC57, 2016-12-26:
Bob,

 

I think we've seen this before. Something in the warrants kicking off a
window. Here is the trace:

2016-12-26 08:52:37,811 util.FileUtilSupport                  INFO  - File
path scripts: is C:\Users\kcameron\workspace\jmri-20160902\JMRI\jython\
[main]

java.lang.Exception

       at jmri.jmrit.logix.NXFrame.<init>(NXFrame.java:112)

       at jmri.jmrit.logix.NXFrame.lambda$0(NXFrame.java:97)

       at java.util.Optional.orElseGet(Unknown Source)

       at jmri.jmrit.logix.NXFrame.getDefault(NXFrame.java:96)

       at
jmri.jmrit.logix.WarrantPreferences.setNXdata(WarrantPreferences.java:396)

       at
jmri.jmrit.logix.WarrantPreferences.apply(WarrantPreferences.java:391)

       at
jmri.jmrit.logix.WarrantPreferencesPanel.setGUI(WarrantPreferencesPanel.java
:87)

       at
jmri.jmrit.logix.WarrantPreferencesPanel.<init>(WarrantPreferencesPanel.java
:58)

       at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native
Method)

       at sun.reflect.NativeConstructorAccessorImpl.newInstance(Unknown
Source)

       at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(Unknown
Source)

       at java.lang.reflect.Constructor.newInstance(Unknown Source)

       at java.lang.Class.newInstance(Unknown Source)

       at java.util.ServiceLoader$LazyIterator.nextService(Unknown Source)

       at java.util.ServiceLoader$LazyIterator.next(Unknown Source)

       at java.util.ServiceLoader$1.next(Unknown Source)

       at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:180)

       at apps.Apps$3.lambda$0(Apps.java:376)

       at java.util.Optional.ifPresent(Unknown Source)

       at apps.Apps$3.run(Apps.java:375)

       at java.lang.Thread.run(Unknown Source)

bobjacobsen, 2017-01-02:
There must be something in your Warrant preferences settings that's triggering this.  What do yours look like?  Could you attach the XML file please?  

Mine looks like:

[image]

The points to `jmri.jmrit.logix.WarrantPreferences 391 for making the NXFrame; that seems to always happen, so it sounds like the normal path is to create the NXFrame (not great, but probably OK), and then only show it when desired.  I don't understand why it's sometimes being shown, and sometimes not after that. (I haven't found the `setVisible(..)` call for the NXFrame)

So here's a really weird thing in NXFrame:

```
    public static NXFrame getDefault() {
        if (GraphicsEnvironment.isHeadless()) {
            return null;
        }
        NXFrame instance = InstanceManager.getOptionalDefault(NXFrame.class).orElseGet(() -> {
            return InstanceManager.setDefault(NXFrame.class, new NXFrame());
        });
        if (!instance.isVisible()) {
            instance.setTrainInfo(null);
            instance.clearRoute();
        }
        return instance;
    }
```

See how it's doing certain things only there's a GUI, that the window isn't displayed?  When asked for the default instance?  I have no idea what the intention is there.

KenC57, 2017-01-02:
I couldn't see any warrant related stuff but here is the whole directory. So
could it be triggered from nothing being there (default setups?).

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2017-01-02:
The NXFrame is a singleton (I don't know why the warrants code does not use a single non-GUI manager with windows that listen for properties to update themselves, but it doesn't so the management of the warrants is tangled into the GUI code). It used to maintain a static reference to its singleton instance, but FindBugs did not like that, so now its singleton instance is maintained by the InstanceManager.

spassmo, 2017-01-05:
I've been noticing the untitled window as well for the last few 4.5.x versions.
Loconet & C/MRI
Basic startup, prior to loading panel. no startup actions defined.
4.6,  Win 10

bobjacobsen, 2017-01-21:
@rhwood "t used to maintain a static reference to its singleton instance, but FindBugs did not like that, so now its singleton instance is maintained by the InstanceManager."

Could that be creating the window at a different time, so that the initializion takes a different path & shows the window?  I still can't recreate this.

KenC57, 2017-01-21:
I see the window showing in the 'Window' menu list about 1 to 20 seconds
after the main window opens after initialization. It sometimes gets in the
list before my first chance to check. Most of the time I can check the list
at least once and it is not showing. By the second or third time, it is
there in the list. I should note this is still happening with a fresh
environment as after a few months it seems I get files showing as conflicts
but diff to the same. So I make a new workspace and move on.

-ken c

KenC57, 2017-03-12:
With 4.7.2ish-20170311* the window doesn't show up any more.

closed 2017-04-16 by bobjacobsen