JMRI/JMRI #2732: Initialisation of JOAL can take a long time on some systems

mattharris, 2016-12-20:
From comment in #2723:

> I'm finding the init of Joal is taking a very long time, like up to a
minute. This hangs opening the tables view. Are others seeing this??

bobjacobsen, 2017-02-12:
Is this related to #1880?

KenC57, 2017-02-12:
Looks like the same thing. The OP (and I) were seeing it as the tables were taking a long time to open the first time around and saw Joal was the delay. Here Matt was seeing it the other way around. But same results. After this change by Matt I was still seeing issues with the tables view taking a long time at first open, but it seemed to be something other than Joal.

KenC57, 2017-03-12:
This seems to have appeared again, at least on my current code (almost 4.7.2). Takes about a minute before I get the tables. If I pause the thread (table lister) it is always in the JOAL part.

bobjacobsen, 2017-03-12:
> If I pause the thread (table lister) it is always in the JOAL part.

Could you post one of those trac backs, please?

KenC57, 2017-03-12:
Here is a trace from pausing the table generator:

Thread [Listed Table Generation] (Suspended)       

               owns: Class<T> (com.jogamp.common.util.IOUtil) (id=416)

               owns: PrintStream  (id=432)          

               owns: Class<T> (com.jogamp.common.util.cache.TempJarCache)
(id=418)               

               owns: Class<T> (com.jogamp.openal.util.ALut) (id=423)     

               owns: DefaultAudioManager  (id=433)      

               ProcessImpl.waitForInterruptibly(long) line: not available
[native method] 

               ProcessImpl.waitFor() line: 449    

               IOUtil.testDirExec(File) line: 909  

               IOUtil.testDirImpl(File, boolean, boolean, String) line: 939

               IOUtil.getSubTempDir(File, String, boolean, String) line: 999

               IOUtil.getTempDir(boolean) line: 1114     

               TempFileCache.<clinit>() line: 80 

               TempJarCache.initSingleton() line: 88        

               Platform$1.run() line: 309             

               AccessController.doPrivileged(PrivilegedAction<T>) line: not
available [native method]        

               Platform.<clinit>() line: 287           

               ALFactory.<clinit>() line: 68           

               ALut.alutInit() line: 64      

               JoalAudioFactory.init() line: 217   

               DefaultAudioManager.init() line: 140         

               AudioTableAction.lambda$0(AudioManager) line: 122        

               2002295261.accept(Object) line: not available      

               Optional<T>.ifPresent(Consumer<? super T>) line: 159       

               AudioTableAction.createModel() line: 120              

               AudioTableAction.getPanel() line: 137       

               ListedTableFrame$TabbedTableItem.addPanelModel() line: 420

               ListedTableFrame$TabbedTableItem.<init>(String, String,
boolean) line: 377             

               ListedTableAction$1$1(ListedTableFrame).initComponents()
line: 117          

               ListedTableAction$1.run() line: 68              

               Thread.run() line: 745

bobjacobsen, 2017-08-11:
See #3880 for a failed attempt at fixing this.

klk32003, 2017-08-14:
@bobjacobsen Thanks for the attempt in #3880. 

I'm still convinced that ```-Djogamp.gluegen.UseTempJarCache=false``` is the right workaround today to resolve the JOAL hang-at-start problem under Windows. Two of three JMRIusers concerned with this problem have confirmed that. They are using the JVM option via ```jmri.conf``` individually.

I do admit that I cannot explain for now why #3880 failed. What I'd like to mention is that a change in ```LaunchJMRI.nsi``` needs a recreation of ```LaunchJMRI.exe``` (as long as PR #3919 isn't be merged). But this doesn't explain the failures.

bobjacobsen, 2017-08-14:
> I'm still convinced that -Djogamp.gluegen.UseTempJarCache=false is the right workaround today to resolve the JOAL hang-at-start problem under Windows. Two of three JMRIusers concerned with this problem have confirmed that. They are using the JVM option via jmri.conf individually.

Is it possible that option only works on Windows?  The problems I saw were on Travis CI, which runs Linux.  The AppVeyor (Windows) tests were fine.

KenC57, 2017-08-14:
I've got 4 systems that it has helped on Windows. All were very slow waiting
for JOEL to setup, adding the option cured that.

 

I've not had the slow issue on any Linux systems I work with. Any chance the
format of the option changes between systems or where (relative placement)
might impact some systems??

-ken c

klk32003, 2017-08-14:
> Is it possible that option only works on Windows? The problems I saw were on Travis CI, which runs Linux. The AppVeyor (Windows) tests were fine.

I don't think so. The syntax may differ. The JOAL hang-at-start problem only occurs on Windows systems.

rhwood, 2017-08-15:
Can someone running Windows confirm (by setting the debug logging for jmri.jmrit.audio.DefaultAudioManager) that JOAL audio is enabled if the option ```-Djogamp.gluegen.UseTempJarCache=false``` is used? I'm beginning to suspect that option might be simply disabling JOAL audio and forcing JMRI to using Javax Sound instead.

klk32003, 2017-08-15:
Thanks for looking on this.
Here is the session.log. I used jmri.conf to disable the cache. Don't know how to document that the cache wasn't created.

```2017-08-15 12:24:38,938 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-08-15 12:24:38,954 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Klaus\JMRI\log\messages.log [main]
2017-08-15 12:24:38,954 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Klaus\JMRI\log\session.log [main]
2017-08-15 12:24:38,969 apps.Apps                             INFO  - InstallTest version 4.9.2+Rff24066 starts under Java 1.8.0_144 on Windows 7 x86 v6.1 at Tue Aug 15 12:24:38 CEST 2017 [main]
2017-08-15 12:24:41,294 apps.Apps                             INFO  - Starting with profile Meine_JMRI_Bahngesellschaft.3eece4ab [main]
2017-08-15 12:24:41,481 node.NodeIdentity                     INFO  - Using jmri-94DE80D43F51-3eece4ab as the JMRI Node identity [main]
2017-08-15 12:24:41,980 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-8]
2017-08-15 12:24:42,308 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2017-08-15 12:24:42,308 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Klaus\JMRI\Meine_JMRI_Bahngesellschaft\ [main]
2017-08-15 12:24:42,308 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Klaus\JMRI\Meine_JMRI_Bahngesellschaft\ [main]
2017-08-15 12:24:42,308 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Klaus\JMRI\ [main]
2017-08-15 12:24:42,308 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Klaus\ [main]
2017-08-15 12:24:42,308 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2017-08-15 12:24:46,878 vsdecoder.VSDecoderPreferences        INFO  - Did not find VSDecoder preferences file.  This is normal if you haven't save the preferences before [AWT-EventQueue-0]
2017-08-15 12:24:46,894 audio.DefaultAudioManager             DEBUG - Try to initialise JoalAudioFactory [AWT-EventQueue-0]
2017-08-15 12:24:47,112 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [AWT-EventQueue-0]
2017-08-15 12:24:47,112 audio.DefaultAudioManager             DEBUG - sysName: IAL$ userName: Default Audio Listener [AWT-EventQueue-0]
2017-08-15 12:24:47,112 audio.DefaultAudioManager             DEBUG - Initialised AudioFactory type: JoalAudioFactory [AWT-EventQueue-0]
2017-08-15 12:25:06,564 audio.DefaultAudioManager             INFO  - Shutting down active AudioFactory [AWT-EventQueue-0]
2017-08-15 12:25:06,725 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

rhwood, 2017-08-17:
@klk32003 what is the audio device attached to your system and do you have ALSOFT's software-based OpenAL implementation installed separately from JMRI on your system?

klk32003, 2017-08-17:
> @klk32003 what is the audio device attached to your system and do you have ALSOFT's software-based OpenAL implementation installed separately from JMRI on your system?

For the above session.log there was no audio device attached and no ALSOFT available for JMRI other than ```\lib\windows\x86\soft_oal.dll```, which produced the string "ALSOFT 1.15.1" .

For sound tests I use a Creative Soundblaster audio device and ALSOFT 1.18.1, but on another machine.

klk32003, 2017-08-17:
I'd like to summarize a bit, just to keep track of this for myself.

1) There are some few cases where JMRI Audio (based on JogAmp-JOAL) doesn't startup as expected. This happens on Windows systems only.

2) Someone detected a JogAmp-process as source of the problem, while creating a TempJarCache

3) The JogAmp developer team is aware of this problem and recommend to ...
 3a) switch from ```Automated Native Library Loading``` to ```Traditional Native Library Loading```
 3b) disable the TempJarCache

4) To disable the TempJarCache a JMRI user can use ```jmri.conf``` with the line ```-J-Djogamp.gluegen.UseTempJarCache=false```

JMRI is already using ```Traditional Native Library Loading``` (at least on Windows). The TempJarCache, designed for ```Automated Native Library Loading``` and created by default, is never used on a JMRI Windows machine. Hence my conclusion is to disable the TempJarCache, as a workaround.
At least this is my understanding.

bobjacobsen, 2017-08-17:
To add a bit to that summary:

5. PR #3880 attempted to add a `-Djogamp.gluegen.UseTempJarCache=false` line to all the build and startup scripts.  That PR was never merged because although it passed Windows CI on AppVeyor, it caused Linux CI on Travis to consistently fail to load during audio tests.  It's not clear why.

6. I think this all means that we _should_ add the `-Djogamp.gluegen.UseTempJarCache=false` to Windows build and startup process, but not add it for other platforms.  That's easy for the startup scripts we distribute in releases because they're separate from each other. It's harder with e.g. the files for Ant, Maven, NetBeans, Eclipse, etc. Those are not (currently) fully separated, and we don't have a strong maintainer community for some of them.

7. And in turn I think that all means that we should build the (equivalent of) `-Djogamp.gluegen.UseTempJarCache=false` into the code itself:  JMRI should set that property, if it detects that it's running on Windows, before or during the launch of the Audio support.

closed 2021-04-14 by stale[bot]