JMRI/JMRI #2733: How to bring in PJC changes (#936)

mattharris, 2016-12-20:
>> #936 was last updated from JMRI/master on 12 Dec 2016. 4.5.7 was branched from master before that, and I don't know which changes between 4.5.7 and 12 Dec 2016 in master would be in the release (I think most were unit testing changes).
> 
> Thanks! I think both the following are straight-forward:
> 
> - A test release that’s v4.6 and #936
> - A test release that’s head-of-master and #936, either before, after or simultaneous with a regular head-of-master test release
> 
> I’m leaning toward 4.6+#936 because it’ll be most reliable on _other_ things.
> 
> Then
> 
> - put that out as 4.7.1, and decide in a month about whether 4.7.2 branches from that (but that delays a bunch of other change)
> - put that out as a special release, e.g. 4.7.0 which we label as a special test which we’d like our friends to help us with, and put out an all-contents 4.7.1 at the same time
> 
> I’m leaning toward the second of those.  I’d be happy to build and release Wednesday/Thursday this week. I think early is good, as people might have time to test over the next 10 days.
> 
> Should we move this to JMRIdevelopers?

mattharris, 2016-12-20:
Continuing discussion from #2699

mattharris, 2016-12-20:
I would also lean towards the later, 4.7.0 plus 4.7.1

rhwood, 2017-01-13:
We have some users reporting issues with SerialIO and with RXTX. Can we get 4.7.0 out to see if their issues are addressed?

bobjacobsen, 2017-01-13:
> We have some users reporting issues with SerialIO and with RXTX. Can we get 4.7.0 out to see if their issues are addressed?

Could they try the PJC-branch installers [built by Jenkins}(http://jmri.tagadab.com/jenkins/job/TestReleases/job/alpha-builds/job/rxtx-to-pjc/)?

rhwood, 2017-01-13:
I'd forgotten about that. Thanks.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

rhwood, 2017-02-27:
Because the existing PJC branch is *not* based off 4.6, and was last merged off master while master was likely in some not-right state at the time of the last merge, we should probably delay this until after 4.7.1, or even 4.7.2.

I had been testing with a "known to work with PJC" profile to verify it wasn't broken, but that was too simple to catch that Preferences are just broken for some reason that I can't see.

bobjacobsen, 2017-07-03:
PJC was merged into the 4.7.* series, so am closing this.

closed 2017-07-03 by bobjacobsen