JMRI/JMRI #2746: Decoder Pro 4.6 using Spog 11V3 WIthrottle not Working

peterriley2, 2016-12-22:
> As requested
I would be interested in some assistance
> Downloaded new production version of JMRI 4.6
> I'm using windows 10 and a new sprog11V3 
> I am struggling to use WIThrottle am I doing some thing wrong?
> I get am error on both the iPhone and iPad
> Network has disconnected, trying to reconnect
> The standard throttle is working and the loco operates as required

Comments from Randel

I see from the information you posted that JMRI cannot determine its own address.

You may find that either disabling IPv6 in Windows networking configuration or disabling the firewall or both will resolve your issue. If disabling the firewall resolves your issue, re-enable it, but make exceptions for ports 12080 and 12090, the JMRI web and WiThrottle ports.

Tried al this and still the same
> Here is the log file

JMRI Version: 4.6-R81496dc   
JMRI configuration file name: null   
JMRI main window name: DecoderPro: All Entries   
JMRI Application: DecoderPro   
Connection 0: SPROG DCC connected via SPROG Command Station on COM3 Disabled false   
Available Communication Ports:
 Found 1 serial ports
 Port: COM3 - in use by: JMRI app   
Active profile: Sprog Command   
Profile location: C:\Users\Peter Riley\JMRI\Sprog_Command   
Profile ID: Sprog_Command.3eb9fa78   
Preferences directory: C:\Users\Peter Riley\JMRI\Sprog_Command\   
Program directory: C:\Program Files\JMRI   
Roster index location: C:\Users\Peter Riley\JMRI\Sprog_Command\roster.xml   
Current panel file: [none]   
Audio factory type: [not initialised]   
java.version: 1.8.0_111   
java.vendor: Oracle Corporation   
java.home: C:\Program Files\Java\jre1.8.0_111   
java.vm.version: 25.111-b14   
java.vm.vendor: Oracle Corporation   
java.vm.name: Java HotSpot(TM) Client VM   
java.specification.version: 1.8   
java.specification.vendor: Oracle Corporation   
java.specification.name: Java Platform API Specification   
java.class.version: 52.0   
java.class.path: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.0.6.jar;lib\jackson-core-2.0.6.jar;lib\jackson-databind-2.0.6.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbee-api-0.9.jar;lib\xbjlib-1.1.1.jar;lib\xercesImpl.jar   
java.library.path: .;lib\x86;lib   
java.compiler: null   
java.ext.dirs: C:\Program Files\Java\jre1.8.0_111\lib\ext;C:\WINDOWS\Sun\Java\lib\ext   
file.encoding: UTF-8   
os.name: Windows 10   
os.arch: x86   
os.version: 10.0   
python.home: C:\Users\Peter Riley\JMRI\systemfiles   
python.path: null   
python.cachedir: null   
python.cachedir.skip: null   
python.startup: null   
user.name: Peter Riley   
user.home: C:\Users\Peter Riley   
user.dir: C:\Program Files\JMRI   
user.country: US   
user.language: en   
user.timezone: Europe/London   
jmri.log.path: C:\Users\Peter Riley\JMRI\log\   
Screen size h:864, w:1152 Inset t:0, b:40; l:0, r:0
Screen resolution: 96
Environment max bounds: java.awt.Rectangle[x=0,y=0,width=1152,height=824]
bounds[0] = java.awt.Rectangle[x=0,y=0,width=1152,height=864]
bounds[1] = java.awt.Rectangle[x=0,y=0,width=1152,height=864]
bounds[2] = java.awt.Rectangle[x=0,y=0,width=1152,height=864]
bounds[3] = java.awt.Rectangle[x=0,y=0,width=1152,height=864]
bounds[4] = java.awt.Rectangle[x=0,y=0,width=1152,height=864]
bounds[5] = java.awt.Rectangle[x=0,y=0,width=1152,height=864]
Device: \Display0 bounds = java.awt.Rectangle[x=0,y=0,width=1152,height=864] sun.awt.Win32GraphicsConfig@1647f75[dev=Win32GraphicsDevice[screen=0],pixfmt=2]
JmriInsets t:0, b:40; l:0, r:0
Network Interface: eth1
 Long Name: Intel(R) 82562V-2 10/100 Network Connection
 Host Name: PeterRiley-PC
 IP address: 192.168.1.2
ZeroConfService host: PeterRiley-PC running 1 service(s)
ZeroConfService: My JMRI Railroad._withrottle._tcp.local.  
 Name: My JMRI Railroad   
 Address: [unknown due to NPE]
 Port: 12090   
 Server: PeterRiley-PC.local.   
 Type: _withrottle._tcp.local.   
 URL: [unknown due to NPE]
 Published: yes

rhwood, 2016-12-23:
@peterriley2 Thanks for reporting this. When you run JMRI, it creates a file called session.log in C:\Users\Peter Riley\JMRI\log\ Can you attach that file to this issue or copy and paste its contents into a comment on this issue?

I think the clue to understanding what is happening is in that log based on these statements from what you have already provided:
```
ZeroConfService: My JMRI Railroad._withrottle._tcp.local.
Name: My JMRI Railroad
Address: [unknown due to NPE]
Port: 12090
Server: PeterRiley-PC.local.
Type: _withrottle._tcp.local.
URL: [unknown due to NPE]
Published: yes
```
where JMRI cannot figure out the address.

peterriley2, 2016-12-23:
Randel
Many thanks for the reply and directing me to where the log file is located
Here is the log file as I have just repeated the same process again
2016-12-20 14:32:42,150 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter Riley\JMRI\ [main]
2016-12-20 14:32:42,150 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter Riley\ [main]
2016-12-20 14:32:42,150 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-12-20 14:32:42,384 util.prefs                            WARN  - Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5. [Thread-5]
2016-12-20 14:53:19,191 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
2016-12-20 14:53:21,425 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-12-20 14:53:21,441 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter Riley\JMRI\log\messages.log [main]
2016-12-20 14:53:21,441 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter Riley\JMRI\log\session.log [main]
2016-12-20 14:53:21,457 apps.AppsBase                         INFO  - DecoderPro version 4.6-R81496dc starts under Java 1.8.0_111 on Windows 10 x86 v10.0 at Tue Dec 20 14:53:21 GMT 2016 [main]
2016-12-20 14:53:33,082 profile.ProfileManagerDialog          INFO  - Automatically starting with profile Programming_Track.3e599da2 after timeout. [AWT-EventQueue-0]
2016-12-20 14:53:33,238 gui3.Apps3                            INFO  - Starting with profile Programming_Track.3e599da2 [main]
2016-12-20 14:53:33,550 node.NodeIdentity                     INFO  - Using jmri-001AA092EFFF-3e599da2 as the JMRI Node identity [main]
2016-12-20 14:53:34,140 serialdriver.SerialDriverAdapter      INFO  - COM3 port opened at 9600 baud, sees  DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2016-12-20 14:53:40,218 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-12-20 14:53:40,218 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter Riley\JMRI\Programming_Track\ [main]
2016-12-20 14:53:40,218 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter Riley\JMRI\Programming_Track\ [main]
2016-12-20 14:53:40,281 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter Riley\JMRI\ [main]
2016-12-20 14:53:40,281 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter Riley\ [main]
2016-12-20 14:53:40,281 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-12-20 14:53:41,765 util.prefs                            WARN  - Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5. [Thread-6]
2016-12-20 14:56:18,484 throttle.AddressPanel                 INFO  - Asked for 3 got 3(S) [AWT-EventQueue-0]
2016-12-20 14:56:26,922 sprog.SprogCommandStation             INFO  - Changing currentSprogAddress (for pseudo-idle packets) to 1(L) [AWT-EventQueue-0]
2016-12-20 15:00:06,734 jmri.NmraPacket                       ERROR - invalid address 15227 [AWT-EventQueue-0]
2016-12-20 15:00:06,734 symbolicprog.CvValue                  WARN  - Exception during write CV '17' to '223' [AWT-EventQueue-0]
java.lang.NullPointerException
 at jmri.jmrix.sprog.SprogSlot.addressFromPacket(SprogSlot.java:457)
 at jmri.jmrix.sprog.SprogSlot.getAddr(SprogSlot.java:377)
 at jmri.jmrix.sprog.SprogCommandStation.notifySlotListeners(SprogCommandStation.java:365)
 at jmri.jmrix.sprog.SprogCommandStation.opsModepacket(SprogCommandStation.java:297)
 at jmri.jmrix.sprog.SprogOpsModeProgrammer.writeCV(SprogOpsModeProgrammer.java:53)
 at jmri.jmrix.AbstractProgrammer.writeCV(AbstractProgrammer.java:97)
 at jmri.jmrit.symbolicprog.CvValue.write(CvValue.java:536)
 at jmri.jmrit.symbolicprog.LongAddrVariableValue.writeAll(LongAddrVariableValue.java:292)
 at jmri.jmrit.symbolicprog.LongAddrVariableValue.writeChanges(LongAddrVariableValue.java:256)
 at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.executeWrite(PaneProgPane.java:858)
 at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.nextWrite(PaneProgPane.java:1149)
 at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.writePaneChanges(PaneProgPane.java:1083)
 at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane$3.itemStateChanged(PaneProgPane.java:379)
 at javax.swing.AbstractButton.fireItemStateChanged(Unknown Source)
 at javax.swing.AbstractButton$Handler.itemStateChanged(Unknown Source)
 at javax.swing.DefaultButtonModel.fireItemStateChanged(Unknown Source)
 at javax.swing.JToggleButton$ToggleButtonModel.setSelected(Unknown Source)
 at javax.swing.JToggleButton$ToggleButtonModel.setPressed(Unknown Source)
 at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
 at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
 at java.awt.Component.processMouseEvent(Unknown Source)
 at javax.swing.JComponent.processMouseEvent(Unknown Source)
 at java.awt.Component.processEvent(Unknown Source)
 at java.awt.Container.processEvent(Unknown Source)
 at java.awt.Component.dispatchEventImpl(Unknown Source)
 at java.awt.Container.dispatchEventImpl(Unknown Source)
[... cut ...]
 at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
 at jmri.jmrit.symbolicprog.VariableValue.setState(VariableValue.java:302)
 at jmri.jmrit.symbolicprog.LongAddrVariableValue.propertyChange(LongAddrVariableValue.java:352)
 at java.beans.PropertyChangeSupport.fire(Unknown Source)
 at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
 at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
 at jmri.jmrit.symbolicprog.CvValue.setState(CvValue.java:

dheap, 2016-12-23:
> 2016-12-20 14:56:18,484 throttle.AddressPanel                 INFO  - Asked for 3 got 3(S) [AWT-EventQueue-0]

The user has opened a Throttle for short address 3
 
> 2016-12-20 14:56:26,922 sprog.SprogCommandStation             INFO  - Changing currentSprogAddress (for pseudo-idle packets) to 1(L) [AWT-EventQueue-0]

The SPROG does not support IDLE packets so we have to keep the track alive by sending speed 0 commands to an unused address. The command station code changes this address as needed. It now avoids long address 0 and also the complementary version of any address (e.g. long address 3 if short address 3 is in use) after it was found that some Hornby decoders apparently confuse the two.

> 2016-12-20 15:00:06,734 jmri.NmraPacket                       ERROR - invalid address 15227 [AWT-EventQueue-0]

Address 15227 is the result of a Read Sheet of a roster entry using a Simulator connection (returns 123 in every byte).

What follows below indicates that the user has attempted to use Program on Main to change the long address of the decoder using a roster entry that was saved with the invalid 15227 address.

Dave

> 2016-12-20 15:00:06,734 symbolicprog.CvValue                  WARN  - Exception during write CV '17' to '223' [AWT-EventQueue-0]
> java.lang.NullPointerException
>  at [...]

peterriley2, 2016-12-24:
Dave

Thank you for your comments 

On the 20 December as you have explained the user using the system and made wrong selections

I was trying to set up a new loco and decoder from new 

Therefore the new decoder is set up as standard short address 03. Then I wanted to change to long address 8019. I had not even attempted to use WiThrottle 

If you follow the log further down to December 22 and 23. I am using address 8019 to operate a standard throttle without an issue then select WiThrottle 

This is where I have an problem or maybe not?

Peter

peterriley2, 2016-12-24:
Dave

I have just review the d

2016-12-21 12:11:29,424 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-12-21 12:11:29,580 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter Riley\JMRI\log\messages.log [main]
2016-12-21 12:11:29,580 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter Riley\JMRI\log\session.log [main]
2016-12-21 12:11:29,611 apps.AppsBase                         INFO  - DecoderPro version 4.6-R81496dc starts under Java 1.8.0_111 on Windows 10 x86 v10.0 at Wed Dec 21 12:11:29 GMT 2016 [main]
2016-12-21 12:11:33,408 gui3.Apps3                            INFO  - Starting with profile Programming_Track.3e599da2 [main]
2016-12-21 12:11:33,893 node.NodeIdentity                     INFO  - Using jmri-001AA092EFFF-3e599da2 as the JMRI Node identity [main]
2016-12-21 12:11:38,565 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-12-21 12:11:38,565 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter Riley\JMRI\Programming_Track\ [main]
2016-12-21 12:11:38,565 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter Riley\JMRI\Programming_Track\ [main]
2016-12-21 12:11:38,565 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter Riley\JMRI\ [main]
2016-12-21 12:11:38,565 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter Riley\ [main]
2016-12-21 12:11:38,565 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-12-21 12:11:39,565 util.prefs                            WARN  - Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5. [Thread-4]
2016-12-21 12:12:22,862 withrottle.UserInterface              INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleGUIServer]
2016-12-21 12:19:45,090 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
2016-12-21 12:19:51,293 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-12-21 12:19:51,308 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter Riley\JMRI\log\messages.log [main]
2016-12-21 12:19:51,308 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter Riley\JMRI\log\session.log [main]
2016-12-21 12:19:51,324 apps.Apps                             INFO  - PanelPro version 4.6-R81496dc starts under Java 1.8.0_111 on Windows 10 x86 v10.0 at Wed Dec 21 12:19:51 GMT 2016 [main]
2016-12-21 12:19:51,808 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3e87c222 [main]
2016-12-21 12:19:51,949 node.NodeIdentity                     INFO  - Using jmri-001AA092EFFF-3e87c222 as the JMRI Node identity [main]
2016-12-21 12:19:55,543 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-8]
2016-12-21 12:19:56,605 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-12-21 12:19:56,605 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter Riley\JMRI\My_JMRI_Railroad\ [main]
2016-12-21 12:19:56,605 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter Riley\JMRI\My_JMRI_Railroad\ [main]
2016-12-21 12:19:56,605 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter Riley\JMRI\ [main]
2016-12-21 12:19:56,605 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter Riley\ [main]
2016-12-21 12:19:56,605 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-12-21 12:19:57,027 util.prefs                            WARN  - Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5. [init prefs]
2016-12-21 12:25:30,045 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
2016-12-21 12:25:33,842 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-12-21 12:25:33,858 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter Riley\JMRI\log\messages.log [main]
2016-12-21 12:25:33,858 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter Riley\JMRI\log\session.log [main]
2016-12-21 12:25:33,858 apps.Apps                             INFO  - PanelPro version 4.6-R81496dc starts under Java 1.8.0_111 on Windows 10 x86 v10.0 at Wed Dec 21 12:25:33 GMT 2016 [main]
2016-12-21 12:25:34,233 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3e87c222 [main]
2016-12-21 12:25:34,389 node.NodeIdentity                     INFO  - Using jmri-001AA092EFFF-3e87c222 as the JMRI Node identity [main]
2016-12-21 12:25:37,545 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-8]
2016-12-21 12:25:38,452 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-12-21 12:25:38,452 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter Riley\JMRI\My_JMRI_Railroad\ [main]
[... cut ...]
2016-12-23 16:16:34,049 throttle.AddressPanel                 INFO  - Asked for 8019 got 8019(L) [AWT-EventQueue-0]
2016-12-23 16:16:36,580 sprog.SprogCommandStation             INFO  - Changing currentSprogAddress (for pseudo-idle packets) to 1(L) [AWT-EventQueue-0]
2016-12-23 16:16:53,658 throttle.AddressPanel                 INFO  - Asked for 8019 got 8019(L) [AWT-EventQueue-0]
2016-12-23 16:17:51,768 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

peterriley2, 2016-12-24:
Dave,  Randel

apologies just reviewed the log information I up loaded yesterday. The information around Withrottle was missing from the information I uploaded. I am also having trouble with the internet 
really appreciate you assistance. 
Peter

peterriley2, 2016-12-24:
wrong button

dheap, 2016-12-24:
Sorry for that , this is a problem when you are looking at a long messages.log rather than session.log - I didn't look at the dates.

A good idea to edit out the older stuff before posting in this case.

Dave

> On 24 Dec 2016, at 11:09 AM, peterriley2 <notifications@github.com> wrote:
> 
> On the 20 December as you have explained the user using the system and made wrong selections
> 
> I was trying to set up a new loco and decoder from new 
> 
> Therefore the new decoder is set up as standard short address 03. Then I wanted to change to long address 8019. I had not even attempted to use WiThrottle 
> 
> If you follow the log further down to December 22 and 23. I am using address 8019 to operate a standard throttle without an issue then select WiThrottle

peterriley2, 2016-12-24:
Dave

No worries I have a long way to go in learning

I have edited the JMRI log file into the issue raised on the web site as Randell had requested 

Peter

danielb987, 2020-08-19:
I'm closing this issue since there has been no comment on the problem for over three years.

closed 2020-08-19 by danielb987