JMRI/JMRI #2747: Current master DecoderPro does not load roster from absolute path

pabender, 2016-12-23:
The in the current master, I can't load my roster.

The roster has been (for a very long time) at (on a linux box)

/home/Paul/.jmri/roster

The path appears in preferences just fine, but there is no data loaded into the roster table (even after rebuilding the roster index).

This is shared among several profiles, and none of the profiles will load the roster entries.

rhwood, 2016-12-23:
What does the line starting with ```jmri-jmrit-roster.directory=``` read in the file ```profile/profile.properties``` in the profiles that cannot read the roster?

pabender, 2016-12-23:
On 12/23/2016 04:49 AM, Randall Wood wrote:
> What does the line starting with |jmri-jmrit-roster.directory=| read in
> the file |profile/profile.properties| in the profiles that cannot read
> the roster?

bash-4.3$ cat ~/.jmri/Z21SimulatorTest/profile/profile.properties |grep 
jmri-jmrit-roster.directory
jmri-jmrit-roster.directory=settings\:roster/

And I explicitly set the value on that profile last night.

The preferences correctly show this as /home/Paul/.jmri/roster/

Paul

rhwood, 2016-12-23:
I am unable to recreate this problem in JMRI/master on my computer, using rosters:
- in a profile
- in settings:
- in settings:foo
- in $HOME/.foo/bar

Do the other profiles have different values? Do any of the roster files fail a schema validation?

pabender, 2016-12-23:
Ok, I double checked that profile.  It turns out I had selected a roster.xml file in the roster directory, and not the roster.xml at the same level as the roster directory.

Closing.

closed 2016-12-23 by pabender