JMRI/JMRI #2748: Warning Messages in 4.6

WaltMazzei, 2016-12-23:
Each time I start up JMRI I receive many warning messages in my console file regarding two features in my panel.  I have 1 edge connector and one double-slip turnout.  Both seem to work seamlessly both on my layout and in simulation mode.  I reported this on the Yahoo site and after investigation, I was told that if the issue remained after the recent update, I should open an issue here.  See my problems folder (Walt M) on the Yahoo folder.   I have upgraded to version 4.6 and the warnings remain.  In addition I now see INFO messages about missing route information.  As I said, I have a great JMRI system running with no physical problems to report.  But these messages continue to give me doubts that the JMRI system is working as it should.  I believe that these warnings should either point to a problem I can address or be removed from JMRI console. 
Walt Mazzei
Packer47@att.net
[MMTRR system Dump File JMRI4.6 Dec23.docx](https://github.com/JMRI/JMRI/files/670755/MMTRR.system.Dump.File.JMRI4.6.Dec23.docx)

bobjacobsen, 2016-12-25:
@WaltMazzei: Could you attach your Layout Editor panel please?  I'd like to try to recreate this on my own machine to debug it.

Multiply repeated error messages (from the .docx file attached above):
```
layoutEditor.LayoutEditorAuxTools     WARN  - Did not decode ts.getType2() of 1 (A) [main]

layoutEditor.LayoutEditorAuxTools     WARN  - possible incomplete code: At a double slip we could go in two different directions [main]
```

Plus some specific messages, all INFO from layoutEditor.LayoutBlock:
```
From B-LS656, no routes returned for getRouteByNeighbor(B-LS655) [main]
From B-LS651, no routes returned for getRouteByNeighbor(B-LS653) [main]
From B-LS650, no routes returned for getRouteByNeighbor(B-LS652) [main]
From B-LS655, no routes returned for getRouteByNeighbor(B-LS654) [main]
From B-LS659, no routes returned for getRouteByNeighbor(B-LT345) [main]
From B-NL-Yard, no routes returned for getRouteByNeighbor(B-NL-North) [main]
From B-LS644, no routes returned for getRouteByNeighbor(B-LS643) [main]
From B-NL-Yard, no routes returned for getRouteByNeighbor(B-NL-Engine) [main]
From B-Laredo-Import-Lead, no routes returned for getRouteByNeighbor(B-LT292) [main]
From B-NL-Yard, no routes returned for getRouteByNeighbor(B-NL-South) [main]
From B-LS605, no routes returned for getRouteByNeighbor(B-LS606) [main]
From B-LT200, no routes returned for getRouteByNeighbor(B-LT210) [main]
From B-LT200, no routes returned for getRouteByNeighbor(B-LT210) [main]
From B-LS623, no routes returned for getRouteByNeighbor(B-LT210) [main]
From B-Staging5, no routes returned for getRouteByNeighbor(B-StagingIN) [main]
From B-Staging5, no routes returned for getRouteByNeighbor(B-StagingIN) [main]
From B-LS671, no routes returned for getRouteByNeighbor(B-LS670) [main]
From B-LT292, no routes returned for getRouteByNeighbor(B-LS657) [main]
```

From a quick look at the code, the first are perhaps over-zealous diagnostic info.  Not sure about the 2nd group.

WaltMazzei, 2016-12-26:
I have attached a zip file of my panel plus the signal and resource 
files needed to run the panel.

Thank you for looking into this.  Greatly appreciated.  If I can help, 
Please let me know how.

Merry Christmas

Walt

On 12/25/2016 12:02 AM, Bob Jacobsen wrote:
>
> @WaltMazzei <https://github.com/WaltMazzei>: Could you attach your 
> Layout Editor panel please? I'd like to try to recreate this on my own 
> machine to debug it.
>

geowar1, 2017-01-21:
I have five (5!) double slips on my layout and I get about a page of those errors; if I replace them with "Level crossings" these messages go away… So I suspect bugs in the slip code. I've already found/fixed(?) two bugs in ConnectivityUtil trackSegmentLeadsTo() SLIP_A-C case:
    
    PanelPro was spewing tons of "ERROR - Connectivity error when following
    track in Block Yard [MainThread]" on startup… I tracked it down to two
    bugs in the ConnectivityUtil trackSegmentLeadsTo() routine:
    
    In "case LayoutEditor.SLIP_B:" was testing "if (((TrackSegment)
    ls.getConnectB()).getLayoutBlock() == nlb) {"; that should be
    getConnectD (not getConnectB).
    
    The curObj variable wasn't getting set in the "if (conType >=
    LayoutEditor.SLIP_A && conType <= LayoutEditor.SLIP_D)" block (of code).

I'll look into the above (info? debug? error?) messages as soon as I fix the alltest build (I broke it… /shame).

bobjacobsen, 2017-09-20:
@geowar1: Did this get resolved with the recent changes?

WaltMazzei, 2017-09-20:
The situation is MUCH improved!  However, I'm still getting eight lines in my System Console reporting double slip turnout connection problems.  I have one double slip turnout on  my panel in the yard mail line track.  I have a second double slip in dark territory.  So I believe all 8 lines relate to the one main line doubleslip.
You can reach me at Packer47@att.net if you need more detail than I have already provided.
Thanks for asking!
Walt

geowar1, 2017-09-20:
> layoutEditor.LayoutEditorAuxTools     WARN  - Did not decode ts.getType2() of 1 (A) [main]

This is bad… send me the panel file and I'll see what's happening…

`layoutEditor.LayoutEditorAuxTools     WARN  - possible incomplete code: At a double slip we could go in two different directions [main]
`
This one is benign; IIRC I've downgraded that to just a debug. The same situation happens at the throat of any turnout… (but without the warning).

closed 2021-04-14 by stale[bot]