JMRI/JMRI #2751: Enabling "Identify" button and Programmer modes in DecoderPro needs to be revisited.

pabender, 2016-12-24:
I'm debugging an issue related to why the "Identify" button is not available in DecoderPro with a Roco Z21 Connection.

The Roco Z21 provides a programmer through its XPressNet tunnel.  The XPressNet tunnel appears in the defaults selection as a separate line from the Z21 connection, and it has a menu seperate from the Z21 menu for XPressNet specific operations.  The XPressNet tunnel cannot be configured separately and does not register a ConnectionConfig object because of that (plus, adding the connection config object causes a second. 

I've gone to great lengths to verify that the programmer is actually being registered properly, so there is a programmer available, but the "Identify" button remains disabled.

After tracing the problem, it turns out the issue is this code (from updateProgrammerStatus() in jmri.jmrit.roster.swing.RosterFrame. (with some added debugging statements that don't appear in master.): 

        // Find the connection that goes with the global programmer
        GlobalProgrammerManager gpm = InstanceManager.getNullableDefault(GlobalProgrammerManager.class);
        if (gpm != null) {
            String serviceModeProgrammerName = gpm.getUserName();
            log.debug("GlobalProgrammerManager found of class {} name {} ",gpm.getClass(),serviceModeProgrammerName);
            for (ConnectionConfig connection : InstanceManager.getDefault(ConnectionConfigManager.class)) {
                log.debug("Checking connection name {}", connection.getConnectionName());
                if (connection.getConnectionName() != null && connection.getConnectionName().equals(serviceModeProgrammerName)) {
                    log.debug("Connection found for GlobalProgrammermanager");
                    serModeProCon = connection;
                }
            }
        }

The problem here is that the code above assumes that all connections that register a programmer will also register a connection config object which isn't happening for the Roco Z21's XPressNet connection (again, it can't exist separately).  

The connection object is later used in a call to isConnectionOk in jmri.jmrix.ConnectionStatus to provide the port name for the connection.

So now I have two issues that are going to need to be addressed:
First, We need an alternate method of fetching the port name here.  Using the ConnectionConfig may work for most connections, but it doesn't work for all connections.

Second, It doesn't appear the ConnectionStatus handles port name collisions.  i.e. if I have two simulators configured (which use (none) as the port name) then the connection status for both connections is going to be the same.

I think the right way to handle both of these is to modify the ConnectionStatus so that it uses either the connection prefix or the connection name to provide the status report.  The connection name is retrieved from the programmer in the code above.  (and the Z21's XPressNet tunnel DOES report status using a port name currently).

Does anyone have any thoughts on this?  It's a rather major change, since implementing it will require touching all supported systems.

Paul

KenC57, 2016-12-24:
Paul,

 

I had thought I noticed some issues with multiple simulator connections, but
never checked too deep. Is what you found dependent on the two simulated
connections being of the same connection type or any 2 (or more) simulated
on a system??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-12-24:
On Dec 24, 2016, at 8:56 AM, Ken Cameron <notifications@github.com> wrote:
> I had thought I noticed some issues with multiple simulator connections, but
> never checked too deep. Is what you found dependent on the two simulated
> connections being of the same connection type or any 2 (or more) simulated
> on a system??

Because it is the port name that is registered, the problem is not restricted to connections of the same type.
> 

So, registering a C/MRI simulator and an XPressNet simulator and a LocoNet simulator, all 3 will report the same status because the all use (none) as the port name.

We also have the same problem with multiple network connections to the same server ( they connect to the same hostname and port).

Paul

rhwood, 2016-12-24:
Should not the code you referenced be asking the DefaultsManager for the global programmer instead of trying to determine its own separate from the DefaultsManager?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2016-12-24:
Sent from my iPhone

> On Dec 24, 2016, at 9:57 AM, Randall Wood <notifications@github.com> wrote:
> 
> Should not the code you referenced be asking the DefaultsManager for the global programmer instead of trying to determine its own separate from the DefaultsManager?

You are mis-reading that code.  It is not trying to determine its own default.

The first thing that code does is ask the InstanceManager for the default global programmer.

The rest of the code posted finds the connection associated with that programmer.  Later ( not in the snipit I posted ) the code checks to see if that connection is up or down.

If there is no programmer or there is a programmer but the connection is down, the identify button is disabled.  

There are listeners installed so that if a programmer ( or the connection it is associated with ) availability changes, the button will be enabled or disabled appropriately.

Paul

rhwood, 2016-12-24:
My question is "why is the InstanceManager being asked for that?" I think the DefaultsManager should know that better than the InstanceManager, since the InstanceManager is known to have some issues sorting that out since some connections's managers get registered as the default instance in the InstanceManager when created regardless of the settings in the DefaultsManager.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2016-12-24:
Sent from my iPhone

> On Dec 24, 2016, at 11:02 AM, Randall Wood <notifications@github.com> wrote:
> 
> My question is "why is the InstanceManager being asked for that?" I think the DefaultsManager should know that better than the InstanceManager, since the InstanceManager is known to have some issues sorting that out since some connections's managers get registered as the default instance in the InstanceManager when created regardless of the settings in the DefaultsManager.

My guess is this code predates the DefaultsManager.

Paul

pabender, 2016-12-25:
Another wrinkle here.

I've been doing experiments in test driven development as I've been working on things the last few weeks especially.  That means you write tests that show the desired behavior first, and then you modify the code.

So, in order to test the code for the Identify button, I've loaded a jmri.progdebugger.DebugProgrammerManager which is a valid Programmer, but it isn't associated with a connection at all.

So now I'm questioning why do we even bother to check and see if the connection is enabled if we have a valid programmer object?  Was there some user objection to this?

(With the further thought to this being that UDP services are technically connectionless, and we have at least two systems that use UDP based communication (The Roco Z21 and Marklin).

I can fake a connection in the ConnectionStatus class for purposes of the test if needed.  I'm just trying to determine the correct desired behavior.

Paul

pabender, 2016-12-29:
closed via #2767

closed 2016-12-29 by pabender