JMRI/JMRI #2753: Operations help and other help broken in 4.6

DanielBoudreau, 2016-12-24:
The operation help mapping is broken in 4.6.  From the console:

javax.help.BadIDException: Not valid ID: package.jmri.jmrit.operations.Operations_Settings
	at javax.help.Map$ID.create(Map.java:126)
	at javax.help.CSH.displayHelp(CSH.java:743)
	at javax.help.CSH.access$200(CSH.java:80)
	at javax.help.CSH$DisplayHelpFromSource.actionPerformed(CSH.java:1668)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

DanielBoudreau, 2016-12-24:
I tracked the problem to the Map.jhm file, it is missing over a hundred mapIDs.  This affects not only operations but other help indexes.

DanielBoudreau, 2016-12-24:
Went back and tested 4.5.5 and the help mapping is okay with that test release, but it was broken in 4.5.8.  I don't have 4.5.6 or 4.5.7 on my PC so I don't know exactly when the help mapping broke.

bobjacobsen, 2016-12-24:
The map file is regenerated for each release, so something has started going wrong in that.  Will check.

bobjacobsen, 2016-12-24:
JHelpDev is creating a truncated Map.jhm file, without reporting errors.  Arghh...

mattharris, 2016-12-24:
I think that the issue came in via #2611. And I now recall you seeing some similar issues as noted in #2505 which I hadn't fully realised the impact of at the time of #2611.

bobjacobsen, 2016-12-24:
`git bisect` struggled because the map file changes quite a bit as help files are changed, but the problem seems to have surfaced first in d285f676ff020507c72e48406af042e1957d81bb "run tidy -m over the rest of the help files", which made about 1/2 of the Map entries go away.

bobjacobsen, 2016-12-24:
The problem seems to be that JHelpDev used to create e.g.
```
<mapID target="html.hardware.XBee.index_hardware" url="html/hardware/XBee/index.shtml#hardware"/>
```
from 
```
<a name="hardware"></a><h2>Supported Hardware</h2>
```
but the `tidy -m` operation (or something else in that commit) converted that to
```
<a name="hardware" id="hardware"></a>
```
and  JHelpDev is no longer making the entry.

bobjacobsen, 2016-12-24:
The problem with the local anchors might not be the only problem (Daniel's description is a bit different), but it's one that's big enough that it's masking everything else. So I'm working on it first.

- Removing the `id="hardware"` attribute fixes the JHelpDev problem, but might cause other problems
- Removing the change to the DOCTYPE doesn't fix it.

bobjacobsen, 2016-12-24:
Found the problem.  In the JHelpDev source, `net.sourceforge.jhelpdev.action.CreateMapAction` is using regular expressions (!!) to parse the relevant HTML text.  It pulls the `<A>` tag into a string, then used `">.*<"` to parse for the contents (with DOTALL to handle line ends). That assumes there's only one quoted attribute value inside the tag (the `.*` is a greedy match).  Badness ensures. Changing that to the non-greedy `.*?` (which requires moving to JHelpDev version 0.29 to get a clean compile) fixes that.  Now testing...

bobjacobsen, 2016-12-24:
Also updating to JHelDev 0.69, which looks like the most recent (and, given activity, perhaps last) version.  This cleans up some nagging indexing problems, but doesn't fix the long-standing warning about
```
[Fatal Error] index.html:1:3: The markup declarations contained or pointed to by the document type declaration must be well-formed.
```

bobjacobsen, 2016-12-24:
This update fixes the map generation for #1431, but not the indexing problem; it has no effect on #1432

bobjacobsen, 2016-12-24:
I think that PR #2757 fixes this on master.  It's probably a big enough deal that we should publish a 4.6.1 (perhaps with other fixes?) see Issue #2755 (Create 4.6.1) and PR #2758 (help for 4.6.1 if needed)

DanielBoudreau, 2016-12-24:
Bob,

Thanks for fixing this.

Merry Christmas and Happy Hanukkah,

Dan

PR #2757 (2016-12-25): JHelpDev update and help metadata for master
Fixes #2753 on master
closed 2016-12-25 by bobjacobsen