JMRI/JMRI #2770: Need method to create empty profiles for tests

rhwood, 2016-12-27:
Need methods in JUnitUtil to create and activate (and destroy) a profile so things that rely on having a Profile do not need to be overwritten in tests.

pabender, 2016-12-27:
There is already a temp/profile directory created by some of the tests.  

The created profiles should probably end up there as well.

Paul

rhwood, 2016-12-27:
I wasn't thinking of providing three methods to:
- generate empty profile in random location
- use the known test location
- allow a profile specifically configured for a specific test to be specified

(Really the first two would generate or use a profile and call the last with that profile.)

pabender, 2016-12-27:
Sent from my iPhone
> On Dec 27, 2016, at 8:47 AM, Randall Wood <notifications@github.com> wrote:
> 
> I wasn't thinking of providing three methods to:
> 
> generate empty profile in random location
> use the known test location
> allow a profile specifically configured for a specific test to be specified
> (Really the first two would generate or use a profile and call the last with that profile.)
> 

I'm think that first sentence should have said was not wasn't.

I see very little, if any, value in creating a profile in a random location.

As a matter of testing protocol, randomness leads to uncertainty and a measure of unpredictability in the tests.  Also, it makes it harder to debug the tests.

Being able to find ( and clean up ) the data is the reason we use the temp location in other parts of the testing suite.

Paul

rhwood, 2016-12-28:
Fixed in #2772

closed 2016-12-28 by rhwood