JMRI/JMRI #2775: Issues with Line Endings (again).

dheap, 2016-12-28:
I noticed a `git pull` a couple of days ago made some line ending changes in my local repo. That always bodes ill.

 Sure enough, the problem bit today (just when I'm busy):
> Mac-Pro-2:jmri heap$ git pull
> From https://github.com/JMRI/JMRI
>    03970be..ff87d26  master     -> origin/master
> Updating 03970be..ff87d26
> error: Your local changes to the following files would be overwritten by merge:
> 	xml/demoOperations/Operations.xml
> 	xml/demoOperations/OperationsCarRoster.xml
> 	xml/demoOperations/OperationsEngineRoster.xml
> 	xml/demoOperations/OperationsLocationRoster.xml
> 	xml/demoOperations/OperationsRouteRoster.xml
> 	xml/demoOperations/OperationsTrainRoster.xml
> Please, commit your changes or stash them before you can merge.
> Aborting
> Mac-Pro-2:jmri heap$

I've resolved this somehow in the past, but can't remember the procedure. Tried a few things, but not helping.

Could someone please advise? 

Dave

dheap, 2016-12-28:
P.S. I have command line and NetBeans 8.2 at my disposal.

rhwood, 2016-12-28:
In NetBeans, right click on the xml/demoOperations directory and select Git->Stash Changes (if I recall correctly--I'm not a machine I can check on right now). Once you pull in the changes from master, you should be able to discard the stashed changes using the same menu.

dheap, 2016-12-28:
I did look at using `git stash save` and `git stash drop`from the command line but couldn't see how to restrict it to particular files or folders (I have many uncommitted changes on one of my machines I don't want to lose).

Is anyone able to help me selectively reset the the xml/demoOperations directory to resolve the issue so I can again pull from master?

rhwood, 2016-12-29:
The other reality is that if the line endings on these files are correct in your working copy (are LF characters) you can just commit the changes.

dheap, 2016-12-29:
@rhwood They were changed to CRLF by the older commit. I've tried changing them to LF, with no joy.

I don't  want to commit my other changes but if I just commit the demoOperations changes, that might work, but it may create a conflict with Master if there have been subsequent changes. I now recall doing this last time... I'll look into it//

dheap, 2016-12-29:
It seems the checklineends test we had doesn't work (or doesn't work anymore)...

dheap, 2016-12-29:
`git stash save` doesn't fix the problem. It simply restores the conflicting CRLF form.

dheap, 2016-12-29:
It seems when thing go wrong with Git, they go wrong in a nasty way. As a result of the bad PR #2764 being merged, I have ended up with irreconcilable conflicts in my local checkouts on several machines. On one machine I have a lot of uncommitted local changes in master I neither want to lose, nor to have to reconstruct manually.

If I were to follow Randall's suggestion of fixing the offending line endings and committing those files, where would I push the commit to?  https://github.com/dheap/JMRI ? https://github.com/JMRI/JMRI?

Then I've got to do a PR to reconcile this and I can see conflicts arising. And I've got to repeat the process for several checkouts on different machines...

Where do I go from here?

pabender, 2016-12-29:
@dheap,

I don't think that pull request was necessarily bad.

Since your local copy believes the files have changed,  have you tried just checking out the file again?

I don't know how to do this in netbeans, if that is what you're using, but if you run 'git checkout filename' on the command line for the offending file, that resets the file to the last known state of the file.

dheap, 2016-12-29:
@pabender #2764 incorrectly changed the line endings to CRLF. @rhwood subsequently fixed this as part of #2772 .

The problem is that I did a `git pull` on several machines between the merging of these two PRs. Because of the bad line endings, my local repository immediately marked those files as changed, despite my not having touched them.

As result, `git pull` now flags a conflict and won't let me merge. `git checkout filename` does not help either. `git stash` doesn't help either. They still appear as changed, despite being stashed. I've tried changing the endings to LF, reverting them using NetBeans and now `git checkout filename`.

Nothing I can do convinces Git that they are unmodified.

`git stash` complained when I tried to stash them with CRLF endings and changed them, but still to no avail. I can't find any way to get them back to unmodified status.

bobjacobsen, 2016-12-29:
> On one machine I have a lot of uncommitted local changes in master I neither want to lose, nor to have to reconstruct manually.

Have these been committed already?

dheap, 2016-12-29:
@bobjacobsen I have not committed anything, neither the bad operations files nor my modified/untracked files. Committing the modified/untracked files is inappropriate at this stage (there are quite a lot of them and are WIP). I could manually save all these outside my repository and restore them later, but that is a lot of work and I want to avoid that route if possible.
> Have these been committed already?

pabender, 2016-12-29:
On 12/29/2016 01:34 AM, Dave Heap wrote:
> @bobjacobsen <https://github.com/bobjacobsen> I have not committed
> anything, neither the bad operations files nor my modified/untracked
> files. Committing the modified/untracked files is [...]

You can commit files one at a time, you don't have to commit all open 
changes.

Commit the files you want to save to a branch and push that up to the 
same branch in your github repository.  After you do that, you can clean 
up your local repository as you see fit, and then check the branch out 
on your machine.

When the changes are ready, you can then make a PR against master from 
the branch.

Paul

dheap, 2016-12-29:
@pabender What you suggest is appropriate for those very familiar with branches and keeping them in sync with master. I'm afraid I'm not confident enough with branching to have all my WIP in branch(es), as is best practice.As a result I keep all my long-term WIP in master and only move relevant files to a new branch when I'm ready to do a PR.

Looks like I'll have to wear the consequences of my modus operandi and manually stash all my modified/untracked files elsewhere before resetting my repository to the latest HEAD of master. Will involve a lot of time and care.

But before I do that I need to know how to reset my checkouts completely so I can test it on the machine that has a clean checkout apart from the offending operations files (that I didn't touch). I'm not going to do all that work on my main machine only to find the reset doesn't work.

dheap, 2016-12-29:
Let's try the easy case first:

I had a completely clean checkout on my machine known as Trains. I had previously used it for #2743 and #2750 and subsequently brought it up-to-date with "git pull"s.

> Trains:jmri heap$ git branch
>   athearn-oem
>   eco-tsu2-productID
> \* master
> Trains:jmri heap$ 
> 

 On 27-Dec-1026 at 6:50am AEDT (+1100) I ran a "git pull" against a completely clean checkout. Here is the log:

> git pull
> From https://github.com/JMRI/JMRI
>    6f34dc9..9d4329e  master     -> origin/master
> Updating 6f34dc9..9d4329e
> Fast-forward
> warning: CRLF will be replaced by LF in xml/demoOperations/Operations.xml.
> The file will have its original line endings in your working directory.
> warning: CRLF will be replaced by LF in xml/demoOperations/OperationsCarRoster.xml.
> The file will have its original line endings in your working directory.
> warning: CRLF will be replaced by LF in xml/demoOperations/OperationsEngineRoster.xml.
> The file will have its original line endings in your working directory.
> warning: CRLF will be replaced by LF in xml/demoOperations/OperationsLocationRoster.xml.
> The file will have its original line endings in your working directory.
> warning: CRLF will be replaced by LF in xml/demoOperations/OperationsRouteRoster.xml.
> The file will have its original line endings in your working directory.
> warning: CRLF will be replaced by LF in xml/demoOperations/OperationsTrainRoster.xml.
> The file will have its original line endings in your working directory.
>  .../package/jmri/jmrit/operations/Operations.shtml |  14 +-
>  .../jmri/jmrit/operations/images/Routes.png        | Bin 51619 -> 48582 bytes
>  .../routes/JmritOperationsRoutesBundle.properties  |   2 +-
>  java/src/jmri/jmrit/operations/trains/Train.java   |  24 +--
>  xml/demoOperations/Operations.xml                  |  13 +-
>  xml/demoOperations/OperationsCarRoster.xml         | 140 +++++++-------
>  xml/demoOperations/OperationsEngineRoster.xml      |  40 ++--
>  xml/demoOperations/OperationsLocationRoster.xml    | 206 ++++++++++-----------
>  xml/demoOperations/OperationsRouteRoster.xml       |  29 +--
>  xml/demoOperations/OperationsTrainRoster.xml       |  59 +++---
>  10 files changed, 269 insertions(+), 258 deletions(-)

As a result I now have this:

> Trains:jmri heap$ git status
> On branch master
> Your branch is up-to-date with 'origin/master'.
> 
> Changes not staged for commit:
>   (use "git add <file>..." to update what will be committed)
>   (use "git checkout -- <file>..." to discard changes in working directory)
> 
> 	modified:   xml/demoOperations/Operations.xml
> 	modified:   xml/demoOperations/OperationsCarRoster.xml
> 	modified:   xml/demoOperations/OperationsEngineRoster.xml
> 	modified:   xml/demoOperations/OperationsLocationRoster.xml
> 	modified:   xml/demoOperations/OperationsRouteRoster.xml
> 	modified:   xml/demoOperations/OperationsTrainRoster.xml
> 
> no changes added to commit (use "git add" and/or "git commit -a")
> Trains:jmri heap$ 

and this:

> Trains:jmri heap$ git pull
> From https://github.com/JMRI/JMRI
>    03970be..3a127cf  master     -> origin/master
> Updating 03970be..3a127cf
> error: Your local changes to the following files would be overwritten by merge:
> 	xml/demoOperations/Operations.xml
> 	xml/demoOperations/OperationsCarRoster.xml
> 	xml/demoOperations/OperationsEngineRoster.xml
> 	xml/demoOperations/OperationsLocationRoster.xml
> 	xml/demoOperations/OperationsRouteRoster.xml
> 	xml/demoOperations/OperationsTrainRoster.xml
> Please, commit your changes or stash them before you can merge.
> Aborting
> Trains:jmri heap$ 

How do I fix the checkout on this machine?

rhwood, 2016-12-29:
In this case, commit the changes. When got pulls in an identical change, it will just quietly merge the two together and continue without issue.

dheap, 2016-12-29:
@rhwood But when I do a `git push` to  https://github.com/dheap/JMRI what will happen to that commit? Will it cause problems later?

rhwood, 2016-12-29:
> On Dec 29, 2016, at 6:17 AM, Dave Heap <notifications@github.com> wrote:
> 
> @rhwood <https://github.com/rhwood> But when I do a git push to https://github.com/dheap/JMRI <https://github.com/dheap/JMRI> what will happen to that commit? Will it cause problems later?
> 
I don’t think it will because the commit will merge into JMRI/master without conflicts. The worse case scenario will be that the git history will show two identical changes.

Randall

dheap, 2016-12-29:
@rhwood Past bedtime now. I'll look at it again tomorrow.

bobjacobsen, 2016-12-29:
Sorry to be coming to this late, will try to work on it while you're asleep, but one comment:

> Looks like I'll have to wear the consequences of my modus operandi and manually stash all my modified/untracked files elsewhere before resetting my repository to the latest HEAD of master. Will involve a lot of time and care.

Please DO NOT DO THIS!

Sorry to yell, but I wanted to make sure that was seen in a long & twisty discussion thread. 

I understand that the current problem is frustrating.  I understand that you intend to take "time and care" to get it right.

But the historical record is that people _don't_ get this right.  We've lost changes & introduced bugs several times because of people trying to do this, even experienced people trying to do it carefully.  Version control systems are designed to protect against those problems, and copying files around defeats that protection.

It's particularly dangerous, and therefore even more inappropriate, to be doing it in a repository on an account that has direct commit authority to our main GitHub repository.  

Again, I understand that this is frustrating and time-consuming, and I understand that you intend to take the time and care to do it right.  This is not a statement about your intentions or capability.  But some technical tasks are just too complex and error prone to be 100% sure they're right, and this is one of those.

So please don't do this.  Let's work to fix this right.

bobjacobsen, 2016-12-29:
> As a result I now have this:
 ```
Trains:jmri heap$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

Changes not staged for commit:
(use "git add ..." to update what will be committed)
(use "git checkout -- ..." to discard changes in working directory)

modified: xml/demoOperations/Operations.xml
modified: xml/demoOperations/OperationsCarRoster.xml
modified: xml/demoOperations/OperationsEngineRoster.xml
modified: xml/demoOperations/OperationsLocationRoster.xml
modified: xml/demoOperations/OperationsRouteRoster.xml
modified: xml/demoOperations/OperationsTrainRoster.xml

no changes added to commit (use "git add" and/or "git commit -a")
Trains:jmri heap$
```

I agree with Randall.  Try accepting those modifications and committing:

```
git add xml/demoOperations/*.xml
git commit -m"accepting line ends" xml/demoOperations/*.xml
```

If that commit goes through and eventually merges, it should be OK. We can test that if you push that up to your repository and see what a PR looks like.

If that commit fails, perhaps the error message will give us a next step.

dheap, 2016-12-30:
@bobjacobsen I have no intention of making a rash move. I am fully aware of the possible consequences of things. That is why i have not acted on any advice from @rhwood or @pabender yet. I need to get this right.

rhwood, 2016-12-30:
I've confirmed by submitting additional PRs with the identical changes, that they merge without issue. You are *not* being rash by committing changes that git is configured to essentially require be made.

dheap, 2016-12-30:
@rhwood @bobjacobsen The one thing that I omitted to mention is that the files in my working directory are unchanged from the original pull and have CRLF line endings.

Should I commit them in this state or should I change them to LF line endings before committing?

I will wait for both of you to respond before acting on this.

I have another checkout on a MacBook Pro that should be clean (but well out-of date and certainly preceding  #2764 and #2772). But I don't want to touch that unless needed. Just mentioning it FYI.

rhwood, 2016-12-30:
I had been under the assumption that since you use macs, the files had the correct line endings. Git should change them to be correct when you commit, but if you correct them using dos2unix, it's entirely possible git will decide they can merge cleanly (I.e. You won't need to commit them).

We can sometimes get into this state if a bad commit is made and propagated, and a later commit fixes that, but other working copies are updated in between the bad and the commit making it into JMRI/master.

dheap, 2016-12-30:
@rhwood They had the correct LF line endings immediately prior to the pull that brought in #2764. I retrieved a copy from my machine backups (to elsewhere) to verify this. #2764 sent them bad.

On my Mac Pro I tried changing the CRLF endings to LF (using TextWrangler) but they still flagged as modified and refused to merge (didn't think of dos2unix).

I'm not ready to try dos2unix on these as-yet-untouched (on Trains iMac) files yet.

rhwood, 2016-12-30:
Note that #2764 made substantive (not just line endings) changes to those files, so overwriting them from an earlier version is also wrong.

dheap, 2016-12-30:
I was aware of that. I had no intention of overwriting them.
> Note that #2764 made substantive (not just line endings) changes to those files, so overwriting them from an earlier version is also wrong.

KenC57, 2016-12-30:
For the record, when I get files (signal, decoder, etc..) from others, I
always use dos2unix on them before I bring them into my local repository. It
seems to save a lot of confusion in the system over 'what's changed' when I
do push them out.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dheap, 2016-12-30:
@KenC57 Certainly a good idea.

For the record, the bad line endings files were accidentally introduced by @DanielBoudreau in #2764 .

>For the record, when I get files (signal, decoder, etc..) from others, I
always use dos2unix on them before I bring them into my local repository. It
seems to save a lot of confusion in the system over 'what's changed' when I
do push them out.

dheap, 2016-12-30:
I've just realised my Macs don't have `dos2unix` installed on them.

bobjacobsen, 2016-12-30:
> The one thing that I omitted to mention is that the files in my working directory are unchanged from the original pull and have CRLF line endings.
>
> Should I commit them in this state or should I change them to LF line endings before committing?

Commit them as they are, please.

I just did a `git checkout` of one of the commits that shows the same situation on my Mac, then did `git add` and `git commit`.  It seems to have done the right thing.

DanielBoudreau, 2016-12-30:
Copied the files exactly how they were created by JMRI on my Window’s 7 PC.
Dan

rhwood, 2016-12-30:
The problem is that the built in Git client in Eclipse will not respect the rules in the JMRI .gitattributes file concerning how line endings are to be handled. JMRI developers on Windows really *need* to either manually verify they are checking in the code the way we want it to avoid these situations or to use a git client that works correctly.

KenC57, 2016-12-30:
For those who use a Windows platform, but find the need for common Unix
tools, Cygwin is a great tool set. When I need it, I'm back to the 80's and
life is simple.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2017-01-04:
@dheap Have you made the commits?

rhwood, 2017-01-25:
The ant target ```checklineends``` can now be used on Windows to verify line endings are correct before a commit. Is there need to keep this open?

dheap, 2017-01-25:
No.

closed 2017-01-25 by dheap